1000001101000/Debian_on_Buffalo #61: (Q) possible to reboot to em mode? (Not an issue, just a question)

wodgey, 2020-09-27:
Absolutely loving this. Seems much nippier than stock buffalo os. On my ls420ds.
I'd like to be able to reboot into em mode though, to use parted to shrink my share partition by 5gig, before expanding my / partition into the extra space.
I know I can force it by renaming the .buffalo files in /boot but would like to know if there's an easier method that doesn't require me to remember to rename them back again afterwards

1000001101000, 2020-09-27:
Good to hear from you!

For that type of recover/maintenance work i usually use the Debian installer, I've been meaning to post notes about how to do that somewhere. The installer doesn't load drivers for the disks until it gets to the partitioning screen so you need to run the installer until it gets to that screen, then drop to a shell and do your work.

As far as how to boot into it again, your main options are to replace the *.buffalo files as you described or boot into them via TFTP. I have some notes about TFTP booting here:
https://buffalonas.miraheze.org/wiki/Restoring_Stock_Firmware_via_TFTP

For that device you would set up your TFTP server, pull the drives, boot until it started flashing red, then push the function key twice (i think) until the led starts flashing. It should then load the files and boot into them, you could then insert the drives and continue.

1000001101000, 2020-09-27:
... you do have one other option to do what you described. It will usually let you expand partitions and filesystems while the system is running. I typically use gdisk to:
delete the partition I want to expand (assuming I already have free space after it).
re-create it with the same starting sector and the proper new size
run `resize2fs /dev/sdx2`

I've done this a few times without running into issues.

wodgey, 2020-09-28:
Thanks. The simplicity!? I'd already wiped my device once and had to move my data off my share, after my first attempt failed (I'd moved the*.buffalo files to my http server and wget'd from there. I must have forgotten to chown\chmod something as I just got constant blinking led.)
I had failed to even think of reinserting the drives during boot up and mdadm detection. Lol thanks again.
However I think I can easily live with the 5 gigs afterall. I found my disk-hog log file and moved it to my share. 
Good to know that method for the future though.

closed 2020-09-28 by wodgey

1000001101000, 2020-09-28:
I forgot you were likely using raid arrays, those you can't really resize like that. 

Honestly my best advice for a recovery environment is still the installer, but TFTP introduces so many network complications it still comes down to replacing your boot files and then remembering to restore them when done. 

Either that or use a serial console, but I aim to deliver solutions that don't require soldering.

wodgey, 2020-10-15:
Once again, thank you for pointing me at this solution for booting to the installer. 
I just needed to edit my /etc/network/interfaces file, following an install/uninstall of openmediavault and this method saved me from having to connect the drives/array into my desktop.
If anyone is interested, openmediavault installed just fine (by adding a PPA etc to APT sources, update/upgrade etc) and looked to be working. However it was awfully slow on the Marvel 370, with it being such a low power device. Hence the reason for the uninstall (besides, I saw how many services it was starting on boot up and decided "less is more")