1000001101000/Debian_on_Buffalo #62: LS-Q8.0TL/R5 with LVM2

markblinkhorn, 2020-10-20:
I have managed to install Buster on a Linkstation LS-Q8.0TL/R5 with only a little bit of difficulty. I have installed mdadm, tgt and lvm2 on it. I ran into a problem after adding lvm2. The LS fails to boot after update-initramfs -u has been run by the lvm2 install process. A bit of debugging and inspiration has led me to believe this is likely due to runsize limitation in the ramfs as the uncompressed size of the initrd.buffalo file is around 22MB (without lvm2 it is around 14MB). I have worked around the problem by removing lvm2 and thin-provisioning-tools from /etc/share/initramfs-tools/hooks. This allows the LS to boot as the /boot and /root partitions are not lvm enabled.

I'm guessing the runsize parameter need to be adjused to make the ramfs larger but I'm fumbling in the dark looking for:
a) how to adjust it and
b) how much to adjust (I know RAM is at a premium on this box)

Any clues where to look, or is my work round as good as it gets?

1000001101000, 2020-10-21:
Good to hear from you!

The older devices, and the LS-QL in particular run into issues with large initrd's. I'm actually surprised that a 14mb would boot, I thought the limit was more like 7mb for that device. Are you sure it is a LS-QL and not LS-QVL?

...I don't think the LS-QVL would run into the runsize issue so I assume you're right about the model.

There are some options for reducing the size of the initrd to allow lvm to work.

Make sure you're including as few modules as possible:
**/etc/initramfs-tools/conf.d/modules**
```
MODULES=dep
```

switch to xz compression instead of gzip:

**/etc/initramfs-tools/conf.d/compress**
```
"COMPRESS=xz"
"XZ_OPT=-2e"
"export XZ_OPT"
```

If that isn't enough you can try downgrading kmod to the Stretch version. Newer versions of kmod include a bunch of big crypto libraries to support signed kernel modules (I assume needed for secure boot etc) which end up in the initrd. I assume you could also recompile to remove that feature but downgrading seemed to work when I tried it. 

Lately I've been adjusting the /run size in the initrd so that it is correct before systemd starts:

**/etc/initramfs-tools/scripts/init-bottom/runsize.sh**
```
mount -o remount,nosuid,noexec,size=26M,nr_inodes=4096 /run
```

markblinkhorn, 2020-10-21:
I may have misled you slightly by stating the uncompressed sizes of the initrd - without lvm it is just over 6GB compressed and with lvms, just over 9GB - but those are some quality tips right there. Thank you. I will let you know how I get on.

1000001101000, 2020-10-21:
That makes sense. 

There are two main constraints for the initrd size on these devices. The first constraint comes from the bootloader, it will typically crash if an image past a certain size is loaded. That could probably be fixed by making adjustments to the bootloader but that would involve recompiling/reflashing uboot which would likely require having buffalo’s uboot source which they don’t publish. 

The other main constraint is the total amount of system memory. You need to have enough space for the uncompressed kernel, the compressed initrd, the uncompressed initrd and the space needed for xz decompression. Higher levels of xz compression require more memory to decompress. The settings I posted work for devices with 64mb ram, so the LS-QL with 128mb should be fine. 

You could try setting it even higher but I found higher settings end up adding a lot of time to generate the initrd for pretty small size reduction after a certain point.

markblinkhorn, 2020-10-21:
I'm pleased to report that it now works. 

Changing the runsize didn't make any difference; it still crashed during boot (6 red flashes - it is impossible to know why without a console, and it dies before it brings the network interface up, so netconsole is not an option).

Switching to XZ compression did work the trick though. The resulting initrd is now just under 7MB and the bootloader seems to be happy with that.

Thank you for your insightful comments and help!

Mark

1000001101000, 2020-10-21:
red flashes indicates it failed in the bootloader, way before the settings for /run come into play.

I've had to learn a lot about this piece because the installer image has to fit within the same constraints.

markblinkhorn, 2020-10-21:
Makes sense - I feel I've learned a lot on this project too!

1000001101000, 2021-06-29:
Is there anything else you want me to look at for this issue?

markblinkhorn, 2021-06-29:
Hi, All good here. System has run flawlessly since. Thanks for your input.
Mark

closed 2021-06-29 by markblinkhorn