1000001101000/Debian_on_Buffalo #63: Won't boot with 2 hdds

dejanzelic, 2020-10-21:
Hello! 

I successfully installed debian on my Buffalo LS421e. The installation was done on a single hdd and I didn't use the stock partitions. Instead, I set up the ext3 boot partition, ext3 root partition, swap partition, and ext4 mount partition. 

Now I want to add a second (ext4 partitioned) hard drive. I'm able to get it to mount, but only if I insert it just after I start boot up. I'm guessing the linkstation tries to boot off it and since it doesn't have a boot partition, it won't boot. Any suggestions on how I can force the linkstation to always boot off a specific drive? Or any other workarounds? 

The data that will be stored isn't critical and I don't need RAID.

1000001101000, 2020-10-21:
Good to hear from you!

I've never set up a serial console on this generation of device so I'm not 100% certain of the logic of how it chooses which drive to boot from. I had previously assumed that it just boots from the first drive it finds with valid boot files. If you haven't already, try switching the drive positions and see if that helps.

If that doesn't work I have some theories about why it might not work and things we can try to correct it. 

Did either of these drives previously have the stock firmware on them (even if you reformated/repartitioned after)?

 
but I believe it lists the attached drives and then tries to boot off of the first one it finds with valid boot files.

dejanzelic, 2020-10-22:
Thanks for the quick response!

Switching the drives up was my first thought too. But unfortunately that doesn't work. I just tried again to be 100% sure and I waited at least 10 minutes for it to boot up.

If it helps at all, I found the following bootlogs on openwrt: https://openwrt.org/toh/buffalo/ls421de#bootlogs

I tried openwrt which flashes itself to the nand flash but ran into a similar issue (except it wouldn't boot if I had any hdd plugged in since it was loaded on the nand flash)

dejanzelic, 2020-10-22:
In the original post I mentioned I had the LS421e but I meant LS421DE*

1000001101000, 2020-10-22:
If you reflashed the boot loader then some of what I say might not be correct. I believe the openwrt bootloader was based on the source for another device with some modifications to make it work like Buffalo's. The guy who made it wouldn't give me the source so I'm guessing about part of that. 

One thing I've noticed in the past is that the stock firmware writes the devices MAC address to the partition table. THe bootloader checks this at boot and prints a message telling whether it matches the current device's MAC or not:
https://buffalonas.miraheze.org/w/index.php?title=Terastation_TS-WXL&action=edit&section=8

it looks like the ls400 also does this:
```
MAC Address : XX:XX:XX:XX:XX:XX
MAC Address read from block device 0 : 
** Bad partition 1 **
MAC Address read from block device 1 : 
** Bad partition 1 **
```

I know the devices are fine with booting off of disks that don't have the MAC set, but I wonder if it would favor one with a matching MAC over one without even if that would otherwise make sense.

You can check the patition tables with:
```
hexdump -C /dev/sda | head -n 10
hexdump -C /dev/sdb | head -n 10

```

dejanzelic, 2020-10-22:
That is.... very interesting

Both of my hdd's have the mac address:

```
service@debian:~$ sudo hexdump -C /dev/sda | head -n 10
[sudo] password for service: 
00000000  01 00 00 00 00 00 00 00  20 00 00 00 00 00 00 00  |........ .......|
00000010  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000020  64 69 73 6b 31 00 00 00  30 30 3a 35 30 3a 34 33  |disk1...00:50:43|
00000030  3a 62 34 3a 31 33 3a 66  32 00 00 00 64 69 73 6b  |:b4:13:f2...disk|
00000040  31 00 00 00 6e 6f 72 6d  61 6c 00 00 00 00 00 00  |1...normal......|
00000050  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000001c0  01 00 ee fe ff ff 01 00  00 00 af 88 e0 e8 00 00  |................|
000001d0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
service@debian:~$ sudo hexdump -C /dev/sdb | head -n 10
00000000  01 00 00 00 00 00 00 00  20 00 00 00 00 00 00 00  |........ .......|
00000010  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000020  64 69 73 6b 31 00 00 00  30 30 3a 35 30 3a 34 33  |disk1...00:50:43|
00000030  3a 62 34 3a 31 33 3a 66  32 00 00 00 64 69 73 6b  |:b4:13:f2...disk|
00000040  31 00 00 00 6e 6f 72 6d  61 6c 00 00 00 00 00 00  |1...normal......|
00000050  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000001c0  02 00 ee ff ff ff 01 00  00 00 2f 60 38 3a 00 00  |........../`8:..|
000001d0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*

```

`/dev/sda` is my primary drive with Debian, so do you think that I should wipe `/dev/sdb` with 0's and then repartition it?

1000001101000, 2020-10-22:
I think that would be worth a try. 

As far as i know the actual code for how it makes the decision is contained in whatever customization buffalo made to that version of u-boot. unfortunately, they don't publish their u-boot source. I believe the only source they posted are just unmodified copies of mainline uboot for a couple devices if i remember correctly.

1000001101000, 2020-10-22:
more accurately I'd suggest:

1. Wipe the second drive and then try booting with it installed (trying both positions as needed)
2. if that works partition it but don't format and make sure it still boots.
3. format and see what happens.
4. if the above fail, try wiping out the MAC from the other drive too and repeat.

If all of the above fail... I guess put /boot / and swap as raid one volumes spanning the drives... hopefully you aren't forced into that.

Actually if you get to that point we should probably look at what else would be different about those drives.

dejanzelic, 2020-10-23:
OK so I was able to test it out a bit. 

1. Wipe the second drive and then try booting with it installed (trying both positions as needed) - **It Boots!** 
2. if that works partition it but don't format and make sure it still boots. - **It Boots!** 
3. format and see what happens. - It does not boot

Delete Mac and then try again:
1. Wipe the second drive and then try booting with it installed (trying both positions as needed) - **It Boots!** 
2. if that works partition it **starting at 2gb** but don't format and make sure it still boots. - **It Boots!** 
3. format and see what happens. - **It Boots!** 

Trying  again, without starting at 2gb:
1. Wipe the second drive and then try booting with it installed (trying both positions as needed) - **It Boots!** 
2. if that works partition it **starting at 1** but don't format and make sure it still boots. - **It Boots!** 
3. format and see what happens. -  It does not boot

So it looks like if there is no partition in the first 2 gigs of the hdd, the linkstation won't try to boot from that drive. I'm ok with this (either that or I waste the first 2 gigs with raid boot drive).

dejanzelic, 2020-10-24:
Nevermind about the first 2 gigs, it's no longer booting so that must have been a fluke (or I messed up on something)

1000001101000, 2020-10-24:
that does make me wonder. what happens if you label your first partition as partition 2?

1000001101000, 2020-10-24:
my current theory is that in the step where it scans each disk for partition 1 then tries to read partition 1 looking for *.buffalo files it's failing to read the partition because it's too big and is halting the process (maybe it crashes) rather than just continue to the next disk. having there be no partition 1 might help, though I'm not sure how uboot handles a partition table starting with partition 2.

dejanzelic, 2020-11-16:
Sorry for the slow response, I've been busy and haven't had a chance to mess with this. I appreciate you making yourself available to help out!

I tried looking up how to force a partition to be labels as "2" rather then "1" but couldn't find anything. Do you have any suggestions on how I should do that?

1000001101000, 2020-11-16:
one option might be to just make a really small partition1 (like 1MB) and leave it unformatted or empty.

Otherwise both gdisk and fdisk seem to be fine with making just a partition 2:
```
# gdisk /dev/sdg
GPT fdisk (gdisk) version 1.0.1

Partition table scan:
  MBR: not present
  BSD: not present
  APM: not present
  GPT: not present

Creating new GPT entries.

Command (? for help): n
Partition number (1-128, default 1): 2
First sector (34-15155166, default = 2048) or {+-}size{KMGTP}: 
Last sector (2048-15155166, default = 15155166) or {+-}size{KMGTP}: 
Current type is 'Linux filesystem'
Hex code or GUID (L to show codes, Enter = 8300): 
Changed type of partition to 'Linux filesystem'

Command (? for help): p
Disk /dev/sdg: 15155200 sectors, 7.2 GiB
Logical sector size: 512 bytes
Disk identifier (GUID): 800BC5FD-26C9-45AB-BD26-BE192A2C4E33
Partition table holds up to 128 entries
First usable sector is 34, last usable sector is 15155166
Partitions will be aligned on 2048-sector boundaries
Total free space is 2014 sectors (1007.0 KiB)

Number  Start (sector)    End (sector)  Size       Code  Name
   2            2048        15155166   7.2 GiB     8300  Linux filesystem

Command (? for help): q
# fdisk /dev/sdg

Welcome to fdisk (util-linux 2.29.2).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Device does not contain a recognized partition table.
Created a new DOS disklabel with disk identifier 0xdac14793.

Command (m for help): p
Disk /dev/sdg: 7.2 GiB, 7759462400 bytes, 15155200 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xdac14793

Command (m for help): n
Partition type
   p   primary (0 primary, 0 extended, 4 free)
   e   extended (container for logical partitions)
Select (default p): p
Partition number (1-4, default 1): 2
First sector (2048-15155199, default 2048): 
Last sector, +sectors or +size{K,M,G,T,P} (2048-15155199, default 15155199): 

Created a new partition 2 of type 'Linux' and of size 7.2 GiB.

Command (m for help): p
Disk /dev/sdg: 7.2 GiB, 7759462400 bytes, 15155200 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xdac14793

Device     Boot Start      End  Sectors  Size Id Type
/dev/sdg2        2048 15155199 15153152  7.2G 83 Linux

Command (m for help):
```
hopefully one of those two options does the trick.

dejanzelic, 2020-11-21:
I corrupted my Debian install somehow and had to reinstall it. I think this worked though! I have the partition and formatted it without any issue. Everything mounts after reboot! 

Thank you for the help, I really appreciate it!

1000001101000, 2020-11-21:
One thing to remember when making filesystem changes is to remember to test your fstab entries and run update-initramfs before rebooting. Most of the times I’ve rendered a device unable to boot it has been related to that.

Sadly, in those cases it actually boots into a recovery shell so you can try to fix the issue. Unfortunately without a serial console you would never see that shell.

Glad to hear it’s working now. What does your partition layout look like now?

dejanzelic, 2020-11-21:
Thanks for the heads up, I did not run update-initramfs yet.

Yeah I’ve had a couple times where I’m pretty sure it’s booted in the
recovery shell and I’ve had to reinstall everything.

Right now, I have a 2 TB drive that has:
1GB, ext3 partition, mounted to boot.
5gb, ext3 partition, mounted to root.
1GB, swap partition.
The remaining is mounted to /mnt as ext4.

I have another 500gb drive that just has partition 2 formatted as ext4.

I don’t need any raid for the data I’ll be storing on this. It’s all
temporary data that can be lost in case of a failure. And I don’t need
crazy speeds. So this set up will be perfect.

1000001101000, 2020-11-21:
Great!

I'll try to remember to put a note about the "partition 2 trick" in the docs somewhere in case someone else wants to run in a similar configuration.

1000001101000, 2021-06-29:
Is there anything else you want me to look at for this issue?