1000001101000/Debian_on_Buffalo #65: Problem libmicon with TS5600D

vemax78, 2020-11-01:
Hi,
I have a TeraStation TS5600D (5000 Series). I try examples of libmicon python library but I found some problems.

Work:
* Indication/ETH LED
* Brightness display LCD
* Color display LCD
* HDD LED
* Buzz

Don't work
* Text display LCD (It's always empty, no text)
* Catch pression button

1000001101000, 2020-11-01:
Good to hear from you!

It sounds like it probably isn’t setting up the communication successfully. There could be a few reasons for this.

Could you send me some more information about your device?
 Specifically:
what OS is running and how did you install it?
Also, did you set up the python serial library and if so how did you install it (apt-get pip etc)?

1000001101000, 2020-11-01:
Also, did you install debian using my installer or are you running something else?

If so it includes a script showing how to access the buttons:
https://github.com/1000001101000/Debian_on_Intel_Terastations#buttons-button_examplesh-it87-gpiopatch-build-modulessh

For the intel devices the buttons are connected to two different gpio controllers instead of the microcontroller.

vemax78, 2020-11-01:
Hi, thank you for response. I use official debian and only now I realized that I need to patch the it87 driver for the display and button.

1000001101000, 2020-11-01:
I’m guessing the LCD text will be an easy fix, that feature typically works without trouble.

vemax78, 2020-11-01:
With your build-module.sh script I compile it87, gpio-it87 and gpio-ich modules on 5.8 kernel version of my Debian. Now button_example.sh work. The LCD display still does not work, unable to show text. I already check with lsmod that the three modules were correctly loaded in memory.

1000001101000, 2020-11-01:
Could you post the code you are using to try to set the lcd?

1000001101000, 2020-11-01:
and also the console output when you run it.

vemax78, 2020-11-01:
I try to launch "micon_scripts/micon_startup.py". I have enable debug libmicon output:

Sending Preamble...
sending :  8083fd       checksum:  0    length:  0
response:  908354532d41202020202020566572312e3557       checksum:  0    length:  16     value: 54532d41202020202020566572312e35

sending :  0003fd       checksum:  0    length:  0
response:  0103f408     checksum:  0    length:  1      response code: 0xf4 ERROR

sending :  805030       checksum:  0    length:  0
response:  825000002e   checksum:  0    length:  2      value: 0000

sending :  0250000f9f   checksum:  0    length:  2
response:  015000af     checksum:  0    length:  1      response code: 0x0

sending :  80522e       checksum:  0    length:  0
response:  8252090023   checksum:  0    length:  2      value: 0900

sending :  02520900a3   checksum:  0    length:  2
response:  015200ad     checksum:  0    length:  1      response code: 0x0

sending :  80512f       checksum:  0    length:  0
response:  825100002d   checksum:  0    length:  2      value: 0000

sending :  02510000ad   checksum:  0    length:  2
response:  015100ae     checksum:  0    length:  1      response code: 0x0

sending :  20905465726173746174696f6e2058383620202044656269616e2031302e36202020f4       checksum:  0    length:  32
response:  0190006f     checksum:  0    length:  1      response code: 0x0

sending :  0025db       checksum:  0    length:  0
response:  012500da     checksum:  0    length:  1      response code: 0x0

sending :  013220ad     checksum:  0    length:  1
response:  013200cd     checksum:  0    length:  1      response code: 0x0

sending :  013affc6     checksum:  0    length:  1
response:  013a00c5     checksum:  0    length:  1      response code: 0x0

sending :  02500007a7   checksum:  0    length:  2
response:  015000af     checksum:  0    length:  1      response code: 0x0

sending :  02510002ab   checksum:  0    length:  2
response:  015100ae     checksum:  0    length:  1      response code: 0x0

sending :  805030       checksum:  0    length:  0
response:  8250000727   checksum:  0    length:  2      value: 0007

sending :  025009079e   checksum:  0    length:  2
response:  015000af     checksum:  0    length:  1      response code: 0x0

sending :  80522e       checksum:  0    length:  0
response:  8252090023   checksum:  0    length:  2      value: 0900

sending :  02520000ac   checksum:  0    length:  2
response:  015200ad     checksum:  0    length:  1      response code: 0x0

sending :  80512f       checksum:  0    length:  0
response:  825100022b   checksum:  0    length:  2      value: 0002

sending :  02510902a2   checksum:  0    length:  2
response:  015100ae     checksum:  0    length:  1      response code: 0x0

The display is green with 100% brightness but the LCD is empty.

1000001101000, 2020-11-01:
looks like all the commands are working,  that 0x0003 command failing is normal on the intel devices. 

There is a contrast 

Does the LCD show messages during startup? What does it say before you run the script?

1000001101000, 2020-11-01:
For reference, here’s what my TS5600D looks like after running the script:

[image]
[image]

1000001101000, 2020-11-01:
Again adjusting photo brightness:
[image]

vemax78, 2020-11-02:
> What does it say before you run the script?

Before ever using the script the display was always purple with a generic TeraStation message. The LED of power blink for wait complete loading. After using the script (before patch kernel module) the display is always empty even at startup. I have patch the three kernel module but the display don't show text message. Disconnect and reconnect powerline but the LCD is always empty (no text).

1000001101000, 2020-11-02:
The micon_shutdown script blanks the screen but I haven't otherwise seen that happen.

I'd suggest unplugging the device (for several minutes) and removing the drives and see what happens when you boot it.

vemax78, 2020-11-02:
Confirm that with "micon_shutdown.sh poweroff" the display is off.
Power on device without HDD the LED error is on and LCD display blink red. Always the display don't show text.

1000001101000, 2020-11-02:
hmm. I haven't had one go blank like that before. I've got an older device with a blank screen but it was like that before I got it. I've also read forum posts of people whose displays have gone blank.

I don't have much advice for trying to fix it.

vemax78, 2020-11-03:
I found a solution to restore the text on display. I created a USB recovery firmware and booted without HDD. Booting this firmware, it has restored the text on the LCD. I turned off the NAS, removed the usb stick, reconnected the HDDs and turned on. The display continues to work and it has shown all various phases of the boot. I tried the python scripts again and now correctly show the text on LCD display.

1000001101000, 2020-11-03:
Nice! I wonder what the recovery disk did exactly. 

I'll have to remember that trick if someone else runs into this.

1000001101000, 2021-06-29:
Is there anything else you want me to look at for this issue?

vemax78, 2021-06-29:
No thanks

closed 2021-06-29 by 1000001101000