1000001101000/Debian_on_Buffalo #66: /sbin/ missing from path

mas94uk, 2020-11-03:
After install (on LS220d), the path is
/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games

Several tools (mdadm, update-initramfs and its dependencies) are in /sbin/ and hence don't work.

1000001101000, 2020-11-03:
Always good to hear from you!

I believe this was a deliberate change in Buster (I also think it was standard in other distros like RHEL):
https://wiki.debian.org/NewInBuster

The typical solution is to add "-" to su commands
`su - `
`sudo su -`

You could also add it to your non-root user's profile if you want.

1000001101000, 2021-03-04:
is there anything you want me to look at for this issue?

mas94uk, 2021-03-09:
Thanks for getting back. No, it looks like it is indeed a (slightly strange IMO) Debian thing, so I agree it's sensible to inherit it.

In case anyone else hits this in the future, my solution/workaround was to add the following to the end of ~/.bashrc:

```
# Because Debian path doesn't include these, for some reason
export PATH=$PATH:/usr/local/sbin:/usr/sbin:/sbin
```