1000001101000/Debian_on_Buffalo #68: No IP after reboot in LS210D

dplasa, 2020-11-27:
I've tried method #1 (https://github.com/1000001101000/Debian_on_Buffalo/wiki/Preparing-Device-for-Debian-Install#option-1-load-onto-device-running-the-stock-firmware) but after I copy initrd.buffalo and uImage.buffalo to /boot and reboot, the device doesn't get an IP.

I've noticed with original firmware and telnetting to it that it uses eth1 as net device
```
[root@LS210DAE9 ~]# ifconfig 
eth1      Link encap:Ethernet  HWaddr 18:C2:BF:EB:2A:E9  
```
and that both eth0 and eth1 are present:
```
[root@LS210DAE9 ~]# cat /proc/net/dev
Inter-|   Receive                                                |  Transmit
 face |bytes    packets errs drop fifo frame compressed multicast|bytes    packets errs drop fifo colls carrier compressed
    lo:    1548      20    0    0    0     0          0         0     1548      20    0    0    0     0       0          0
  eth1:   41353     347    0    0    0     0          0         0    43447     290    0    0    0     0       0          0
  eth0:       0       0    0    0    0     0          0         0        0       0    0    0    0     0       0          0
```

What are my best options to proceed?

1000001101000, 2020-11-27:
Good to hear from you!

There are a couple of possibilities. The most common issue people run into with this step is accidentally downloading the GitHub HTML page rather than the actual installer files. If the files are correct they should be 4MB and 13MB, if they are only a few KB that's likely what happened. 

If that isn't the problem there are some other possibilities to explore. It is possible (though pretty unlikely) for it to fail t o set the MAC address properly at boot. In that case I'd expect to still see it get an IP address but not the same one as before. You could try rebooting it and see if it comes up as expected the second time. 

If you still have trouble we'll need a little more information:
What state are the leds on the front of the device? are they (red or white) (blinking or solid)?
Does your network use DHCP or do you set your ip addresses?

dplasa, 2020-11-27:
The files are the correct sizes, I checked. 
My router doesn't see a new MAC, it looks like it doesn't come up properly.
When turning on I see the led sequence 
red -- 1/4s
white blinking (fast) - some seconds
white (perm on)

From the hdd access it sounds like it's accessing the hdd only for a very short time. Then it doesn't do much.
The original FW does the same, waits for a short time, then the hdd is accessed again for more...

dplasa, 2020-11-27:
Blimey! And now, doing the same thing for the 3rd time, it just came up and I connect via ssh installer@... 
Soory!!!!

closed 2020-11-27 by dplasa

1000001101000, 2020-11-27:
That is all consistent with it booting into the installer successfully at least.

I believe you're hearing the bootloader load the files from the disk and boot from them. The installer itself runs from memory and doesn't access the disks until you get to the partitioning menu (or shortly before). 

The installer has to unpack itself and runs through a bunch of steps before it brings up the network so it can take a minute or two. I assume you've waited long enough before checking but i thought I'd mention it anyway. 

If it's getting far enough to change the white LED to solid it's getting to about the same point the network should come up. I'll try loading the install on one of my ls200's quick to see if there's something wrong with the latest files before suggesting other things to check. 

Which version of Debian are you trying to install?

1000001101000, 2020-11-27:
hmm. I'd expect greater than 1/3 success rate on booting but it should be fine once it gets going. 

Let me know if you run into any trouble.

dplasa, 2020-11-27:
I'm installing buster -- so far, all seems good. Thank you very much for your fast answer! I really appreciate it!