1000001101000/Debian_on_Buffalo #71: TS-WXL - doesn't boot after installation

retikulumx, 2020-12-28:
Hi.
I tried to install Debian Buster now several times and spent hours but it just won't boot and hangs in EM.

It's the TS-WXL 2-bay. I use the partitions from the original firmware, 1 GB for /boot and 5.1 GB for / . Both in ext3 .

Tell me if you need more information.

Any help is appreciated.

1000001101000, 2020-12-28:
Good to hear from you!

This behavior indicated that something is wrong with the boot partition. There are a few different possible issues, it sounds like you've already looked at a few of them. 

Usually the issue is related to the partition layout, could you start the installer again and then post a screenshot of the partitioning screen before you continue past that point? I can then take a look and see if I can see an issue.

It's also possible that something is going wrong at the end of the installer where the new boot files are created. I'll try and run through it with mine and see if I run into any issues with that part.

I've automated the process of generating installer images which makes it possible to keep up with Debian's changes but also means if there is a problem I don't always notice right away.

retikulumx, 2020-12-28:
Hi.
Thanks for your quick response. The screenshots are attached.

Thanks for your help :)

[image]
[image]

1000001101000, 2020-12-28:
That looks about right, "RAID1 device #0" needs to be made up of sda1 and sdb1, I assume it is but the image doesn't really show.

Could you run through the rest of the install but without rebooting at the end? We can then connect to it in a second ssh session and look around for the problem.

1000001101000, 2020-12-28:
I ran the install on mine and got the same issue, I'm working on a fix now. I should have updated installer images shortly.

retikulumx, 2020-12-28:
Ah great, thank you man! So I'm gonna switch that NAS off for today and wait for your fix :-)

1000001101000, 2020-12-28:
Turns out the fix doesn't require new images, just had to update the kernel repo.... long story.

It should work now, I'm kicking off the installer to verify but it should be working when you get a chance to try it.

retikulumx, 2020-12-28:
Haha, simple fix :-)
I'll give it another try tomorrow and give you feedback. 
It's just after midnight here in Germany. Thanks and good night.

1000001101000, 2020-12-29:
My install succeeded this time.

retikulumx, 2020-12-29:
Great, the installation succeeded for me too now. Thanks very much for the fast fix! :-)
Now I have to set up everything I need ^^.

1000001101000, 2021-03-04:
I'm gonna close this out, let me know if you need anything else!

closed 2021-03-04 by 1000001101000

minnsou, 2025-05-29:
Hi @1000001101000,

First of all, thank you very much for maintaining this project. Your work makes it possible to keep these Buffalo NAS devices useful with modern Linux distributions — I really appreciate it.

I tried to install Debian Bookworm on a TS-XHL device and ran into the same issue where the system fails to boot after installation.

Here are the relevant details of my setup and findings:

I confirmed the partition layout is correct. `md0` is formatted as ext3 and is composed of `sda1`, `sdb1`, `sdc1`, and `sdd1`.
```
cat /proc/mdstat

md0 : active raid1 sda1[0] sdd1[3] sdc1[2] sdb1[1]
      1000384 blocks [4/4] [UUUU]
```

After the installation completed, I noticed that `/boot` was missing the expected kernel and initrd files. 
I dropped into ash and checked `/target/boot`.

```
~ # ls /target/boot/
bootshim    lost+found
~ # 
```

And I also checked `/var/log/syslog`, and found that the linux-image-marvell-buffalo package had not been installed. Here's the relevant part of the log:

```
May 29 04:47:10 in-target: Hit:1 http://security.debian.org/debian-security bookworm-security InRelease
May 29 04:47:10 in-target: Hit:2 http://http.us.debian.org/debian bookworm InRelease
May 29 04:47:10 in-target: Hit:3 http://http.us.debian.org/debian bookworm-updates InRelease
May 29 04:47:10 in-target: Get:4 https://raw.githubusercontent.com/1000001101000/Debian_on_Buffalo/master/PPA bookworm InRelease [1,642 B]
May 29 04:47:16 in-target: Get:5 https://raw.githubusercontent.com/1000001101000/Debian_on_Buffalo/master/PPA bookworm/main armel Packages [6,838 B]
May 29 04:47:17 in-target: Fetched 8,480 B in 7s (1,156 B/s)
May 29 04:47:17 in-target: Reading package lists...
May 29 04:47:40 in-target: 
May 29 04:47:41 in-target: W
May 29 04:47:41 in-target: : 
May 29 04:47:41 in-target: https://raw.githubusercontent.com/1000001101000/Debian_on_Buffalo/master/PPA/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/tru
sted.gpg), see the DEPRECATION section in apt-key(8) for details.
May 29 04:47:41 in-target: 
May 29 04:47:41 in-target: Reading package lists...
May 29 04:47:43 in-target: 
May 29 04:47:43 in-target: Building dependency tree...
May 29 04:47:45 in-target: 
May 29 04:47:45 in-target: Reading state information...
May 29 04:47:45 in-target: 
May 29 04:47:46 in-target: Some packages could not be installed. This may mean that you have
May 29 04:47:46 in-target: requested an impossible situation or if you are using the unstable
May 29 04:47:46 in-target: distribution that some required packages have not yet been created
May 29 04:47:46 in-target: or been moved out of Incoming.
May 29 04:47:46 in-target: The following information may help to resolve the situation:
May 29 04:47:46 in-target: 
May 29 04:47:46 in-target: The following packages have unmet dependencies:
May 29 04:47:47 in-target:  linux-image-marvell-buffalo : Depends: linux-libc-dev (= 6.1.137-33) but 6.1.140-1 is to be installed
May 29 04:47:47 in-target: E
May 29 04:47:47 in-target: : 
May 29 04:47:47 in-target: Unable to correct problems, you have held broken packages.
May 29 04:47:47 in-target: 
May 29 04:47:48 in-target: Created symlink /etc/systemd/system/multi-user.target.wants/micon_boot.service → /etc/systemd/system/micon_boot.service.
May 29 04:47:48 in-target: Created symlink /etc/systemd/system/multi-user.target.wants/micon_fan_daemon.service → /etc/systemd/system/micon_fan_daemon.service.

```

To resolve this, I chrooted into the target system and manually installed the kernel:

```
~ # mount --bind /dev /target/dev
~ # mount --bind /dev/pts /target/dev/pts
~ # mount --bind /proc /target/proc
[... 53 lines cut ...]
Setting up linux-headers-6.1.137 (6.1.137-33) ...
Setting up linux-image-6.1.137 (6.1.137-33) ...
update-initramfs: Generating /boot/initrd.img-6.1.137
bootshim installed to /boot/vmlinuz-6.1.137
flash-kernel: deferring update (trigger activated)
flash-kernel: deferring update (trigger activated)
Setting up linux-image-marvell-buffalo (6.1.137-33) ...
Processing triggers for flash-kernel (3.107) ...
flash-kernel: installing version 6.1.137
Generating kernel u-boot image... done.
Installing new uImage.buffalo.
Generating initramfs u-boot image... done.
Installing new initrd.buffalo.
root@debian:/# ls /boot
System.map-6.1.137  bootshim  config-6.1.137  initrd.buffalo  initrd.img-6.1.137  lost+found  uImage.buffalo  vmlinuz-6.1.137
root@debian:/# 
```

After installing linux-image-marvell-buffalo, Debian 12.11 was able to boot successfully.

It seems like the installer failed to install the correct kernel package automatically. I hope this information is helpful for others facing a similar issue, and possibly for improving the installer process.

Thanks again for your great work!

Best regards,