1000001101000/Debian_on_Buffalo #72: TS-HTGL/R5: Boot from HDD, then run from memory stick, how?

cdwijs, 2020-12-28:
Hi There,

I have a Buffalo Terrastation Pro TS-HTGL/R5 TS-H4.0TGL/R5 BUF-TS-HTGLR5(A). I can use your installer just fine to install Debian 10.7 on it.

In order to save noise and heat, I would like to keep the hard drives in idle, until a file is requested from the network. To this end I would like to put the root filesystem on a thumb drive. Problem is that the debian installer does not allow partitions and mountpoints to be made on the thumb drive. After the NAS is booted completely, Debian can access the thumb drive without issue.

I guess both the installer initrd.buffalo / uImage.buffalo and the installed Debian initrd / uImage pair don't have the driver to access a USB thumb drive.

I've tried to generate an initrd.buffalo with most of the drivers, but that does not boot. I suspect it's too large (>7MB), as stated here: https://github.com/1000001101000/Deb...falo/issues/67

What driver must be included in the initrd to allow for root filesystem on a memory stick?

Here's what I've done:
# cat /etc/initramfs-tools/conf.d/modules 
MODULES=dep
# nano /etc/initramfs-tools/conf.d/modules 
# cat /etc/initramfs-tools/conf.d/modules 
MODULES=most

# apt-get reinstall linux-image-4.12.13
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
Need to get 22.2 MB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 https://raw.githubusercontent.com/1000001101000/Debian_on_Buffalo/master/PPA buster/main armel linux-image-4.12.13 armel 4.12.13-4 [22.2 MB]
Fetched 22.2 MB in 15s (1,495 kB/s)                                                                                
(Reading database ... 27554 files and directories currently installed.)
Preparing to unpack .../linux-image-4.12.13_4.12.13-4_armel.deb ...
Unpacking linux-image-4.12.13 (4.12.13-4) over (4.12.13-4) ...
Setting up linux-image-4.12.13 (4.12.13-4) ...
update-initramfs: Generating /boot/initrd.img-4.12.13
I: The initramfs will attempt to resume from /dev/sda2
I: (UUID=a4ebb9fd-00ba-46ff-a448-165d6e21a7b8)
I: Set the RESUME variable to override this.
flash-kernel: deferring update (trigger activated)
flash-kernel: deferring update (trigger activated)
Processing triggers for flash-kernel (3.99) ...
flash-kernel: installing version 4.12.13
Generating kernel u-boot image... done.
Taking backup of uImage.buffalo.
Installing new uImage.buffalo.
Generating initramfs u-boot image... done.
Taking backup of initrd.buffalo.
Installing new initrd.buffalo.

# ls -l /boot/
total 44140
-rw-r--r-- 1 root root   164677 Nov  3 15:19 config-4.12.13
drwxr-xr-x 2 root root     4096 Dec 28 08:56 hidden
-rw-r--r-- 1 root root 15327846 Dec 28 10:07 initrd.buffalo
-rw-r--r-- 1 root root  6551033 Dec  9 11:32 initrd.buffalo.bak
-rw-r--r-- 1 root root 15327782 Dec 28 10:07 initrd.img-4.12.13
drwx------ 2 root root    16384 Dec  9 05:50 lost+found
-rw-r--r-- 1 root root  1549426 Nov  3 15:19 System.map-4.12.13
-rw-r--r-- 1 root root  2055672 Dec 28 10:07 uImage.buffalo
-rw-r--r-- 1 root root  2055672 Dec  9 11:32 uImage.buffalo.bak
-rwxr-xr-x 1 root root  2055600 Nov  3 15:19 vmlinuz-4.12.13

# file initrd.buffalo
    initrd.buffalo: u-boot legacy uImage, ramdisk 4.12.13, Linux/ARM, RAMDisk Image (Not compressed), 15327782 bytes, Mon Dec 28 15:07:50 2020, Load Address: 0x00000000, Entry Point: 0x00000000, Header CRC: 0x730B771C, Data CRC: 0x7B7484E3

# file initrd.buffalo.bak
initrd.buffalo.bak: u-boot legacy uImage, ramdisk 4.12.13, Linux/ARM, RAMDisk Image (Not compressed), 6550969 bytes, Wed Dec  9 16:32:05 2020, Load Address: 0x00000000, Entry Point: 0x00000000, Header CRC: 0xAF161477, Data CRC: 0x5B936D6E

# mkimage -l initrd.buffalo
Image Name:   ramdisk 4.12.13
Created:      Mon Dec 28 16:07:50 2020
Image Type:   ARM Linux RAMDisk Image (uncompressed)
Data Size:    15327782 Bytes = 14968.54 KiB = 14.62 MiB
Load Address: 00000000
Entry Point:  00000000

# mkimage -l initrd.buffalo.bak 
Image Name:   ramdisk 4.12.13
Created:      Wed Dec  9 17:32:05 2020
Image Type:   ARM Linux RAMDisk Image (uncompressed)
Data Size:    6550969 Bytes = 6397.43 KiB = 6.25 MiB
Load Address: 00000000
Entry Point:  00000000

1000001101000, 2020-12-28:
I believe I omitted the USB drivers from the installer to keep the size below 7MB. That'll make moving things around a little more difficult but it looks like you have the right idea. 

I've done something similar before though only on a temporary basis, here is how I would proceed (you may have already done some of this).

1. Setup the device with RAID1 arrays for /boot, roofs and swap
2. partition your USB drive to match the HDDs for those partitions
3. add the USB drive to all 3 RAID arrays and wait for it to resync
4. remove the HDDs from the rootfs and swap arrays (resize to "1" so they do not appear as degraded) 
   *You still want the HDDs part of /boot since it will still be booting from hdd, you only actually need to mount it when updating kernel/initramfs 
   *swap on usb isn't great but 128MB and no swap also isn't great, you may need to tune things to fit your needs in that regard)
   * keeping rootfs/etc as RAID1 gives you options to switch back in the future or switch usb drives...etc.
5. run update-initramfs -u to get (still with modules=dep) 
6. reboot and hopefully have everything work as desired

if the usb modules still don't get into the initrd, determine the required modules by looking at the output of lsmod. Then add them to /etc/initramfs-tools/modules and generate a new initrd.

if the initramfs is larger than 6MB or so I'd switch to XZ compression to make it smaller:
try creating /etc/initramfs-tools/conf.d/compress with:
COMPRESS=xz
XZ_OPT=-2e
export XZ_OPT

cdwijs, 2020-12-29:
Thanks. What gave you the information that there's a sizelimit of 7MB? Where can I find how much it is exactly, and can this be increased somehow?

1000001101000, 2020-12-29:
I have learned about most of the limitations through trial and error over the years. Most are related to the version/build of uboot these devices use for their bootloader. Buffalo hasn't published their uboot source for most devices which limits the available information to some degree. 

Up until very recently I have avoided messing with uboot entirely on these devices because most models are difficult to get a working serial console and because I don't want any possibility of "bricking" devices by using any of my tools. I did make an exception recently for the ls400 since someone on the openwrt project had created a uboot image that fixed an ethernet problem that affected some of those devices.

Here are some of the limitations I'm aware of :

- Most Orion builds of uboot (not just Buffalo's) will crash if you try to load a kernel >2MB
- The device needs to have enough memory to unpack the kernel and have enough memory to unpack the initrd (packed size+unpacked size + overhead from decompression alg). This is mainly only a problem for devices with <128MB and using XZ with higher settings.
- The initrd location and a maximum size are passed as a kernel parameter ex "initrd=0x00800040,15M" if you try loading it elsewhere you need to change that too.
- I believe the actual 7MB limit has something to do with newer Linux kernels not liking the default address, at one point I was loading a much larger initrd on a Terastation III by moving it elsewhere in memory.

Once I figured out how to get the installer below 7MB by using XZ compression and some other tricks I haven't spent much more time looking into the whole initrd thing. Using the XZ settings listed above normal initrd's come out to like 3MB even with some additional drivers added which is good enough for most use cases.

cdwijs, 2021-01-02:
Thanks. the compress trick got it down from 6.3MB to 4.6MB, and the NAS boots from it, but strangely the image does not look compressed. Next step: adding the USB drivers:

# cat /etc/initramfs-tools/conf.d/compress 
COMPRESS=xz
XZ_OPT=-2e
export XZ_OPT

# cat /etc/initramfs-tools/conf.d/modules 
MODULES=dep

# apt-get reinstall linux-image-4.12.13
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
Need to get 0 B/22.2 MB of archives.
After this operation, 0 B of additional disk space will be used.
(Reading database ... 27775 files and directories currently installed.)
Preparing to unpack .../linux-image-4.12.13_4.12.13-4_armel.deb ...
Unpacking linux-image-4.12.13 (4.12.13-4) over (4.12.13-4) ...
Setting up linux-image-4.12.13 (4.12.13-4) ...
update-initramfs: Generating /boot/initrd.img-4.12.13
I: The initramfs will attempt to resume from /dev/sde2
I: (UUID=f41c8f77-14d3-4906-8acc-0c935040b0d4)
I: Set the RESUME variable to override this.
flash-kernel: deferring update (trigger activated)
flash-kernel: deferring update (trigger activated)
Processing triggers for flash-kernel (3.99) ...
flash-kernel: installing version 4.12.13
Generating kernel u-boot image... done.
Taking backup of uImage.buffalo.
Installing new uImage.buffalo.
Generating initramfs u-boot image... done.
Taking backup of initrd.buffalo.
Installing new initrd.buffalo.

# mkimage -l /boot/initrd.buffalo
Image Name:   ramdisk 4.12.13
Created:      Sat Jan  2 03:58:40 2021
Image Type:   ARM Linux RAMDisk Image (uncompressed)
Data Size:    4809996 Bytes = 4697.26 KiB = 4.59 MiB
Load Address: 00000000
Entry Point:  00000000

# file /boot/initrd.buffalo
/boot/initrd.buffalo: u-boot legacy uImage, ramdisk 4.12.13, Linux/ARM, RAMDisk Image (Not compressed), 4809996 bytes, Sat Jan  2 08:58:40 2021, Load Address: 0x00000000, Entry Point: 0x00000000, Header CRC: 0x7FAB6BBC, Data CRC: 0xADF5BA00

cdwijs, 2021-01-02:
I've added the USB drivers, and the memory stick is detected before the root filesystem is mounted. My hard drive is /dev/sda, my flash drive is /dev/sde. The flash drive is found before /dev/sda3 is mounted as /

Now I have two options:
A) Figure out what in the initrd deceides where the root filesystem is.
B) Add the USB drivers to this file, and install debian on my flash drive:
https://github.com/1000001101000/Debian_on_Buffalo/raw/master/Buster/installer_images/armel_devices/initrd.buffalo

These messages show the USB stick is found before the root filesystem is mounted:
[   21.752980] usbcore: registered new interface driver usbfs
[   21.788739] usbcore: registered new interface driver hub
[  21.830863] usbcore: registered new device driver usb
[   21.891032] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[   21.934245] ehci-orion: EHCI orion driver
[   23.289357] usb-storage 1-1.1:1.0: USB Mass Storage device detected
[   23.296286] scsi host4: usb-storage 1-1.1:1.0
[   23.306380] usbcore: registered new interface driver usb-storage
[   23.321324] usbcore: registered new interface driver uas
[   24.480029] sd 4:0:0:0: [sde] 30464000 512-byte logical blocks: (15.6 GB/14.5 GiB)
[   24.526640] sd 4:0:0:0: [sde] Write Protect is off
[   24.531471] sd 4:0:0:0: [sde] Mode Sense: 43 00 00 00
[   24.544039] sd 4:0:0:0: [sde] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[   24.596973]  sde: sde1 sde2
[   24.628782] sd 4:0:0:0: [sde] Attached SCSI removable disk
[   26.977918] EXT4-fs (sda3): mounted filesystem with ordered data mode. Opts: (null) <= mounting of the root filesystem

# cat /etc/initramfs-tools/modules 
# List of modules that you want to include in your initramfs.
# They will be loaded at boot time in the order below.
#
# Syntax:  module_name [args ...]
#
# You must run update-initramfs(8) to effect this change.
#
# Examples:
#
# raid1
# sd_mod
uas
usb_storage
usbcore
ehci_hcd
ehci_orion
usb_common
nls_base
scsi_mod

# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.12.13 (jeremy@earth) (gcc version 6.3.0 20170516 (Debian 6.3.0-18) ) #4 Tue Nov 3 14:18:55 CST 2020
[    0.000000] CPU: Feroceon [41069260] revision 0 (ARMv5TEJ), cr=0005317f
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine: Buffalo Terastation Pro II/Live
[    0.000000] Clearing invalid memory bank 0KB@0xffffffff
[    0.000000] Clearing invalid memory bank 0KB@0xffffffff
[    0.000000] Clearing invalid memory bank 0KB@0xffffffff
[    0.000000] Ignoring unrecognised tag 0x00000000
[    0.000000] Ignoring unrecognised tag 0x00000000
[    0.000000] Ignoring unrecognised tag 0x00000000
[    0.000000] Ignoring unrecognised tag 0x41000403
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 32768
[    0.000000] free_area_init_node: node 0, pgdat c0598e3c, node_mem_map c7ef9000
[    0.000000]   Normal zone: 256 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 32512
[    0.000000] Kernel command line: console=ttyS0,115200 root=/dev/sda2 rw initrd=0x00800040,15M panic=5 BOOTVER=1.19
[    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
[... 287 lines cut ...]
[   39.920763] raid6: int32x4  xor()    47 MB/s
[   39.993431] raid6: int32x8  gen()    59 MB/s
[   40.065193] raid6: int32x8  xor()    44 MB/s
[   40.069500] raid6: using algorithm int32x8 gen() 59 MB/s
[   40.074839] raid6: .... xor() 44 MB/s, rmw enabled
[   40.079651] raid6: using intx1 recovery algorithm
[   40.116376] xor: measuring software checksum speed
[   40.160569]    arm4regs  :   428.000 MB/sec
[   40.204559]    8regs     :   352.000 MB/sec
[   40.248564]    32regs    :   427.000 MB/sec
[   40.252770] xor: using function: arm4regs (428.000 MB/sec)
[   40.318160] number of JEDEC chips: 1
[   41.019826] Btrfs loaded, crc32c=crc32c-generic
[   41.051280] BTRFS: device label SLAVE_NAS devid 1 transid 5257 /dev/sdd6
[   41.210306] BTRFS: device fsid 7e08214e-3da1-4d58-9858-b8fa09098465 devid 1 transid 104 /dev/sda4
[   42.420385] BTRFS: device label SLAVE_NAS devid 2 transid 5257 /dev/sdc6
[   46.276366] EXT4-fs (sda1): mounting ext3 file system using the ext4 subsystem
[   46.325151] EXT4-fs (sda1): mounted filesystem with ordered data mode. Opts: (null)
[   47.410880] mv643xx_eth_port mv643xx_eth_port.0 eth0: link down
[   47.428766] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   47.632727] NET: Registered protocol family 17
[   50.989573] mv643xx_eth_port mv643xx_eth_port.0 eth0: link up, 1000 Mb/s, full duplex, flow control enabled
[   50.999402] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   70.952595] usb 1-1.1: reset high-speed USB device number 3 using orion-ehci
[   71.722579] Adding 1048572k swap on /dev/sde2.  Priority:-1 extents:1 across:1048572k FS

1000001101000, 2021-01-02:
One thing to note is that the initrd will set the  rootfs based on what you have in your /etc/fstab.

could you post the output of lsblk?

cdwijs, 2021-01-02:
Thanks, I have first regenerated initrd, and only afterwards edited my /etc/fstab. I should have regenerated initrd *after* editing /etc/fstab.

After doing the below steps again, and a reboot it works! I now have a NAS that runs from a memory stick :-)

Thanks for all your help, I doubt if I would have succeeded without your knowledge and encouragement.

I want to boot from /dev/sde1
# lsblk
NAME      MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda         8:0    0   149G  0 disk 
├─sda1      8:1    0   512M  0 part /boot
├─sda2      8:2    0     1G  0 part 
├─sda3      8:3    0    10G  0 part /
└─sda4      8:4    0 137.5G  0 part 
sdb         8:16   0 298.1G  0 disk 
sdc         8:32   0 931.5G  0 disk 
├─sdc1      8:33   0    30G  0 part 
├─sdc4      8:36   0     1K  0 part 
├─sdc5      8:37   0     1K  0 part 
└─sdc6      8:38   0 901.5G  0 part 
sdd         8:48   0 931.5G  0 disk 
├─sdd1      8:49   0    30G  0 part 
├─sdd4      8:52   0     1K  0 part 
├─sdd5      8:53   0     1K  0 part 
└─sdd6      8:54   0 901.5G  0 part 
sde         8:64   1  14.5G  0 disk 
├─sde1      8:65   1     5G  0 part 
└─sde2      8:66   1     1G  0 part [SWAP]
mtdblock0  31:0    0   256K  0 disk

# ls -l /dev/disk/by-uuid/
total 0
lrwxrwxrwx 1 root root 10 Jan  2 07:09 24a727b4-771b-4e81-8822-a6d241623136 -> ../../sde1

# mount /dev/sde1 /mnt/sde1
# rsync -aAXv /* /mnt/sde1/ --exclude={/boot/*,/dev/*,/proc/*,/sys/*,/tmp/*,/var/tmp/*,/run/*,/mnt/*,/media/*,/lost+found,/home/cedric/*,/swapfile,/storage/*,/opt/*,/root/*}
# umount /dev/sde1

# cat /etc/fstab 
# /etc/fstab: static file system information.
# <file system> <mount point>   <type>  <options>       <dump>  <pass>
UUID=24a727b4-771b-4e81-8822-a6d241623136 /               ext4    errors=remount-ro 0       1
UUID=2bb94806-50be-46b4-a036-c82ec89bc765 /boot           ext3    defaults        0       2
UUID=f41c8f77-14d3-4906-8acc-0c935040b0d4 none            swap    sw              0       0
#69 

# apt-get reinstall linux-image-4.12.13
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
Need to get 0 B/22.2 MB of archives.
After this operation, 0 B of additional disk space will be used.
(Reading database ... 27775 files and directories currently installed.)
Preparing to unpack .../linux-image-4.12.13_4.12.13-4_armel.deb ...
Unpacking linux-image-4.12.13 (4.12.13-4) over (4.12.13-4) ...
Setting up linux-image-4.12.13 (4.12.13-4) ...
update-initramfs: Generating /boot/initrd.img-4.12.13
I: The initramfs will attempt to resume from /dev/sde2
I: (UUID=f41c8f77-14d3-4906-8acc-0c935040b0d4)
I: Set the RESUME variable to override this.
flash-kernel: deferring update (trigger activated)
flash-kernel: deferring update (trigger activated)
Processing triggers for flash-kernel (3.99) ...
flash-kernel: installing version 4.12.13
Generating kernel u-boot image... done.
Taking backup of uImage.buffalo.
Installing new uImage.buffalo.
Generating initramfs u-boot image... done.
Taking backup of initrd.buffalo.
Installing new initrd.buffalo.

The UUID from my memory stick is now correct in initrd.buffalo:
# tail -c+65 < /boot/initrd.buffalo > ~/out
# pax -J -r < ~/out
# cat conf/param.conf 
ROOT="/dev/disk/by-uuid/24a727b4-771b-4e81-8822-a6d241623136"

#/sbin/reboot
# df -h
Filesystem      Size  Used Avail Use% Mounted on
udev             53M     0   53M   0% /dev
tmpfs            26M  1.7M   25M   7% /run
/dev/sde1       4.9G  935M  3.7G  20% /   <YESSSS!!!!!
tmpfs            61M     0   61M   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs            61M     0   61M   0% /sys/fs/cgroup
/dev/sda1       492M   61M  406M  13% /boot
tmpfs            13M     0   13M   0% /run/user/1000

1000001101000, 2021-01-02:
Great!

closed 2021-01-02 by cdwijs