1000001101000/Debian_on_Buffalo #73: "Connection refused" when attempting to SSH in: LS-WXL

pdaoust, 2021-01-05:
Hello, I've gotten an LS-WXL device and I appear to have successfully installed [the Buster installer and the initrd](https://github.com/1000001101000/Debian_on_Buffalo/blob/master/Buster/installer_images/armel_devices/uImage.buffalo.lswxl) for my device via TFTP -- at least, they weren't on my boot partition when I booted up the machine, and they were when I pulled the hard drive and mounted it on my laptop. The device seems to boot up to a steady blue power light with no flashing function or error lights, but... I can't SSH in. Are there any causes that come to mind, or other ways to tell if my device is indeed booting into the installer image?

Thanks in advance; this project is a beautiful example of human generosity.

1000001101000, 2021-01-05:
Good to hear from you!

If the light turn solid it means that it successfully booted into something. Most likely that would be the installer, if you saw it download the files from your TFTP program then would almost certainly have been the installer. The only other possibility would be it booting from NAND into EM mode in which case you might be able to see it in LSUpdater or NasNavigator but I doubt that's what happened (most of those devices don't have NAND).

I think I know what's going on, the details depend on how the network that the device is connected to is set up.

The installer starts with a very minimal environment which needs to fit into a 7MB file, it needs a connection to the internet so that it can download the ssh modules from a Debian mirror before it can start. If your network configuration doesn't provide DHCP and a path to the internet (I assume you had to set up a special 192.168.11.* for TFTP) this will fail and the installer won't do anything. 

I just pushed a change to the Buster armel installer to allow it to fall back to 192.168.11.150 if DHCP fails, if your network has internet access but not DHCP the new installer files might fix the issue. If it doesn't have internet access you'll need to connect it to a network that does before it finishes booting.

Let me know if that makes sense, I'm having trouble organizing my thoughts today and am not sure if I'm being clear.

1000001101000, 2021-01-05:
Yeah uImage.buffalo.lswxl (renamed to uImage.buffalo) and initrd.buffalo are the right files. 

Getting the stock firmware working and installing from there is the easiest method for most people, as long as the disk layout the stock firmware creates works for them. On the really old devices (older than yours) the roofs is too small for a decent Debian install which causes some issues. 

If you have a linux system already I've also created a method that just generates a disk image rather than using the installer:
https://github.com/1000001101000/Debian_on_Buffalo/wiki/Alternate-install-method-via-debootstrap-script

It makes it a little easier to customize the disk layout and doesn't require getting the installer working but is generally more complicated.

As far as getting paid goes....lol, I wish! 

I mostly count my time as trying to give back to the open source community generally since it has benefited me so much over the years.

I did enable the sponsorship feature so people can throw me a few dollars if they want to though I haven't pushed it beyond enabling the button.

pdaoust, 2021-01-06:
IT WORKS!!!!!!!! I can now boot up into my LS-WXL; feels so good to have an up-to-date Debian distro on that thing. Thanks so much for the tip; somehow I missed that debootstrap method the first time around (probably glossed over it because I didn't know what 'debootstrap' was). Honestly it was the easiest method I'd tried by far. There were a couple tweaks -- I had to install `mdadm` and run your script in root -- but otherwise it worked just dandy on my Ubuntu 20.10 system. Thank you, thank you, thank you.

closed 2021-01-06 by pdaoust

1000001101000, 2021-01-06:
Nice!

I need to start pushing the debootstrap method more, I prefer it a lot over the installer now that I have it working.