1000001101000/Debian_on_Buffalo #74: Stuck in Bootloop after exchanging boot files

hakimh02, 2021-01-05:
Hi guys, 

strange one here I believe. So I've followed the guide and have replaced the boot files with the ones from the git. 
After rebooting the NAS, it's somehow stuck in a bootloop. The power LED is constantly blinking (for now more than 40 mins) and I didn't succeed in establishing a connection to the NAS yet.

1000001101000, 2021-01-05:
Good to hear from you!

That generally means that it was able to find the boot files but failed to boot from them. What is the file size of the two *.buffalo devices?

hakimh02, 2021-01-05:
I have two 3 TB hard drives. I tried plugging in just one to see if it boots, with the same result.

1000001101000, 2021-01-05:
What model is the device?

hakimh02, 2021-01-05:
Sorry, forgot to mention that. It's the LinkStation LS220D.

hakimh02, 2021-01-05:
Well, the first issue is that I cannot connect to the nas anymore. It's not showing up in this state. I scanned the network, and it's nowhere to be found. 

I've downloaded the files from here:
https://github.com/1000001101000/Debian_on_Buffalo/tree/master/Buster/installer_images/armhf_devices

1000001101000, 2021-01-05:
if you just copied the links on that page you would have ended up with the HTML rather than the real files. If you clicked through to each and then copied the "download" link you would get the correct ones. I think Github makes that more confusing than it has to be.

If you have a linux pc and sata->usb adapter handy you can download the files and put them in place manually (that process is covered in the instructions along with the stock fw method).

If it's easier you can also restore teh stock firmware first then try again:
https://buffalonas.miraheze.org/wiki/Restoring_Stock_Firmware_without_TFTP
https://buffalonas.miraheze.org/wiki/Restoring_Stock_Firmware_via_TFTP

hakimh02, 2021-01-06:
That's what I did. Clicked on each and downloaded the files (13MB and 4MB respectively). 

I think I'll plug the HDD in a PC and move the files manually. Let's see. I'll report once done.

Thank you, really appreciate it!

hakimh02, 2021-01-06:
So, seems I can establish a connection to the NAS locally. Pinging it with a fixed ip works. 
I've deactivated all firewalls (on Windows and a Mac). But whenever I send the command to exchange the files via acp_commander, the following is replied (on both Windows and Mac):

`A SocketTimeoutException usually indicates bad firewall settings.
Check especially for *UDP* port 22936 and make sure that the connection to your LS is working.
ERROR: Exception: SocketTimeoutException (Receive timed out) [ACP Send/Receive (Packet:80A0 = Special_CMD_EN_ONECMD)]`

1000001101000, 2021-01-06:
Did you load the boot files for the Buffalo firmware or for the debian installer?

hakimh02, 2021-01-06:
I've tried to move both firmware files.

1000001101000, 2021-01-06:
If you're trying to install the stock firmware acp commander or LSupdater should see the device. If not it's possible you're network doesn't forward udp broadcast packets (mesh networks tend not to). 

If you're trying to get the debian installer working you just need to look for port 22 to open and then connect via ssh. 

If the light is turning solid something is booting successfully. 

If it's just flashing constantly (not red) then it is successfully loading some boot files but failing to boot. Loading for the wrong model could do that, but it sounds like you;re already checking for that.

hakimh02, 2021-01-07:
quick update. I've managed to plug the HDD into a Linux machine and moved the files manually. The NAS now boots so that I could install Debian and Openmediavault into it. 
Are there any recommended settings for the FAN? I've run gpio_fan and set it up, but it's still running on 100%. 
Also, do you recommend updating the kernel or not if it's running?

hakimh02, 2021-01-07:
Also, for some weird reason, when plugging in the second drive, the NAS won't boot. After removing it, booting works just fine. 
And when plugging it in, it's being recognized as a ext4 drive etc. 
I guess it didn't create a RAID array

1000001101000, 2021-01-07:
Great!

I’m behind on testing newer kernels on these devices so i’m not sure if you’ll run into issues or not. I am aware of one issue that happens if your device has NAND which might affect your device.

Personally I tend to stick with the distros version.

....that does remind me it’s probably time to start working on Bullseye...

I believe I has a section in post-installation options about setting up fancontrol to set the fan speed. I usually leave the fan on full for smaller devices but it depends on what i’m doing.

1000001101000, 2021-01-07:
someone else ran into the same thing recently:
https://github.com/1000001101000/Debian_on_Buffalo/issues/63

At boot the bootloader scans partition 1 on each disk and decides which one it want to boot from. In some configurations that scan can fail if it finds a partition 1 that is too large for it to read or otherwise decides to pick the invalid one over the valid one. 

I typically recommend setting up /boot as a raid1 on all the partition 1's (the same as the stock firmware) which avoids this issue. However, we did determine that if you rename the partition to partition 2 the bootloader will just ignore it and boot from the other disk.

hakimh02, 2021-01-07:
oh great, I'll try that out!

But there's no way for me to use the partition on the primary drive and the second HDD as RAID1, is there?

1000001101000, 2021-01-07:
I don't understand the question

hakimh02, 2021-01-07:
so, is there a way for me to use both drives as raid1? meaning after managing to boot normally while having both drives plugged in?

1000001101000, 2021-01-07:
definitely.

The easiest way might be to do a fresh install and set it up that way from the start but all you have to do is create raid1 arrays for /boot swap and /. The /boot array needs to be on all the partition 1's, can't be to big (not sure the actual limit on this model but 1GB is usually plenty of space for /boot) and needs to be formatted ext3.

The stock firmware uses this same layout, that's partly why I've defaulted to having folks re-use the stock layout because it does some of that work for them.

I've also created an alternate method that generates a disk image which already has the layout configured. you can then write it to a disk, boot from it, then add a blank drive and follow the instructions to extend the arrays to the new drive:
https://github.com/1000001101000/Debian_on_Buffalo/wiki/Alternate-install-method-via-debootstrap-script

hakimh02, 2021-01-08:
I'm sorry, but this is really frustrating. I've tried EVERYTHING really. Even tried to go back to stock, while everything worked I didn't suceed in getting LSUpdater to load the firmware to the device (via Debug). It recognized it but when clicking on Update it always gave the the error that there's no Partition ot be found etc. 

Anyhow, I've now restarted the whole process and I'm reinstalling Debian as we speak. My questions: 
I have both Disks installed in the Tray: 
-Disk 1: 4 Partitions as advised (1 boot, 1 root, 1 swap, the rest is files)
-Disk 2: 1 Partition that I want to use as Raid1 of the last partition of Disk1

In the Debian Installer, I've mounted each partition to where it should and created a Software RAID1 with last Partition of Disk1 and Disk2. 

Will this work? 
If not, how can I create a RAID with both disks so it works. 

Thank you!

1000001101000, 2021-01-08:
A while back I started working on a step by step guide to manually creating a raid setup from within the installer. I haven't had a chance to finish that work but I did upload screenshots walking through the whole process, they are here:
https://github.com/1000001101000/Debian_on_Buffalo/tree/master/.images

These two show what the partitions and raid devices look like:
https://github.com/1000001101000/Debian_on_Buffalo/blob/master/.images/5-all_partition.png
https://github.com/1000001101000/Debian_on_Buffalo/blob/master/.images/17-all_volumes.png

hakimh02, 2021-01-08:
Thanks, that's exactly what I'm looking for.
I was able to assign one partition to another but wasn't able to define the
mount for each. How did you manage to do that ?

1000001101000, 2021-01-08:
you create the raid arrays and then enter each in the menu and define the filesystem and mount point.

If you look through all the screenshots there's one for each.

hakimh02, 2021-01-09:
Thanks again, really appreciate the support. 

I'll go ahead with the last attempt tomorrow. 

The pictures help alot, but would you recommend creating the necessary partitions on both disks via gparted first then start the installer und create the raid arrays etc in the menu or do everything within the installer.

1000001101000, 2021-01-09:
Creating partitions in the debian installer is in some ways easier but is more time consuming and annoying.

you can certainly use parted if thats easier

hakimh02, 2021-01-11:
Thank you for your help and support! Worked out just fine. The screens helped a lot! 

Vielen lieben Dank!

closed 2021-01-11 by hakimh02