1000001101000/Debian_on_Buffalo #76: USB Boot + rootfs on compatible models

lis6502, 2021-01-11:
Hello. 
Thanks to you i successfully unchained my TS3400d out of OFW's limitations. 
However, i do use 4x HDD setup already as a NAS in my server and they're partitioned already. I planned to use Buffalo as substitue for server's NAS role, it is to pull disks out of HP Z800, plug them into TS3400 and happily serve data over NFS and iSCSI as before. 

tl;dr; it's possible to boot device out of pendrive (plugged into USB2 socket). Prerequisites: initrd.buffalo and uImage.buffalo present on first partition of USB stick (formatted as vfat). Flick rocker switch from HDD to USB position and start your junk. boom, 

```Model: WDC WD10JPVX-XXXXXX                  Firm: 01.01A01 Ser#:         XXXXXXXXXXXXX
            Type: Hard Disk
            Supports 48-bit addressing
            Capacity: 953869.7 MB = 931.5 GB (1953525168 x 512)

usb boot.
Net:   egiga0 [PRIME], egiga1
Hit any key to stop autoboot:  3  2  1  0 
(Re)start USB...
USB:   Active port:	1
Register 10011 NbrPorts 1
USB EHCI 1.00
scanning bus for devices... 2 USB Device(s) found
       scanning bus for storage devices... 1 Storage Device(s) found
reading /uImage.buffalo

5023173 bytes read
reading /initrd.buffalo

6041921 bytes read
<<stop_sound>>
## Booting kernel from Legacy Image at 01200000 ...
   Image Name:   kernel 5.10.0-1-armmp-lpae
   Created:      2021-01-09  22:26:46 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)```

i just wanted to singal that's possible. My next test will be booting from USB3 plugged pendrive/ actual rootdisk.

1000001101000, 2021-01-11:
Good to hear from you!

Thanks for giving that a try! 

I've been meaning to look into doing that. I actually forgot that it has a switch for that. Some other models appear to have the capability too but I'm not sure how to enable it (without messing with the uboot environment).

Did you run into any issues using 5.10 on this device? I ran into some issues on the LS400 which cause the MAC address to be set wrong in some situations with ~5.1 but haven't gotten around to doing much testing on modern kernels yet..... I should probably start working on Bullseye support pretty soon.

It's good to know someone working with a TS3000, It's the most powerful of the device's in the project but also the one I hear the least about.

lis6502, 2021-01-11:
Additionally: i successfully booted with one pendrive acting as vfat /boot and rootdisk hooked via USB3 enclosure
https://pastebin.com/cLwxe5AD <- (almost) full bootlog grabbed over front UART. Happy AF :D
tl;dr, adding this
```usbcore
ehci_orion
usb_storage
ehci_hcd
xhci_pci
uas
xhci_hcd
```
into `/etc/initramfs-tools/modules` < here (and triggering initrd rebuild with something alike ```dpkg-reconfigure $(dpkg -l |awk -v VAR=`uname -r` '$2 ~ VAR {print $2}')```) and copying resulting files into vfat usbstick plugged into USB2 socket, enables booting off USB3 connected rootdisk.

> Some other models appear to have the capability too but I'm not sure how to enable it (without messing with the uboot environment).
Probably it depends on micon/ other "platform controller" rather than uboot. I don't know why, but i am able only to read from console, but no keystrokes "reach" Buffalo. i am pretty sure that i've attached UART properly (according to PCBs imprintings marking which goldpin is for what), therefore i can't say much about uboot configuration and it's features. Maybye it's my dongle. 

> Did you run into any issues using 5.10 on this device?
Absolutely none. I've bootstrapped vanilla Buster using your script, baked resulting image onto HDD and proceeded as with regular desktop Debian - enable testing and do `dist-upgrade` ;). 
my `sources.list`
```deb http://ftp.cz.debian.org/debian/ buster main contrib non-free
deb-src http://ftp.cz.debian.org/debian/ buster main contrib non-free

deb http://ftp.cz.debian.org/debian/ bullseye main contrib non-free
deb-src http://ftp.cz.debian.org/debian/ bullseye main contrib non-free
```

One thing that i can complain at is mdcrypt throwing stacktraces in kernel log from time to time and `cryptsetup benchmark` hanging so hard that even `kill -9` does not remove it from process list, however i suppose it's rather support of crypt* engines in particular kernel on particular platform.

https://pastebin.com/pUNjgpZQ <- crypt*'s stacktraces. Unfortunately, i don't know how to take advantage of there traces, maybye there'll be someone more familiar with kmod development.

> on the LS400 which cause the MAC address to be set wrong 
Given that i am booting into Debian, i always receive same IP address (and my home Mikrotik does bind between MAC and IP while DHCPing); i tried to boot into recovery (to obtain know-to-be-working kernel and initrd straight outta mtd) and this mode actually identifies itself with totally different MAC, as my IP have changed.

> It's the most powerful of the device's in the project 
Maybye i am lucky, but i bought it cheaper than 2013's QNAPs and Synologies. Of course in "broken/ not tested" condition, withouth drive caddies.

This way or another, i hope that my input will useful for project - i know this and that about daily linux usage, so feel free to ask more or less questions ;)

To summarize: tonight will be first night when i'll let it be running till morning. We'll see whether it crashes. Tomorrow i plan to populate internal SATAs with my precious WD Reds (2x2TB + 2x4TB) and do some stress testings :)

1000001101000, 2021-06-29:
Is there anything else you want me to look at for this issue?

closed 2023-06-07 by 1000001101000