1000001101000/Debian_on_Buffalo #77: Stuck at partitioning

poadb, 2021-01-12:
Hi, as requested please find a attached screenshots of the installer disk partitioning stage.

[image]
[image]
[image]
[image]
[image]

1000001101000, 2021-01-12:
I think I've seen that before. Go through and make each of the #5 partitions as "do not use" so it doesn't try to format them as swap (it should only do so for the raid device.

poadb, 2021-01-12:
[image]

poadb, 2021-01-12:
[image]

1000001101000, 2021-01-12:
I believe I've seen that when the rootfs volume runs out of space, but it looks like you've got plenty of space. I may have also seen it when someone tried the install without reformatting the rootfs. Make sure to select format for the / and /boot filesystems.

You should be able to find the actual error and get more detail if you connect to a second session and take a look at the end of /var/log/syslog from a shell.

lis6502, 2021-01-12:
Try to recreate these steps on x86 debian installation within VM - it might be related to installer itself, i've seen similar errors trying to PXE debian onto one machine. 
On the other hand, try on NAS to partition disks using regular partitions, not MD devices.

1000001101000, 2021-01-12:
I suspect the issue is related to the history of previous attempts rather than an installer bug, if it is an installer bug it's probably something fairly specific to the device/architecture and wouldn't be easily reproduced in an x86 VM.

If the problem isn't fixed by what I mentioned above I expect it to be pretty obvious when we look at the syslog output.

If a package failed to download for some reason (momentary network issue/etc) it could also fail at that point (you'd see it in the log). It's possible it work on the next try even without changes.

If you keep having trouble I can give it a try on my TS-XL, but I'm guessing we'll be able to resolve the issue and move forward before it gets to that point.

DavidDah, 2021-01-12:
You might want to check my guide about installing Debian on TS-HDTGL, I have run into similar problems with partitioning
https://github.com/DavidDah/Debian_on_Buffalo/wiki/How-to-install-Debian-on-TeraStation-PRO-II-TS-HTGL-R5

1000001101000, 2021-01-12:
@DavidDah That's awesome, I should see if I can integrate that into the main instructions at some point.

We should probably point out that the main limitation for the TS-HTGL is that the default partitions are too small for a Debian install. Partitions in @poadb 's screenshots are more than large enough.

DavidDah, 2021-01-12:
@1000001101000  please feel free to integrated it to main instructions, that would be awesome.
I will try to improve it little bit along in the meantime.
@poadb can you post content of /var/log/syslog ?

1000001101000, 2021-01-12:
@poadb I do have one other suggestion regarding partitioning.

I'd suggest against using the RAID0 array as /home. It'll be easier to recover when one of the disks eventually fails if you move it to something like "/mnt/array" (something user logons won't depend on). I'd also add the "optional" option to the array in your fstab so that the device will boot successfully without it if it fails (I think I have some info about that on the wiki).

poadb, 2021-01-13:
How do I get the contents of syslog out of putty? I can't seem to select all and the copy all to clipboard only captures what's on the screen.

I'll change the RAID0 array now so that it's not /home. What do you mean by the rest of your instructions.."something like /mnt/array" - I saw this in the youtube video, but I'm sure he did this on the single disk he had in his linkstation?

poadb, 2021-01-13:
Jan 13 09:31:20 main-menu[15046]: INFO: Menu item 'bootstrap-base' selected
Jan 13 09:31:20 base-installer: warning: Unknown armel subarchitecture 'generic'.
Jan 13 09:31:21 base-installer: warning: attempting to install to unclean target
Jan 13 09:31:22 debootstrap: /usr/sbin/debootstrap --components=main --debian-installer --resolve-deps --exclude=pciutils --keyring=/usr/share/keyrings/archive.gpg buster /target http://http.us.debian.org/debian
Jan 13 09:31:23 debootstrap: mkdir: can't create directory '/target/var/lib/apt/': No such file or directory
Jan 13 09:31:26 base-installer: error: exiting on error base-installer/debootstrap-failed
Jan 13 09:31:27 main-menu[15046]: WARNING **: Configuring 'bootstrap-base' failed with error code 1
Jan 13 09:31:27 main-menu[15046]: WARNING **: Menu item 'bootstrap-base' failed.
Jan 13 09:31:29 main-menu[15046]: INFO: Modifying debconf priority limit from 'medium' to 'low'
Jan 13 09:31:29 debconf: Setting debconf/priority to low
Jan 13 09:31:29 main-menu[15046]: DEBUG: resolver (ext2-modules): package doesn't exist (ignored)
Jan 13 09:31:29 main-menu[15046]: DEBUG: resolver (fat-modules): package doesn't exist (ignored)

1000001101000, 2021-01-13:
> Jan 13 09:31:21 base-installer: warning: attempting to install to unclean target
I think this warning indicates that the volume wasn't reformatted which caused the failure.

Look at the rootfs and /boot volumes again, there should be a format option which defaults to something like "no, keep data". Change it so that it reformat the volume (on both). I believe this will resolve the issue.

The post installation changes i was talking about earlier can be found here:
https://github.com/1000001101000/Debian_on_Buffalo/wiki/Post-Installation-Options

poadb, 2021-01-13:
Post-Installation Options are irrelevant at this point, right? Since I've not done an installation?

So if we take the disk partitioning one step at a time, can you help with what each should be, please? Here's the first screenshot of RAID1:

[image]
[image]

1000001101000, 2021-01-13:
correct, post-installation is irrelevant for now.

The first RAID device needs to be /boot, it also needs to be ext3.

The second one should be mounted as "/", it can technically be any filesystem you want, I recommend ext4.

poadb, 2021-01-13:
Right - I forgot that bit.

I've gone ahead and set RAID1 #0 as /boot, #1 as / (root) , each set to ext4 and formatted. And then set each SWAP (#5) of each disk to "do not use".

It seems to be installing now?

[image]

1000001101000, 2021-01-13:
Looks good!

poadb, 2021-01-13:
Classic "Problem In Chair", but I was genuinely stuck. So the only difference I had to make was remove swap from each disk? And remember to format the ext3 root and boot of course.

Will see how I get on with the rest of the guide 👍 
Thanks for your help.

1000001101000, 2021-01-13:
I think that is all that was required.

I believe I have seen the swap issue before but I don't remember the situation. I don't think it happens all the time, otherwise I would probably get more questions about it. 

....actually I wonder the change I made a while back to (unsuccessfully) add support for devices with <128MB ran to the installer is causing it to do this. 

When I get some time I'll restore the stock firmware on mine and see if I can do anything to prevent it from happening in the future.

poadb, 2021-01-14:
OK - I've got WebMin installed to give me a GUI.
I'm not stuck at post installation options:

`6. Presumably you'll already have used some of these arrays as your /boot,/ and swap during install and don't need to configure them further, if you have a data array/disk create a mountpoint for it similar to:

   mkdir /mnt/array
`

How would I know if I have a data array/disk? And if I don't (and I presume i don't because of the installation process I did thus far) how do I create one?

Here's the earlier steps results:

`root@debian:/# mdadm --detail --scan
ARRAY /dev/md0 metadata=0.90 UUID=241328c6:d3892594:5b99ebed:45004f3e
ARRAY /dev/md/1 metadata=1.2 name=TS-XL-EMCE5:1 UUID=26b12580:dad660b7:d106c6a3:3dde3433
ARRAY /dev/md/10 metadata=1.2 name=TS-XL-EMCE5:10 UUID=70817612:43a85e63:cb7656b9:ff9bc52b
ARRAY /dev/md/2 metadata=1.2 name=TS-XL-EMCE5:2 UUID=1118f9ad:010bd659:c3fa5004:c7f410d8
root@debian:/# nano /etc/mdadm/mdadm.conf
root@debian:/# ls /sys/block
md0  md1  md10  md2  sda  sdb  sdc  sdd
`

And screen shot of WebMin

[image]

1000001101000, 2021-01-14:
looks like the array exists already. The next step is to add it to the fstab (you may have already done this with the installer) so that it mounts automatically.

take a look at the output of `lsblk` which will show all your block devices and associated mount points to see if it's mounted already.

The other thing I recommended was adding the "optional" option to the array in fstab, this tells the system it is okay to keep booting even if the array fails. Someday one of your disks will die causing the RAID0 to fail, it will be much easier to recover if the device still boots (there's no easy way to hook up a keyboard/mouse/monitor to fix it). 

also, remember to generate a new initrd if you make any changes to mdadm.conf or fstab:
`update-initramfs -u`

poadb, 2021-01-14:
root@debian:/# lsblk
NAME     MAJ:MIN RM   SIZE RO TYPE  MOUNTPOINT
sda        8:0    0 931.5G  0 disk
├─sda1     8:1    0   977M  0 part
│ └─md0    9:0    0   977M  0 raid1 /boot
├─sda2     8:2    0   4.8G  0 part
│ └─md1    9:1    0   4.8G  0 raid1 /
├─sda3     8:3    0     1M  0 part
├─sda4     8:4    0     1M  0 part
├─sda5     8:5    0   977M  0 part
│ └─md10   9:10   0   977M  0 raid1 [SWAP]
└─sda6     8:6    0 917.2G  0 part
  └─md2    9:2    0   3.6T  0 raid0
sdb        8:16   0 931.5G  0 disk
├─sdb1     8:17   0   977M  0 part
│ └─md0    9:0    0   977M  0 raid1 /boot
├─sdb2     8:18   0   4.8G  0 part
│ └─md1    9:1    0   4.8G  0 raid1 /
├─sdb3     8:19   0     1M  0 part
├─sdb4     8:20   0     1M  0 part
├─sdb5     8:21   0   977M  0 part
│ └─md10   9:10   0   977M  0 raid1 [SWAP]
└─sdb6     8:22   0 917.2G  0 part
  └─md2    9:2    0   3.6T  0 raid0
sdc        8:32   0 931.5G  0 disk
├─sdc1     8:33   0   977M  0 part
│ └─md0    9:0    0   977M  0 raid1 /boot
├─sdc2     8:34   0   4.8G  0 part
│ └─md1    9:1    0   4.8G  0 raid1 /
├─sdc3     8:35   0     1M  0 part
├─sdc4     8:36   0     1M  0 part
├─sdc5     8:37   0   977M  0 part
│ └─md10   9:10   0   977M  0 raid1 [SWAP]
└─sdc6     8:38   0 917.2G  0 part
  └─md2    9:2    0   3.6T  0 raid0
sdd        8:48   0 931.5G  0 disk
├─sdd1     8:49   0   977M  0 part
│ └─md0    9:0    0   977M  0 raid1 /boot
├─sdd2     8:50   0   4.8G  0 part
│ └─md1    9:1    0   4.8G  0 raid1 /
├─sdd3     8:51   0     1M  0 part
├─sdd4     8:52   0     1M  0 part
├─sdd5     8:53   0   977M  0 part
│ └─md10   9:10   0   977M  0 raid1 [SWAP]
└─sdd6     8:54   0 917.2G  0 part
  └─md2    9:2    0   3.6T  0 raid0

Does this look right?

1000001101000, 2021-01-14:
looks like the array exists but isn't mounted.

This is pretty much what I'm suggesting you do to make it available:
https://github.com/1000001101000/Debian_on_Buffalo/wiki/Post-Installation-Options#Setup_your_existing_raid_arrays_for_use_on_the_new_system

poadb, 2021-01-14:
And that's where I am stuck again. Number 6.

6. Presumably you'll already have used some of these arrays as your /boot,/ and swap during install and don't need to configure them further, if you have a data array/disk create a mountpoint for it similar to:

   mkdir /mnt/array

So, I think the presumption is correct? Would you agree? So I don't need to configure them?
But I don;t know if I have a data array or whether i need to create a mount point. If I do need to create a mount point - is running the cmd "mkdir /mnt/array" all that is neccessary?

Sorry for the hand feeding but the guide is higher level than my understanding of debian and array's.

1000001101000, 2021-01-14:
Don't feel bad, software raid is confusing.

You are correct, your data array is the only one you need to work on, the installer handled the others for you.

yours is /dev/md2

poadb, 2021-01-15:
Thanks for your patience :)

Next at number 9 - syntax error:

root@debian:/# mount | grep /mnt/array
/dev/md2 on /mnt/array type ext3 (rw,relatime,stripe=512,data=ordered)
root@debian:/# /dev/md2 on /mnt/array type xfs (rw,relatime)
bash: syntax error near unexpected token `('
root@debian:/#

1000001101000, 2021-01-15:
that second line was an example of output, not another command to execute.

looks like you've got md2 mounted on /mnt/array as ext3 already. If you specified all that inside the installer you might be all set already.

can you post your /etc/fstab?

poadb, 2021-01-15:
I'm sorry, that wasn't clear in the guide, my bad.

# /etc/fstab: static file system information.
#
# Use 'blkid' to print the universally unique identifier for a
# device; this may be used with UUID= as a more robust way to name devices
# that works even if disks are added and removed. See fstab(5).
#
# <file system> <mount point>   <type>  <options>       <dump>  <pass>
# / was on /dev/md1 during installation
UUID=43b09dab-5166-4d61-9d8e-08e70510725f /               ext3    errors=remoun$
# /boot was on /dev/md0 during installation
UUID=e4f2c7af-be0f-45ac-abc7-6bb88f22bd0f /boot           ext3    defaults     $
# swap was on /dev/md10 during installation
UUID=ce404ea6-f248-4158-9960-d0b9cde04c8e none            swap    sw           $

1000001101000, 2021-01-15:
alright, you must have formatted it in the installer on your first try but didn't add it in the final try so it isn't in the fstab yet.

you should be able to more or less follow the rest of my instructions to add it but be careful. If your fstab is incorrect your device won't boot. I believe my instructions describe how to test it but be sure it's right and generate a new initrd before rebooting.

poadb, 2021-01-15:
Thanks, but I've no confidence I'll know if it's right.

I only did what I said I did in the above posts. i.e make /boot and / and format them and then change all the swap drives in the scsci drives to do not use.

I'll try following the rest of the guide. Are there any other steps that state a "result" of a command that I may mistake as a command you want me to submit?

Thanks again for your support.

poadb, 2021-01-15:
And the result:

root@debian:/# nano /etc/fstab
root@debian:/# mount /mnt/array
mount: /mnt/array: wrong fs type, bad option, bad superblock on /dev/md2, missing codepage or helper program, or other error.
root@debian:/#

contents of the fstab after I followed your guide:

`# /etc/fstab: static file system information.
#
# Use 'blkid' to print the universally unique identifier for a
# device; this may be used with UUID= as a more robust way to name devices
# that works even if disks are added and removed. See fstab(5).
#
# <file system> <mount point>   <type>  <options>       <dump>  <pass>
# / was on /dev/md1 during installation
UUID=43b09dab-5166-4d61-9d8e-08e70510725f /               ext3    errors=remoun$
# /boot was on /dev/md0 during installation
UUID=e4f2c7af-be0f-45ac-abc7-6bb88f22bd0f /boot           ext3    defaults     $
# swap was on /dev/md10 during installation
UUID=ce404ea6-f248-4158-9960-d0b9cde04c8e none            swap    sw           $

/dev/md2        /mnt/array           xfs    defaults        0       2
`

poadb, 2021-01-15:
being an idiot. I have to now change everything from xfs to ext3, as that's the fs mine is formatted too.. right?

1000001101000, 2021-01-15:
can you post your final fstab?

unmount and then mount /mnt/array to ensure you have everything right before proceeding.

poadb, 2021-01-15:
```
# /etc/fstab: static file system information.
#
# Use 'blkid' to print the universally unique identifier for a
# device; this may be used with UUID= as a more robust way to name devices
# that works even if disks are added and removed. See fstab(5).
#
# <file system> <mount point>   <type>  <options>       <dump>  <pass>
# / was on /dev/md1 during installation
UUID=43b09dab-5166-4d61-9d8e-08e70510725f /               ext3    errors=remoun$
# /boot was on /dev/md0 during installation
UUID=e4f2c7af-be0f-45ac-abc7-6bb88f22bd0f /boot           ext3    defaults     $
# swap was on /dev/md10 during installation
UUID=ce404ea6-f248-4158-9960-d0b9cde04c8e none            swap    sw           $

/dev/md2        /mnt/array           ext3    defaults        0       2

```

I have samba setup on /mnt/array/home now which is accessible from my windows machine with authentication working.

I've not been brave enough to reboot the NAS yet.....

Many thanks for your help.

1000001101000, 2021-01-15:
The way to be confident with an fstab change is to do the following:

1. verify the fstab entry by unmounting and then mounting the volume (which uses the fstab)
2. verify it's actually mounted and working via df/cp/etc
3. update the initramfs to ensure the boot environment gets the changes

updating the initramfs will automatically ensure that the proper modules/drivers get loaded and will ensure any volumes needed for booting are configured properly. This is why you want to update it anytime mdadm.conf or fstab change. 

if you do all of the above and don't get errors you should be ready to proceed

poadb, 2021-01-15:
Done - there were no errors and samba stopped and started before and after those steps too.

I will now try to restart the NAS ........

poadb, 2021-01-15:
It's working :) thank you!

Hopefully my idiocy helps others, or maybe helps with the instructions? You can see where layman like myself falls over :) 

But thank you for this project - hopefully get some more years out of this old NAS yet!

1000001101000, 2021-01-15:
Nice!

1000001101000, 2021-03-04:
I'm going to close out this issue, let me know if you need anything else!

closed 2021-03-04 by 1000001101000