1000001101000/Debian_on_Buffalo #79: Kernel of Buffalo Firmware in Debian Buster ?

VR-Punk, 2021-01-13:
hey,
how can i install the Kernel of Buffalo Firmware in Debian Buster ?
The buffalo kernel has better performance on my system !

VR-Punk, 2021-01-13:
A few years ago i installed debian jessy with the buffalo kernel, but the boards with the "how to" doesn't exist anymore.

1000001101000, 2021-01-13:
Good to hear from you!

I haven't tried running Debian with the stock kernel in a long time, it's not something I plan to spend much time researching/supporting. 

I am curious about performance differences between that kernel and a Debian kernel though. I'd be curious to see what benchmarks you see it under and how big the difference is. It would be interesting to see if recompiling a Debian kernel without specktr/meldown fixes gives similar gains without needing to use an ancient kernel.

I would expect to see compatibility issues with the stock kernel on any version newer than Jessie. 

The one time I did it I used this process:
https://github.com/rogers0/OpenLinkstation
It would probably take some work to make it work now.

You could also generate a Jessie disk image using my deboostrap script (with a few tweaks):
https://github.com/1000001101000/Debian_on_Buffalo/wiki/Alternate-install-method-via-debootstrap-script

You can probably find an old Jessie roofs elsewhere (I briefly used a stretch image meant for a raspberrypi before I got the installer working).

Whatever you go with you should be able to use the uImage.buffalo from the stock firmware with a custom initrd.buffalo. The one from the OpenLinkstation method worked for me back when I did that. I assume one generated by initramfs-tools would work too but I've never generated one with a manually installed kernel, I don't know if there are any complications. You would also need to install the modules, that should be as easy as copying the module directory from /lib/modules/

1000001101000, 2021-01-13:
lol, I don't really have any of those things.

What model device are you using? 
Are these transfers over samba or something else?

VR-Punk, 2021-01-13:
Samba on a LS-QVL !

VR-Punk, 2021-01-13:
any very big file ? it's limited to 40mb/s on debian samba, with buffalo it copy's at 70-100mb/s

1000001101000, 2021-01-13:
A few months ago, before I got distracted by other projects, I was working on the LS-QVL device tree to include some elements that are missing from the mainline version. I don't think it included anything that would help with performance but would fix some hotswap issues. I'll try to get back to that work in the near future.

While I'm at it I can take a look at samba performance when I do. I wonder if there are any test suites that make that process less painful.

VR-Punk, 2021-01-13:
NFS from LS-QVL to Ubuntu was 5mb/s faster. If transmission and a movie stream runs in the backround, than samba copys at 20mb/s. The Buffalo kernel is much faster and copys always above 50mb/s. The LS-QVL is too weak for 2021, but original drivers in the buffolo kernel are better optimized for samba than in debian buster.

1000001101000, 2021-01-13:
Did you see these speeds in the Buffalo kernel on Jessie or in the stock firmware?

If it's just the kernel that narrows down some of the possibilities.

VR-Punk, 2021-01-13:
On Jessy with the Buffalo Kernel and on stock firmware, too !