1000001101000/Debian_on_Buffalo #82: LS210D nothing after reboot

skamphax0r, 2021-02-11:
i was able to finish up the install, but after reboot it seems like network never comes up. 
i have one white light on the front - just not responding to ping 

[image]

/target/etc # cat fstab
# /etc/fstab: static file system information.
#
# Use 'blkid' to print the universally unique identifier for a
# device; this may be used with UUID= as a more robust way to name devices
# that works even if disks are added and removed. See fstab(5).
#
# <file system> <mount point>   <type>  <options>       <dump>  <pass>
# / was on /dev/sda2 during installation
UUID=aaee5f9a-5017-4964-a142-cf6bc5d8de1d /               ext4    errors=remount-ro 0       1
# /boot was on /dev/sda1 during installation
UUID=0d72437f-d7c2-4b57-a050-e76104923f0c /boot           ext3    defaults        0       2
# /mnt/disk was on /dev/sda4 during installation
UUID=870e9030-9a58-4818-8dfa-fb3307ee234f /mnt/disk       ext4    defaults        0       2
# swap was on /dev/sda3 during installation
UUID=e364c21d-142a-47a5-8afc-baa16f319c25 none            swap    sw              0       0
/target/etc # 

/target/boot # ls
System.map-4.19.0-14-armmp  dtbs                        initrd.img.old              vmlinuz-4.19.0-14-armmp
config-4.19.0-14-armmp      initrd.buffalo              lost+found                  vmlinuz.old
dtb                         initrd.img                  uImage.buffalo
dtb-4.19.0-14-armmp         initrd.img-4.19.0-14-armmp  vmlinuz

1000001101000, 2021-02-11:
Good to hear from you!

Your partitions look fine, so does the contents of your /boot. I don't see an issue there.The fact that you're getting to the solid LED means it's getting most of the way through the boot process. 

The mostly explanation I can think of is that it's getting a different IP address when after it reboots. I've noticed that buster uses a different format for DHCP client names than older versions (and the stock firmware), depending on your router this can cause a mis-match which results in getting a different IP address. Another possibility is that the MAC address isn't getting set properly at boot (the way that works on these devices is weird) that can result in a random mac address which can also result in getting a different DHCP address? 

Are you able to scan your network for ssh servers (or look at DHCP leases in your router if possible) and see if you can find it that way? 

It's conceivable (though unlikely) that Debian's recent kernel/installer update broke something. I can look into that as well (I just need to find my ls200 and try it out).

skamphax0r, 2021-02-12:
@1000001101000 Thanks! i found it - apparently my router didnt want to update to show the new IP, even though it had the MAC right. 
All good now, thanks!

closed 2021-02-12 by skamphax0r