1000001101000/Debian_on_Buffalo #83: Siren after install Stretch or Booster on LS-GL

markmal, 2021-02-16:
Hi, I've installed Booster or Stretch on LS-GL. The installation itself run without any issue.
However after final reboot the device sirens with flashing red LED.
I remember I had this issue before and cause was initrd.buffalo bigger than 7MB or so.
I've re-packed initrd.img-4.19.0-14-marvell with XZ recompression.

`mkdir initrd.img-4.19.0-14-marvell.d`
`cd initrd.img-4.19.0-14-marvell.d`
`xz -dc < ../initrd.img-4.19.0-14-marvell.d | cpio -idmv`
`find . 2>/dev/null | cpio -o -c -R root:root | xz -9 --format=lzma > ../initrd.img-4.19.0-14-marvell.xz`
`mkimage -A arm -O linux -T ramdisk -C lzma -n "ramdisk 4.19.0-14-marvell" -d initrd.img-4.19.0-14-marvell.xz initrd.buffalo-4.19.0-14-marvell`
`cp initrd.buffalo-4.19.0-14-marvell initrd.buffalo`

`file initrd.buffalo-4.19.0-14-marvell
initrd.buffalo-4.19.0-14-marvell: u-boot legacy uImage, ramdisk 4.19.0-14-marvell, Linux/ARM, RAMDisk Image (lzma), 5604525 bytes, Mon Feb 15 22:57:10 2021, Load Address: 0x00000000, Entry Point: 0x00000000, Header CRC: 0xBFC0162C, Data CRC: 0xAFDA3E0D`

With the smaller initrd.buffalo (5928550 bytes) it does not siren, but keeps rebooting.
I do not have serial cable, do not know what is going on there.
Any advise?   

THANKS!

1000001101000, 2021-02-16:
Good to hear from you!

Next time I make changes to the installer (hopefully fairly soon) I plan to make xz compression the default for initrd on these devices. It makes generating initrd take a while but it avoids these types of issues in many cases. 

It's possible the device is failing to unpack the initrd because "xz -9" requires too much memory to unpack. I did some experiments last summer and found -2e gave the best compression/speed while still working on devices with as low as 64MB ram. I use "-7" when making installer images (which work on all the 128mb devices like the LS-GL), i imagine that will take a long time on the device though.

Do you know why your initramfs is so big? usually folks don't have a problem, especially on stretch.

Here are the settings changes I use to generate XZ compressed initramfs on these devices:

**/etc/initramfs-tools/conf.d/modules**
```
MODULES=dep
```

switch to xz compression instead of gzip:

**/etc/initramfs-tools/conf.d/compress**
```
"COMPRESS=xz"
"XZ_OPT=-2e"
"export XZ_OPT"
```

You could try this by making those changes in the /target filesystem at the end of the install process then generate a new initramfs before rebooting. I can go into more detail about how to do that if needed.

2boot, 2021-02-16:
Sorry for butting in. My experience - I saw a warning somewhere (don't know where) that because of the hardware limitations of the buffalo-lsgl that most recent version of Debian to be installed should be STRETCH as Buster had some items discontinued for armel. Whereas Stretch was OK for ARMEL.
Today I followed the procedure in conjunction with the the earlier part of the video mentioned on `https://github.com/1000001101000/Debian_on_Buffalo/wiki/Running-the-installer`. 
I started from the point that the drive was formatted ntfs, I then loaded buffalo software to put the lsgl to stock. Then following the procedure and video I downloaded initrd.buffalo and uImage.buffalo.lsgl from these github pages served them to the lsgl and although the complete installation took quite some time, it all ended good. I then installed samba and ufw by following other websites. The result one working Debian file server. Thank you for your effort with acp_commander etc I now have a working 2006 buffalo linkstation with 1GB drive running Debian 9 This may not help this case, but I can say the software available at todays date was good.

1000001101000, 2021-02-16:
Good to hear from you!

A few people (including me) have installed Buster on the LS-GL without too much trouble, though  it's currently pretty easy to make the initrd too big by adding drivers for lvm/etc. The XZ compression should help with that part.

You may have been reading something about Bullseye or newer kernels in general. It's getting harder (if not impossible) to build a kernel that will fit in the 2M kernel size limit of the really old versions of uboot used by most Orion devices. I believe Debian did announce they are dropping some devices because the Marvell kernel will no longer be small enough. 

I plan to look into what works and what doesn't when I build the first Bullseye installer and go from there.

markmal, 2021-02-17:
Tried following options:
1. xz lzma -2e initrd.buffalo
2. xz lzma -2e initrd.buffalo (removed /sbin/lvm)
3. gzip -9 initrd.buffalo (removed /sbin/lvm)
4. bzip2 -9 initrd.buffalo (removed /sbin/lvm)
all have initrd.buffalo <= 7MB

All keep rebooting.

/sbin/lvm had size 2.7MB. I am not sure why it was included. Probably because data is on LVM. But I've set it "not used" in Installer. To avoid inclusion LVM into initrd.
My disk is 3.2TB. it has /data in LVM

I suspect that either kernel or initrd got created wrongly. Comparing with Installer's initrd, mine has pretty empty /usr/lib/ . 
drwxr-xr-x 2 mark mark  4096 Feb 15 17:51 arm-linux-gnueabi/
-rwxr-xr-x 1 mark mark 75364 Feb  1  2019 klibc-TypMgZXqx9gUdGDfK-WB8jYZpwQ.so*
lrwxrwxrwx 1 mark mark    28 Feb 15 17:51 ld-linux.so.3 -> arm-linux-gnueabi/ld-2.28.so*
drwxr-xr-x 2 mark mark  4096 Feb 15 17:51 modprobe.d/
drwxr-xr-x 3 mark mark  4096 Feb 15 17:51 modules/
drwxr-xr-x 3 mark mark  4096 Feb 15 17:51 systemd/
drwxr-xr-x 3 mark mark  4096 Feb 15 17:51 udev/

only ld-linux.so.3 and klibc-*.so. Is it normal?

If you have working (not installer) initrd.buffalo and uImage.buffalo for 4.19.0-14 could you place it on github?

2boot, 2021-02-17:
Hello, Please do not confuse me with someone who may know what they are doing (my drive is 1TB not 1GB ;) ).
I purely followed the video sticking rigidly to Debian stretch for armel, as I need 4T whatever that is and it appears I don't have floating point.
The only interface between me and LVM was during partitioning of the drive I selected "Guided - use entire disk and set up LVM"
Concerning 4.19.0-14 I would not have a clue, however the best I can offer are the attached picture files as to what I have - the folders are selected from your above comment. Sorry I cannot help directly...
[image]
[image]
[image]

markmal, 2021-02-17:
@2boot thank you for reply, though my request for initrd.buffalo and uImage.buffalo was addressed to 1000001101000.
But any way thank you. Could you put here text output of 
`file initrd.img-4.9.0-14-marvell`
`lsinitrd initrd.img-4.9.0-14-marvell`

2boot, 2021-02-17:
Will try tomorrow, just in case it is not obvious could you write the commands for me - thanks (now 6:30pm here)

markmal, 2021-02-17:
@2boot 
these are commands:
`cd /boot/`
`file initrd.img-4.9.0-14-marvell`
`lsinitrd initrd.img-4.9.0-14-marvell`

1000001101000, 2021-02-17:
I'm planning to bring mine up on Buster/4.19 to help compare. Might be a day or two, need to reconfigure my desk somewhat.

2boot, 2021-02-18:
@markmal 
It didn't quite go to plan but I hope I have done enough to allow you to get what you need.
I didn't have access to the lsinitrd command and based upon my limited knowledge I didn't want to try adding something that may affect my working system negatively. So hopefully you have the means to extract what you need from the file.

[image]
[initrd.img-4.9.0-14-marvell.zip](https://github.com/1000001101000/Debian_on_Buffalo/files/6001201/initrd.img-4.9.0-14-marvell.zip)

markmal, 2021-02-18:
Hi 2boot - THANK YOU!

markmal, 2021-02-19:
@2boot could you attach also your uImage.buffalo ? 
I want to try to boot with your kernel and initrd.

1000001101000, 2021-02-19:
I believe the initrd will try to mount the roofs based on the UUID. Since your root partitions will have different UUID’s it will likely fail to mount the rootfs.

1000001101000, 2021-02-19:
I think your best bet in the short term is to re-run your install without telling it about your lvm setup to get a working system. Then  you should be able to add the xz logic to hopefully generate working initrds. If the xz isn’t enough you may be able to blacklist lvm from being added to the initrd (it shouldn’t be needed at that stage if your rootfs isn’t on lvm)

markmal, 2021-02-19:
Yes, I am going to put my UUID into 2boot's initrd.

Currently my root is not on LVM. It is on extended partition. Here is my fdisk -l

Device     Boot      Start        End    Sectors  Size Id Type
/dev/sdd1  *          2048     194559     192512   94M 83 Linux
/dev/sdd2           196606   32962559   32765954 15.6G  5 Extended
/dev/sdd3         32962560 4039161855 4006199296  1.9T 8e Linux LVM
/dev/sdd4       4039163568 7814037167 3774873600  1.8T 8e Linux LVM
/dev/sdd5           196608   31444991   31248384 14.9G 83 Linux
/dev/sdd6         31447040   32962559    1515520  740M 82 Linux swap / Solaris

/dev/sdd1 /boot
/dev/sdd5 /
/dev/sdd3 /dev/sdd4 - two PVs for VG vg_data 
/dev/sdd6 swap

When I installed, I've chosen "do not use" for /dev/sdd3 and /dev/sdd4
Probably I need to change partition type to something alien to linux. I suspect the installer saw LVM and included it even for partitions not in use.

It is interesting that 2boot's initrd includes LVM, it is gzipped and is still very small - 5229451. 
His root is on LVM

markmal, 2021-02-19:
I found why initrd-4.9 is smaller than 4.19. The latter has libsystemd.so that needs also 
librt.so.1 
liblzma.so.5 
liblz4.so.1
libgcrypt.so.20
libpthread.so.0
libc.so.6

2boot, 2021-02-19:
Hi, Your discussion is way over the top of my head, so I just attach the requested file, I would however like to know the eventual outcome and whether Buster could be loaded on my LSGL.
[uImage.zip](https://github.com/1000001101000/Debian_on_Buffalo/files/6008400/uImage.zip)

markmal, 2021-02-19:
Thank you 2boot.

Whole discussion is about size of initrd. For some reason uboot of LS-GL does not like if initrd is bigger than 7+MB. Probably if it is bigger it overrides some other data in memory that uboot places there.
To make it smaller we need to remove whatever is unnecessary for kernel boot process, like LVM, encryption, etc. But in this case root can not be on LVM and/or encrypted. Probably our partition layout should be like 
sda1 /boot ext2
sda2 / ext3
sda3 swap
sda4 /data - primary or extended partition(s) with LVM, depending on HDD size, encrypted or whatever. 
The /data will be mounted after kernel boot process is done.

Bottom line you should be very careful with OS upgrades, because newer versions use systemd that has larger footprint.

2boot, 2021-02-19:
Hi, Whether this helps or not you will have to judge, but since you mentioned partitions and UUID I offer this from my setup.
[info.txt](https://github.com/1000001101000/Debian_on_Buffalo/files/6011464/info.txt)
Thanks for advice.

markmal, 2021-02-20:
I've set disk uuid into initrd /conf/param.conf, coped /lib/modules/4.9.0-14-marvell to my root and boot the system from the 2boot's images of 4.9.0-14. Almost successful.
Thing is my root is 4.19.0-14, it is configured for systemd, it looks like it could not find network settings, so the system was not available on network.

markmal, 2021-02-20:
I tried to set LVM partition types as Non-FS, Empty, Hidden FAT16. Nothing helps. Partitioner recognizes them as LVM any way. It looks like it pays little attention on partition types, rather scanning and detects data itself.

1000001101000, 2021-02-20:
I finally got around to bringing my LS-GL online. Let me know if there's something you want me to try.
'''
Linux lsgl-Buster 4.19.0-14-marvell #1 Debian 4.19.171-2 (2021-01-30) armv5tel
Last login: Sat Feb 20 16:13:29 2021 from 192.168.1.146
root@lsgl-Buster:~# ls -lh /boot | grep 4.19 
-rw-r--r-- 1 root root 166K Jan 30 03:35 config-4.19.0-14-marvell
lrwxrwxrwx 1 root root   53 Feb 20 16:11 dtb -> dtbs/4.19.0-14-marvell/./orion5x-linkstation-lsgl.dtb
lrwxrwxrwx 1 root root   53 Feb 20 16:11 dtb-4.19.0-14-marvell -> dtbs/4.19.0-14-marvell/./orion5x-linkstation-lsgl.dtb
-rw-r--r-- 1 root root 4.1M Feb 20 16:11 initrd.img-4.19.0-14-marvell
-rw-r--r-- 1 root root 1.6M Jan 30 03:35 System.map-4.19.0-14-marvell
-rw-r--r-- 1 root root 2.0M Jan 30 03:35 vmlinuz-4.19.0-14-marvell
root@lsgl-Buster:~#
'''

markmal, 2021-02-20:
That's great!
could you put the files here?
initrd.img-4.19.0-14-marvell
vmlinuz-4.19.0-14-marvell

PS/ is it Buster? I thought Buster is 5.10.0-3.

1000001101000, 2021-02-20:
5.10 is bullseye or buster-backports, buster is 4.19 and stretch is 4.9.

Out of curiosity I updated to 5.10, it seems to work fine which was somewhat surprising since the kernel is now 2.4mb. My initrd is only 4.1m, it seems like these devices should be viable for a while yet. (at least for basic configs)

I attached my 4.19 and 5.10 boot files

[boot.zip](https://github.com/1000001101000/Debian_on_Buffalo/files/6016877/boot.zip)

markmal, 2021-02-21:
no luck. it still reboots. :( 
what am I doing wrong?

I've copied your *buffalo.bak to dir 4.19.0-14/

cd 4.19.0-14/
# uncompress initrd
dd if=initrd.buffalo.bak bs=64 skip=1 of=initrd.img-4.19.0-14-marvell
mkdir initrd.img-4.19.0-14-marvell.d
cd initrd.img-4.19.0-14-marvell.d
xz -dc < ../initrd.img-4.19.0-14-marvell | cpio -idmv

# put my root UUID
echo 'ROOT="/dev/disk/by-uuid/f59971e5-722b-4521-8660-c04a5157e56f"' > conf/param.conf

# wrap
find . 2>/dev/null | cpio -o --quiet -H newc -R root:root | xz -2e > ../initrd.img-4.19.0-14-marvell.xz
cd ..
mkimage -A arm -O linux -T ramdisk -C none -n "ramdisk 4.19.0-14-marvell" -d initrd.img-4.19.0-14-marvell.xz initrd.buffalo-4.19.0-14-marvell

file initrd.buffalo*
initrd.buffalo-4.19.0-14-marvell: u-boot legacy uImage, ramdisk 4.19.0-14-marvell, Linux/ARM, RAMDisk Image (Not compressed), 4103736 bytes, Sun Feb 21 14:17:56 2021, Load Address: 0x00000000, Entry Point: 0x00000000, Header CRC: 0x27540E0A, Data CRC: 0xC8E46416
initrd.buffalo.bak:               u-boot legacy uImage, ramdisk 4.19.0-14-marvell, Linux/ARM, RAMDisk Image (Not compressed), 4207096 bytes, Sat Feb 20 22:41:21 2021, Load Address: 0x00000000, Entry Point: 0x00000000, Header CRC: 0x404EE718, Data CRC: 0xAB2217EF

cp initrd.buffalo-4.19.0-14-marvell $TARGET_BOOT/initrd.buffalo 
cp uImage.buffalo.bak $TARGET_BOOT/uImage.buffalo

markmal, 2021-02-21:
TA DAAA! Finally I got there! 
I've recompressed initrd with gzip -9. Size is still small - 5703221.
And created boot/vmlinuz-4.19.0-14-marvell and boot/initrd.img-4.19.0-14-marvell so links in root can find them.
I do not know what helped. I'll try with xz compression later.

How did you managed to create this small initrd?

1000001101000, 2021-02-21:
with the settings listed here:
https://github.com/1000001101000/Debian_on_Buffalo/issues/83#issuecomment-779977362

Then just running `update-initramfs -u`

(I haven't done anything with LVM at this point)

I don't usually manually pack/unpack initrds, but you can see the code I use to create installer images here:
https://github.com/1000001101000/Debian_on_Buffalo/blob/master/Buster/installer_images/build/generate_images_armel.sh

markmal, 2021-02-21:
it does not work with the xz initrd I've created on my x86_64 ubuntu. gzip initrd works.
However when LSGL have boot from gzipped initrd, I run `update-initramfs -u`  it created xz initrd that works.
It has included LVM into this initrd. The size of this initrd.buffalo is 5721132.

I found why XZ initrd produced manually on x86_64 was not bootable.
It needs to add `--check=crc32` to xz command otherwise kernel rejects this initrd!
`xz -2e --check=crc32` this works. It looks like update-initramfs uses this key behind a scene.

OMG, finally I've sorted this out. May be I'll need to filter out LVM from update-initramfs settings. later.

Thank you!

2boot, 2021-02-22:
@markmal
Just in case I am being too simple in my requirements for my lsgl (file server based on debian-samba-ufw), could you please give a brief simple description of your desired configuration and purpose?

@1000001101000
For the record, if I wanted to upgrade my stretch to buster and I had the contents of your above boot.zip available (or the same files from elsewhere) is there a simple way to do this? or would I have to start over from the beginning of the video I followed?

I know you have both put a lot of effort into this subject and is great that you have solved it.

markmal, 2021-02-22:
@2boot, I have Linuxes at home, so my file server is NFS.

On your upgrade question. Just boot.zip is not enough and most likely your system wont start. Your root also needs to have libs and kernel modules for kernel 4.19. So, if you done major upgrade using apt-get, do not rush to reboot. Check size of **/boot/initrd.buffalo**. If it is greater than 7MB you will need 
> with the settings listed here: #83 (comment)
> Then just running update-initramfs -u

and check size again

closed 2023-06-07 by 1000001101000