1000001101000/Debian_on_Buffalo #84: Installer not running on LS-WXL

wilkgr76, 2021-02-18:
Hi,

I have a LS-WXL device. A while ago I had mounted its raid array in a desktop pc and enabled ssh in its original OS. 

I copied `uImage.buffalo.lswvl` and `initrd.buffalo` to the devices `/boot` through scp, and renamed as described in the wiki. I rebooted, but wasn't able to ssh into the installer account. The root account still worked. Just in case, I also copied over `vmlinuz` into `/boot`. 

I've also checked the md5 sum to verify that the files are correct -- and they are!

```
PS D:\> ssh installer@192.168.8.204
installer@192.168.8.204's password:
Permission denied, please try again.
installer@192.168.8.204's password: # (I pressed Ctrl-C here)
PS D:\> ssh root@192.168.8.204
root@192.168.8.204's password:
root@LS-WVL03B:~# ls /boot/
conf_save.tgz   initrd.buffalo  uImage.buffalo  vmlinuz
root@LS-WVL03B:~# md5sum /boot/initrd.buffalo
702da16415a46c00fb775c82cf53e548  /boot/initrd.buffalo
root@LS-WVL03B:~# md5sum /boot/uImage.buffalo
4f104a3eafacfea833178d845276a774  /boot/uImage.buffalo
root@LS-WVL03B:~# md5sum /boot/vmlinuz
7234f0b46cfc9a6d08252185120ab043  /boot/vmlinuz
```

wilkgr76, 2021-02-22:
I tried installing with only a single drive in the system rather than both. It works now!

closed 2021-02-22 by wilkgr76

1000001101000, 2021-02-22:
Sorry for the delayed response!

It sounds like your /boot RAID1 wasn't syncing which resulted in you loading the installer files to just one drive instead of both. Unfortunately it was then booting of the drive with the old files.

wilkgr76, 2021-02-22:
Ohh I see. Thank you for the explanation!