1000001101000/Debian_on_Buffalo #87: Not an Issue - It is a Help

iamnorbiato, 2021-02-26:
Looks who is back !!!

Hi there - my storage is working pretty fine except my disk #4 crashed few days ago.
I got a brand new disk (same model, same size)
Let me describe again to you what we did in the past:

RAID 1 for OS 
Device name | Status                  | RAID level            | Usable size | Member disk devices
/dev/md0       | clean, degraded | RAID1 (Mirrored) | 976.43 MiB | /dev/sda1, /dev/sdb1, /dev/sdc1.
/dev/md1       | clean, degraded | RAID1 (Mirrored) | 4.76 GiB       | /dev/sda2, /dev/sdb2, /dev/sdc2.
/dev/md2       | clean, degraded | RAID1 (Mirrored) | 976 MiB       | /dev/sda5, /dev/sdb5, /dev/sdc5.

My /dev/md3 should be my RAID 5
all 4 disks has 2TB each

Is it possible to rebuild the RAID 5 after one disk failure ? The storage (OS) is up and running but I don't want to "screw up" because I have backup of personal files, photos, videos - however one filesystem (with some media stuffs) I did not backed up (some personal videos playing drums...)

Can you help me out ?

Thanks.

MN

1000001101000, 2021-02-26:
Good to hear from you again!

Take a look at the output of 
`mdadm --detail /dev/md3`
and
`mdadm -E /dev/sda6`

That should let you know the status of your RAID5 array.

You should be able to add the new disk to all the arrays by:
1. copy the partition table from one of the other disks to it (or just create with the same sizes)
2. run `mdadm --add /dev/mdx /dev/sddx` as needed for each array
3. watch the output of `mdadm --detail /dev/mdx` to watch them resync

iamnorbiato, 2021-02-26:
I confess that is a relief hear back from you dude !!!
As I don't want to loose (was a lazy dumb ass taking too long to start backup the other stuffs ... anyway) some data from my raid5 - I can, but I don't want to.

Here are the outputs:

$ sudo mdadm --detail /dev/md3
/dev/md3:
        Version : 1.2
     Raid Level : raid0
  Total Devices : 3
    Persistence : Superblock is persistent

          State : inactive

           Name : HomeNAS:3
           UUID : ded6066c:c07a6b24:564084ee:f3856551
         Events : 0

    Number   Major   Minor   RaidDevice

       -       8       38        -        /dev/sdc6
       -       8       22        -        /dev/sdb6
       -       8        6        -        /dev/sda6

$ sudo mdadm --detail /dev/md3
/dev/md3:
        Version : 1.2
     Raid Level : raid0
  Total Devices : 3
    Persistence : Superblock is persistent

          State : inactive

           Name : HomeNAS:3
           UUID : ded6066c:c07a6b24:564084ee:f3856551
         Events : 0

    Number   Major   Minor   RaidDevice

       -       8       38        -        /dev/sdc6
       -       8       22        -        /dev/sdb6
       -       8        6        -        /dev/sda6
 

All the "family" data is safe (have 2 copies !!!)

again !!! one million dollar question each !
How do I excite #1 (copy partition table from other disks) ?

iamnorbiato, 2021-02-26:
oh.. crap...

here are the rest

$ sudo mdadm -E /dev/sda6
/dev/sda6:
          Magic : a92b4efc
        Version : 1.2
    Feature Map : 0x0
     Array UUID : ded6066c:c07a6b24:564084ee:f3856551
           Name : HomeNAS:3
  Creation Time : Sat Feb 23 15:49:05 2019
     Raid Level : raid0
   Raid Devices : 4

 Avail Dev Size : 3876362240 (1848.39 GiB 1984.70 GB)
    Data Offset : 262144 sectors
   Super Offset : 8 sectors
   Unused Space : before=262056 sectors, after=0 sectors
          State : clean
    Device UUID : 8ea07652:e28088ab:8434487f:49ed72d0

    Update Time : Sat Feb 23 15:49:05 2019
  Bad Block Log : 512 entries available at offset 72 sectors
       Checksum : 7022309 - correct
         Events : 0

iamnorbiato, 2021-02-26:
DONE - 1 - copy the partition table from one of the other disks to it (or just create with the same sizes)

2- run mdadm --add /dev/mdx /dev/sddx as needed for each array
should it be
mdadm --add /dev/md0 /dev/sdd1
mdadm --add /dev/md1 /dev/sdd2
mdadm --add /dev/md2 /dev/sdd5
mdadm --add /dev/md3 /dev/sdd6

Based on this ?
/dev/md0 | clean, degraded | RAID1 (Mirrored) | 976.43 MiB | /dev/sda1, /dev/sdb1, /dev/sdc1.
/dev/md1 | clean, degraded | RAID1 (Mirrored) | 4.76 GiB | /dev/sda2, /dev/sdb2, /dev/sdc2.
/dev/md2 | clean, degraded | RAID1 (Mirrored) | 976 MiB | /dev/sda5, /dev/sdb5, /dev/sdc5.

will run after step #2
3 - watch the output of mdadm --detail /dev/mdx to watch them resync

1000001101000, 2021-02-26:
> How do I excite #1 (copy partition table from other disks) ?

gdisk has a function for this and is what I typically use. You can also just open a disk with fdisk and record all the start/end sectors for each partition and use that to create the partitions on the new disk.

Your MD3 array is a problem, ccording to your output it is RAID0 not RAID5. RAID0 cannot handle any disk failure, all must be working for the array to function.

iamnorbiato, 2021-02-26:
somehow the output is wrong.. it is a raid5 

I'll submit 

mdadm --add /dev/md0 /dev/sdd1
mdadm --add /dev/md1 /dev/sdd2
mdadm --add /dev/md2 /dev/sdd5
mdadm --add /dev/md3 /dev/sdd6

and see what will happen

iamnorbiato, 2021-02-26:
I had a progress...

but md3 issue an error when I was adding the sdd6

all others has sync'd and status Ok as follow:

$ sudo mdadm -D /dev/md2
/dev/md2:
        Version : 1.2
  Creation Time : Wed Oct 31 04:01:33 2007
     Raid Level : raid1
     Array Size : 999424 (976.00 MiB 1023.41 MB)
  Used Dev Size : 999424 (976.00 MiB 1023.41 MB)
   Raid Devices : 4
  Total Devices : 4
    Persistence : Superblock is persistent

    Update Time : Fri Feb 26 17:06:32 2021
          State : clean
 Active Devices : 4
Working Devices : 4
 Failed Devices : 0
  Spare Devices : 0

           Name : LS441D-EMC92:2
           UUID : e853aa99:3839c877:8f1e6de9:5ee2c859
         Events : 220

    Number   Major   Minor   RaidDevice State
       0       8        5        0      active sync   /dev/sda5
       1       8       21        1      active sync   /dev/sdb5
       2       8       37        2      active sync   /dev/sdc5
       4       8       53        3      active sync   /dev/sdd5

iamnorbiato, 2021-02-26:
here is what I'm having...

$ sudo mdadm -D /dev/md3
[sudo] password for norbiato:
/dev/md3:
        Version : 1.2
     Raid Level : raid0
  Total Devices : 3
    Persistence : Superblock is persistent

          State : inactive

           Name : HomeNAS:3
           UUID : ded6066c:c07a6b24:564084ee:f3856551
         Events : 0

    Number   Major   Minor   RaidDevice

       -       8       38        -        /dev/sdc6
       -       8       22        -        /dev/sdb6
       -       8        6        -        /dev/sda6

norbiato@homenas:/home/norbiato
$ ^Cdo mdadm -D /dev/md3

norbiato@homenas:/home/norbiato
$ sudo mdadm --add /dev/md3 /dev/sdd6
mdadm: Cannot get array info for /dev/md3

iamnorbiato, 2021-02-26:
looks like my md3 has gone... correct ?

1000001101000, 2021-02-26:
Do you still have your failed disk?

It is sometimes possible to get a little more life out of a dying disk, maybe long enough restart the array and read some files off. There are also companies that can try to repair drive electronics/etc for $$$$.

iamnorbiato, 2021-02-26:
Well... I'll rebuild all from scratch - and create another backup plan or this time - a raid5 with 3 disks and one as spare.... will follow the standard procedure to setup all over again... :-)

iamnorbiato, 2021-02-28:
Hi there.... do you mind to help me out with the Debian image files ? looking into the Debian.org (following the prepare device for install and (im too dumb) I can't find the files - do you mind ?

1000001101000, 2021-02-28:
Which model do you have?

I think what you want will be here:
https://github.com/1000001101000/Debian_on_Buffalo/tree/master/Buster/installer_images

iamnorbiato, 2021-02-28:
YES!!!!! thanks a lot !!

BTW - are you the one in the YouTube videos explaining how to do it ?

1000001101000, 2021-02-28:
Great!

That’s not me. It was pretty neat to see it when it showed up, I had never heard of him before.

iamnorbiato, 2021-02-28:
I think you should get monetised for that !!!!!!

Let me tell you - I'm getting better and better now with install and mdadm :-)
Rebuilding from scratch for the 3rd time today...

1000001101000, 2021-02-28:
I used to think mdadm was the most confusing program ever. Once you get some practice creating arrays and adding/removing drives etc it starts to make sense. 

The youtube video drove more traffic/interest than almost anything else, I'm tempted to see If I can get him to do a similar video for the Intel-based devices since that project doesn't get nearly as much attention. To bad it doesn't translate into $$, just shipping him a Terastation to do a video on would cost far more than I've received in donations up to this point.

1000001101000, 2021-06-29:
Is there anything else you want me to look at for this issue?

iamnorbiato, 2021-06-29:
Lets close it !
My puppy is working fine since last rebuild and I'm more than happy with it !

closed 2021-06-29 by iamnorbiato