1000001101000/Debian_on_Buffalo #90: Issue with LS-QL

masterlog80, 2021-03-04:
Hello guys, 
I have just tried to install Debian on my Buffalo LS-QL with an existing RAID. 
Everything was fine until the steps after "Install the base system" which was stuck on 16% for hours. At this point I just opened another SSH shell which started from the beginning (Start installer) but still getting stuck as per attached image.
As it wasn't possible to continue I have to forcibly reboot the NAS which now, with an OS partially installed, seems not booting (RED light blinking) and doesn't get an IP Address on the Network.
Can you suggest how to get back, at least to original firmware so I will try again to reinstall Debian later?

Regards

[image]

1000001101000, 2021-03-04:
Good to hear from you!

A few times I've seen installs fail the way you described. When I looked closer in those cases it was usually because one of the packages was failing to download because of either a network issue or some issue with the mirror. It seems that the Debian Installer doesn't have a good way to deal with that scenario (though it is rare). I think to get the installer working again in that scenario you'd probably have to kill off the first installer from a second terminal and unmount the target volume etc. 

Since you ended up having to reboot after you had reformatted /boot but before new boot files were generated this left you with no boot files, which explains why it couldn't boot afterwards. You can get the device booting again either by connecting the drives to a PC and loading files back into /boot or by booting into TFTP. I have some guides for both approaches:

https://buffalonas.miraheze.org/wiki/Restoring_Stock_Firmware_without_TFTP
https://buffalonas.miraheze.org/wiki/Restoring_Stock_Firmware_via_TFTP

masterlog80, 2021-03-04:
Thanks!
I will take a look as soon as I have time or in the weekend.
I was thinking to put one HDD (or should I do the same on all the four HDDs?) on my Linux device creating basic partitions (e.g. gparted) and copy the Debian files to /boot to start from the installer directly, but I am wondering if it works... What do you think? 

Also about the URLs you provided, do you believe it's there a way to keep the existing RAID array? I have a backup but I would avoid to copy the data back.

Regards,

1000001101000, 2021-03-04:
If you have a system that can accommodate all your drives it's fairly easy to just mount the volumes in linux and move over the installer files to /boot and move forward without touching your data array (or even back it up directly while it's connected).

You might be able to restore the stock firmware over TFTP without reformatting your data array but I'm not sure if that will work. Since it will need to reformat the root partition I believe you'll need to enable formatting via a debug option and I'm not sure if it would preserve the data array or not. 

You could probably also:
- connect just one drive to a linux system
- adjust the boot,swap and rootfs arrays to be only one drive (so they don't show as degraded)  `mdadm --grow /dev/md0 -n 1 --force` 
- load the installer files to the /boot array
- boot the device with just that drive inserted and start the install like normal
- on the partition menu just leave the partition for the data array as unused
- complete the install as normal and boot into debian.
- re-insert the other 3 drives and extend your boot,swap.roofs back onto them `mdadm --grow /dev/md[012] -n 4` `mdadm --add /dev/md0 /dev/sd[bcd]1` etc
- reboot to make sure everything still works.
- run `mdadm --detail --scan` to get info about your data array, start it manually if needed
- add the details of the array to mdadm.conf, run `update-initramfs -u` and reboot to make sure it's recognized at boot
- add the array to your fstab, test it by mounting it, then run `update-initramfs -u` again

masterlog80, 2021-03-04:
Thank you for getting back.
So basically if I clean the "/boot", "/" and "swap" (existing?) partitions and copy the 2 files for debian to all the 4 drives from a Linux device, it should start from the installer?

Regards

1000001101000, 2021-03-04:
sort of?

You can either:
load the files into raid array (md0) by attaching all the drives at once or by reducing it to 1 drive like I described
or
destroy the raid on partition 1 or the first (or all) disks, reformat it as ext3 with inode=128 then load it there. (you can rebuild the array within the installer)

masterlog80, 2021-03-04:
Thanks. I will try first attaching all the drives (one by one as I don't have enough SATA cables) to my Linux device, clean the "/boot", "/" and "swap" (existing?) partitions there and copy the 2 files for debian on "/boot".
If it doesn't work, trying other methods. 

Regards

1000001101000, 2021-03-04:
You should be able to leave swap and "/" alone, /boot is the only thing required to get into the installer.

masterlog80, 2021-03-04:
Hello, 
This is the content of my first HDD mounted on Linux. 
I am unable to mount "/dev/sda1" (which should be "/boot") to copy Debian installer as "unknown filesystem type 'linux_raid_member'". 
Need to dig into this. 

[image]
[image]

1000001101000, 2021-03-04:
I put the rough steps in a list a few messages ago.

Basically what I propose is that you change the 3x 4-disk RAID1 arrays to 3x 1-disk RAID1 arrays and then you copy the installer files to the 1-disk md0 and perform your install on one disk only. When finished you can then expand all the RAID1 arrays back to all 4 disks. The RAID5 should be fine as long as you leave those partitions alone until you are back running Debian.

masterlog80, 2021-03-04:
Yeah, I am going in that way right now.

Thanks!

1000001101000, 2021-03-04:
that's normal. the devices names are specific in mdadm.conf. if you add them to another system it assigns one starting with md127

masterlog80, 2021-03-04:
I see, thank you for clear explanation!
However, I get an error running, for example "mdadm --grow /dev/md0 -n 1 --force" as there is no "/dev/md0" device.
But I will take a look later, once I have free time.

Regards

masterlog80, 2021-03-04:
Btw, I have taken a new HDD, created the partition scheme (setting inode 128 for "/boot") by Linux, copied Debian installer files on "/boot" only and put that drive only on NAS as 1 (at the top). However, once I start the NAS it still has the LED red blinking 6 times.
Need to dig into this later!

masterlog80, 2021-03-05:
I couldn't find a way to boot using a single HDD despite I created a 1GB partition formatted as ext3 and put the Debian installer files there.
Even changing size of (first) partition matching existing HDDs [1], once I start the NAS it still has the LED red blinking 6 times.
Curious.... 

[1] - https://user-images.githubusercontent.com/33828929/110036992-27633b80-7d81-11eb-9bac-6892ca97606d.jpg

masterlog80, 2021-03-05:
I was finally able to boot a new HDD with the Debian installer, and I am seeing if I encounter the same issue.
It seems it was very important to partition the HDD per the image below, selecting labe "msdos" and not "gpt", and align partitions to the cylinders.

[image]

masterlog80, 2021-03-05:
I was thinking that I can try to boot the Debian installer from the new disk.
Then, once it has loaded swap the disks with the 4 original ones and proceed with installation: as far as I know the Debian installer once loaded should stay in the memory, don't need to access to the disk anymore. Am I wrong?

Regards,

1000001101000, 2021-03-05:
The partition table must be mbr/dos for this model. Though hybrid can be made to work if needed.

your idea of booting from a different disk should work

masterlog80, 2021-03-05:
Thank you for now! I have noticed a couple of issues not related with this,
so I will create a new thread.

closed 2021-03-05 by masterlog80

masterlog80, 2021-03-06:
Just for information if someone may need, I was able to boot LS-QL from a new HDD, then once the Debian installer loaded I could swap the HDDs and recover the existing partitions and data by using the following partition scheme:
[image]