1000001101000/Debian_on_Buffalo #91: Fan issue with LS-QL

masterlog80, 2021-03-06:
Hello! I have successfully installed Debian on my Linkstation LS-QL and everything work properly except for the fan, which is definitely very noisy comparing to the stock firmware. I have checked the following URL:
https://github.com/1000001101000/Debian_on_Buffalo/wiki/Post-Installation-Options#Fans
Which is mentioning to use the `sensors` command but for me it doesn't seem to work properly, so I don't know how to slow the fan down:

```root@NAS:~# sensors
gpio_fan-isa-0000
Adapter: ISA adapter
fan1:           0 RPM  (min =    0 RPM, max = 5000 RPM)

root@NAS:~#
```

Also, it seems I am missing the `micon_fan_daemon` mentioned in the URL above:

```root@NAS:~# systemctl restart micon_fan_daemon
Failed to restart micon_fan_daemon.service: Unit micon_fan_daemon.service not found.
root@NAS:~#
```

Is there a way to fix this issue?

Regards,

1000001101000, 2021-03-06:
Hmmm, it reading as 0 RPM seems wrong.

could you run `pwmconfig` and post the output?

masterlog80, 2021-03-06:
Not installed:

```
root@NAS:~# pwmconfig
-bash: pwmconfig: command not found
root@NAS:~#
```

Regards,

masterlog80, 2021-03-06:
I have followed the procedure on screen and now:

```
root@NAS:/# sensors
gpio_fan-isa-0000
Adapter: ISA adapter
fan1:        5000 RPM  (min =    0 RPM, max = 5000 RPM)  ALARM

root@NAS:/#
```

It's showing 5000 RPM but definitely it's stopped, so it should be 0 RPM.

Regards,

1000001101000, 2021-03-06:
Sounds like I have the speed pins invert in the device tree.

I’ll push a fix for you to test shortly

masterlog80, 2021-03-06:
Yeah, it seems like you said. Btw, as now fan is stopped, is there a way to temporary set at fixed speed in the meantime?
I guess rebooting may fix, but....

Regards,

1000001101000, 2021-03-06:
[dtb.zip](https://github.com/1000001101000/Debian_on_Buffalo/files/6095414/dtb.zip)

Here's a fix:
download and unzip this DTB file
put the file in /etc/flash-kernel/dtbs/ on your device
run `flash-kernel`
reboot the device.

If your fans work properly with this in place I'll push it to the repo right away.

1000001101000, 2021-03-06:
if possible, could you also re-run pwmconfig and see if the fan sounds like it speeds up properly?

it should go off-low-medium-high but if the pins are wrong it could go off-medium-low-high.

masterlog80, 2021-03-06:
Ok, gimme some time and thank you!

masterlog80, 2021-03-06:
First step done:

```
root@NAS:/etc/flash-kernel/dtbs# flash-kernel
Using DTB: orion5x-linkstation-lsql.dtb
Installing /etc/flash-kernel/dtbs/orion5x-linkstation-lsql.dtb into /boot/dtbs/4.19.0-14-marvell/./orion5x-linkstation-lsql.dtb
Taking backup of orion5x-linkstation-lsql.dtb.
Installing new orion5x-linkstation-lsql.dtb.
flash-kernel: installing version 4.19.0-14-marvell
flash-kernel: appending /etc/flash-kernel/dtbs/orion5x-linkstation-lsql.dtb to kernel
Generating kernel u-boot image... done.
Taking backup of uImage.buffalo.
Installing new uImage.buffalo.
Generating initramfs u-boot image... done.
Taking backup of initrd.buffalo.
Installing new initrd.buffalo.
root@NAS:/etc/flash-kernel/dtbs#
root@NAS:/etc/flash-kernel/dtbs# reboot
root@NAS:/etc/flash-kernel/dtbs#
```

masterlog80, 2021-03-06:
Uh.... after reboot, the red lamp is blinking 6 times and there is some R/W activity on HDD 1. :-( 
Not sure if this is temporary or something has broken....

1000001101000, 2021-03-06:
hmm.

The most likely explanation is something is wrong with those boot files we just generated though that doesn't really match the error we see. 6 red blinks is usually what happens when it fails to find the boot files entirely, if the device tree I generated was bad I would expect the power led to just keep blinking forever since it would crash after getting past the boot loader.

I think the best thing for you to do would be to take a look at your /boot volume. If the problem is the new boot files you should be able to put the *.bak files back in place. 

I'll work on getting my LS-QL out of storage and try the new DTB on mine to validate it before we have you try again. 

I'd also like to take a look at your /boot setup when you get your device working again, something about this seems strange.

masterlog80, 2021-03-06:
Yeah, I will see if I can fix this without reinstalling.
Thank you for now!

1000001101000, 2021-03-06:
I got mine setup and updated, then installed the new DTB. It booted just fine and the fan issue appears to be resolved: 

```
$ ssh root@lsql
Linux lsql-Buster 4.19.0-14-marvell #1 Debian 4.19.171-2 (2021-01-30) armv5tel
Last login: Sat Mar  6 11:33:43 2021 from 192.168.1.146
root@lsql-Buster:~# sensors
gpio_fan-isa-0000
Adapter: ISA adapter
fan1:        5000 RPM  (min =    0 RPM, max = 5000 RPM)

root@lsql-Buster:~#
```

I'm wondering if there is something wrong with your /boot that is making it not extend to all the drives. That might explain why you would get 6 red blinks, if it was trying to boot from a drive with a blank first partition or something. I'm not sure what criteria the device uses to choose which drive it boots from (even in RAID1 it just boots from one initially).

For other devices where I've observed the boot process I've seen that it checks for signature in the MBR to determine if the drive was formatted specifically for this device or elsewhere and will also compare the timestamps of the boot files across the drives. I've never determined how they use this information to pick a drive, and I think it varies between models. It just proves that the logic is somewhat complicated and may be difficult to predict.

1000001101000, 2021-03-06:
I pushed the new dtb to the repo and updated the installer images. 

If you end up needing the installer again make sure to use the new images

masterlog80, 2021-03-06:
I will try, thanks!

masterlog80, 2021-03-07:
The  an issue seems to be fixed now:

```
root@debian:~# sensors
gpio_fan-isa-0000
Adapter: ISA adapter
fan1:        5000 RPM  (min =    0 RPM, max = 5000 RPM)

root@debian:~#
```

 However `pwmconfig` ends with an error mentioning a sensor is missing:

```
root@debian:~# pwmconfig
# pwmconfig revision $Revision$ ($Date$)
This program will search your sensors for pulse width modulation (pwm)
controls, and test each one to see if it controls a fan on
....
The fancontrol script can automatically respond to temperature changes
of your system by changing fanspeeds.
Do you want to set up its configuration file now (y)?
/usr/sbin/pwmconfig: There are no temperature-capable sensor modules installed
root@debian:~# 
```

So I guess I can't set the fan speed in relation with the temperature. Am I wrong?

Regards,

masterlog80, 2021-03-07:
For the moment, I run the following commands to set the speed and I may create a boot script:

```
echo "1" > /sys/devices/platform/gpio_fan/hwmon/hwmon0/pwm1_enable 
echo "1500" > /sys/devices/platform/gpio_fan/hwmon/hwmon0/fan1_target
root@debian:/# sensors
gpio_fan-isa-0000
Adapter: ISA adapter
fan1:        1500 RPM  (min =    0 RPM, max = 5000 RPM)

root@debian:/#
```

Regards,

1000001101000, 2021-03-07:
I believe that the Orion SoCs don’t have a temperature sensor, I believe this is the case for some kirkwoods too.

I believe the stock firmware uses the drive temperature via SMART. You could write a script which does something similar.

masterlog80, 2021-03-07:
Yeah, I will try installing `smartmontools` to address this.
Thanks for the hint!

closed 2021-03-07 by masterlog80