1000001101000/Debian_on_Buffalo #94: TS1400R - Root

darkdrgn2k, 2021-03-14:
Posting here because https://github.com/1000001101000/acp-commander has no issue support.

Wondering if you have any insight into getting root on a TS1400R.

```
-bash-4.2# java -jar acp_commander.jar -t 192.168.50.100 -o
admin password:
Reset root pwd...
Starting Telnet .........Failed!

Unable to detect telnet server.
This could be a firewall issue but more likely this model does not have a telnetd binary installed.
Consider using "-s" as an alternative.
-bash-4.2#
```

1000001101000, 2021-03-14:
Good to hear from you!

I’ve now enabled issues over there.

I think I connected on the TS1400D a long time ago but I don’t remember the details.

 Depending on what you’re looking to do the “-s” shell option is often good enough. Otherwise I have used a statically-linked version of busybox to provide telnetd when it isn’t available.

checkout this for that process:
https://buffalonas.miraheze.org/w/index.php?title=Linkstation_LS520D&mobileaction=toggle_view_desktop#Telnet

darkdrgn2k, 2021-03-14:
I did try  `-s` and i get a `>` prompt but nothing i type seems to work.
The only action i get is `exit` which exists

My goal , to try to do something more then CIFS on this box.  Id love to get rsync working, or maybe some RX/TX stats or something else. Can't even get NFS working.

I find the ui ridiculously limiting.

Ill try the linked posted and report back.

1000001101000, 2021-03-14:
I’d probably start by using -f and -c to verify it is discovered and accepting commands. -s usually works on literally everything provided you have the right password/etc.

Debian works on this device which opens up a lot of possibilies.

darkdrgn2k, 2021-03-14:
Im a little apprehensive doing a full Debian install on the box at the moment. 

`-f` does work
```
-bash-4.2# java -jar acp_commander.jar  -t 192.168.50.100 -f
ACPdiscovery reply
N-ON-DC-NAS     192.168.50.100  TS1400R(GOSANJYOU)      ID=003239       mac: 88:57:XX:XX:XX:XX  FW=  1.670
N-ON-DC-NAS
/192.168.50.100
88:57:XX:XX:XX:XX
TS1400R(GOSANJYOU)
003239
1.670
0BCXXXXXX
Found 9 device(s).
```

i could not get `-c` to work . 

` java -jar acp_commander.jar  -t 192.168.50.100 -c 'ls'` just asks for password then returns next line

darkdrgn2k, 2021-03-14:
Busy Box install attempt:
```
java -jar acp_commander.jar -t 192.168.50.100 -xfer bin/busybox -xferto /tmp
admin password:
-bash-4.2# java -jar acp_commander.jar -t 192.168.50.100 -o
admin password:
Reset root pwd...
Starting Telnet .........Failed!

Unable to detect telnet server.
This could be a firewall issue but more likely this model does not have a telnet                                                                                                                                                             d binary installed.
Consider using "-s" as an alternative.
-bash-4.2# nmap -sT 192.168.50.100

Starting Nmap 6.40 ( http://nmap.org ) at 2021-03-14 18:50 EDT
Nmap scan report for 192.168.50.100
Host is up (0.00089s latency).
Not shown: 991 closed ports
PORT      STATE SERVICE
80/tcp    open  http
111/tcp   open  rpcbind
139/tcp   open  netbios-ssn
443/tcp   open  https
445/tcp   open  microsoft-ds
873/tcp   open  rsync
2049/tcp  open  nfs
8873/tcp  open  dxspider
22939/tcp open  unknown
```

darkdrgn2k, 2021-03-14:
with `-d2`

```
-bash-4.2# java -jar acp_commander.jar  -d2 -t 192.168.50.100 -c 'ls'
Debug level set to 2
Target parameter -t found
Command-line parameter -c given
Using random connid value = 719E7281B7E2
Using cmd-line:
>>ls

Using target:   192.168.50.100/192.168.50.100
Using port:     22936
Using MAC-Address:      FF:FF:FF:FF:FF:FF
Sending 1/2
received ACP special command reply
ACP analysis result: ACP_STATE_OK
Trying to authenticate enonecmd...      ACP_STATE_OK
Password not specified, trying default password.
Sending 1/2
received ACP special command reply
ACP analysis result: ACP_STATE_PASSWORD_ERROR
admin password:
Sending 1/2
received ACP special command reply
ACP analysis result: ACP_STATE_PASSWORD_ERROR
Sending 1/2
received ACP special command reply
ACP analysis result: ACP_STATE_OK
Sending 1/2
received ACP special command reply
ACP analysis result: ACP_STATE_PASSWORD_ERROR
Sending 1/1
Received packet (CA10) has the error-flag set!
For 'authenticate' that is (usually) OK as we do send a buggy packet.
received acpcmd reply
ACP analysis result:
>ls
```

darkdrgn2k, 2021-03-14:
Interesting

I saw the `ACP_STATE_PASSWORD_ERROR` even though the password was correct.

So i went into the UI and reset the password to `password`

it got me root!

1000001101000, 2021-03-15:
Nice!

darkdrgn2k, 2021-03-15:
Any tips to enable ssh for a user?

seems `sftponly_config` blocks ssh and is regenerated every sshd restart
and i have not found a way to inject a non-nas user not to hit against sshd

1000001101000, 2021-03-15:
I think this worked for v-series and ls400 devices, might for ts1000.
https://github.com/rogers0/OpenLinkstation/blob/master/0_get-ssh/get-ssh.sh

even if it doesn’t work the process is probably similar

1000001101000, 2021-03-16:
In case you want to trade up:
https://www.ebay.com/itm/BUFFALO-TS3400R0804-TeraStation-3400-NAS-No-Hard-Drives-/154373440429?_trksid=p2349624.m46890.l49292

1000001101000, 2021-06-29:
Is there anything else you want me to look at for this issue?

closed 2023-06-07 by 1000001101000