1000001101000/Debian_on_Buffalo #97: LS220de upgrade from 4.19.0-6 to 4.19.0-16 doesn't work

Kulturnilpferd, 2021-04-06:
Hey there,
when upgrading to 4.19.0-16 my Linkstation LS220DE doesn't boot up at all. Is there some bugfix for the new version around or should I stick to older version?
Have renamed the bak to get my old stable version running again but really want to upgrade due safety and so on...

1000001101000, 2021-04-06:
Good to hear from you!

That's the first report I've received.... I sure hope dozens of these devices aren't about to start failing. I'll work on bringing my LS200 online and see if I run into the same issue or not.

have you made any changes recently like reformatting/removing a filesystem? 

Assuming there isn't some horrible regression in the LTS kernel (possible though unlikely) the next most likely thing I can think of would be some sort of issue causing a boot failure like a missing filesystem or something causing an issue with the new initrd that would have been generated.

Kulturnilpferd, 2021-04-06:
Awesome, thanks for quick reply and looking into it. 
Only thing I have changed is replacing the stock fan with another because the old one failed after years of using. Due to this change the `systemd-udevd` has after a restart about 100% cpu usage for a few minutes polling `/devices/platform/gpio-fan` a dozen time in a second.
It takes some time like 5 minutes or so, then the polling slows down and the cpu usage goes to 2,5% for that process.
After the installation years ago I didn't changed the filesystem or its components. 
Only `apt update / apt upgrade` from time to time...

1000001101000, 2021-04-07:
I just loaded up my ls220de and upgraded all the packages except the kernel (it was still back on 4.19.0-9-armmp), then rebooted and installed 4.19.0-16-armmp and rebooted again. 

It came up fine and appears to be working properly:

```
# uname -a
Linux ls220de-Buster 4.19.0-16-armmp #1 SMP Debian 4.19.181-1 (2021-03-19) armv7l GNU/Linux
```

Kulturnilpferd, 2021-04-08:
Ok, thanks for testing. I will update my LS again and look into the logs maybe there is some things I have missed.

Kulturnilpferd, 2021-04-08:
Very strange, after `apt-get install --reinstall linux-image-armmp` (and unplugged all USB Devices) the upgrade worked fine without any issues.
So it's working now. thanks for looking into it 👍

closed 2021-04-08 by Kulturnilpferd

1000001101000, 2021-04-08:
must have been something goofy going on with the initrd generated when you upgraded the first time.