1000001101000/Debian_on_Buffalo #105: Starting NAS remotely? (LS-WVL)

ilendemli, 2021-05-30:
Hi, Thanks for this project. I have an LS-WVL running OMV. I was wondering, is it possible to put the NAS to a standby mode where one can simply issue a command (open a link, send a WOL packet, etc.) to remotely power up the device to its default OS? Thanks in advance!

1000001101000, 2021-05-31:
Good to hear from you!

WOL is a topic I haven't spent much time on, and I think the time I've spent I was looking at the topic incorrectly. 

I suspect that many of these devices are configured to allow the bootloader to to wait for a WOL packet and trigger a boot, particularly the devices where something like that was used in the stock firmware.

I think the first step to trying to get that working would be for you to try shutting down the device (which the power switch set to auto if applicable) and try sending a WOL packet to the device's MAC address and see if it boots. If not you can also try enabling WOL using ethtool before shutting it down.

alexhorner, 2021-05-31:
This will be something quite interesting to investigate. @ilendemli Are you looking for an instant wake style sleep state or are you okay with some considerable delay on startup, just the automation is important?

ilendemli, 2021-05-31:
Hey! Thank you for the hint with ethtool @1000001101000, that worked flawlessly for WOL.
@alexhorner delay is fine for me, the "default" behaviour from buffalo would be enough too. I think it replaces some files before rebooting which then has a lightweight webserver hosting a page that triggers a command that again replaces the files back to the original state and reboots. I think the hard part would be to have the HDDs spin down while hosting the lightweight webserver.

1000001101000, 2021-05-31:
Nice!

could you post your ethtool/etc output so I can add it as an example on the wiki?

ilendemli, 2021-06-01:
Sure!

1. SSH into device and install ethtool by issuing `sudo apt install ethtool`
2. Find network interface by running `ip link show`. it is `eth0` for me
3. Enable WOL for `eth0` by running `sudo ethtool -s eth0 wol g`
4. Check if WOL is enabled by executing `sudo ethtool eth0`, Output should have `Wake-on: g`
5. Try shutting down your NAS (keep the switch on AUTO) and send a MagicPacket to wake up your device remotely.

I am not sure if the setting is reset after each boot so one way to enable WOL every time on boot is to edit `/etc/network/interfaces` and add `up ethtool -s eth0 wol g` to your primary network interface.
```# The primary network interface
allow-hotplug eth0
iface eth0 inet dhcp
    up ethtool -s eth0 wol g
```

1000001101000, 2021-06-29:
Thanks again!

closed 2021-06-29 by 1000001101000