1000001101000/Debian_on_Buffalo #108: LS720D Series compatibility?

rlyshw, 2021-06-18:
I'm new to the buffalo/LinkStation ecosystem so apologies if I come across as ill-informed.. The info out there on these systems seems to be a bit sparse.

Related to this project; I'm not seeing an installer specifically for the ls720d series. Is it OK to use some other version, like ls441? The wiki isn't clear on potential device series version interops.

I can go ahead and try it out for myself but wanted to check here before potentially bricking the device.

Either way I'll use the acp_commander system to start poking around with the stock firmware to learn more about the LS system. Maybe I can help out with the 720d series installer, if needed :) 

Thanks!

1000001101000, 2021-06-18:
Good to hear from you!

The current installers will not work with the LS720 series, because of the way hardware detection works on non-X86 systems the kernel images for each device end up being very specific and non-interchangeable ( though devices based on the same SoC usually mostly work).

I'm eager to look at adding LS720D support or at least figuring out what would be required to accomplish it. I usually don't make much progress until devices start hitting the used market for cheap but if you have one and are willing to do some (safe) experiments we can see how far we can get.

The first step is to see if we can get root access with acp_commander and start poking around. See if acp_commander detects the device and if so try using the built in shell feature (-s). If that works we can move to trying to start telnet or ssh.

rlyshw, 2021-06-21:
I got shell access and "open-boxed" the system with acp_commander (the "-o" command). I've now got root-level access to the system via telnet.
(adding our findings from discord here for posterity)
We found that you cannot use the default "broadcast" address (FF:FF:FF:FF:FF:FF) to send acp_commander commands to this device, the system MAC needs to be specified with -m. Oddly enough, the -f flag does work to initially find the device. But specific commands must specify the MAC.

1000001101000, 2021-06-21:
Thanks!

I've made a note of a possible way to get acp_commander to do that automatically in the future, It should be a fairly simple change.

Now that you have root access we can start collecting information about hardware and boot environment and try to determine what the possibilities are for running Debian/etc. I spent a little time looking at the firmware and it looks like it boots from the on-board NAND which will make that more difficult than the older devices with boot from the drives. It does look like there is a USB recovery boot option in the boot loader which might be useful if we can figure out how to activate it. 

Usually what I do at this point I create a Debian install on the device inside a chroot environment. That environment can then provide additional tools for testing. 

We can walk through all of the above in the next discord session. Thanks again for working on this!