1000001101000/Debian_on_Buffalo #109: Getting connection refused when running ssh installer

RobertTTaylor9517, 2021-06-25:
When I try to run the ssh installer@<ip address> command, I get the connection refused on port 22 error. When I run the ping on the ip address, I get a response so the server is up. Am I missing a step???

Running on linkstation ls210d

1000001101000, 2021-06-25:
Good to hear from you!

If you have the terminal output from the steps you took I can take a look for issues. 

There are a few ways to have something like that happen, most of which involve failing to get the correct installer files into place in one way or another.

- If the device was in EM mode for some reason /boot would just have been a ramdisk and rebooting would likely have returned it to EM mode when booting again.
- If you didn't rename the kernel (uImage.Buffalo) properly you could have booted the stock kernel with the installer which would likely fail early in the process 
- If you put the files somewhere other than /boot I'd expect it to just boot back into the stock fw (which would respond to ping but not ssh)
- etc

Does the device show up in ACP Commander? If so we should be able to connect and figure out what happened and fix it. If not it'll take some work to get it back into an accessible state to try again.

RobertTTaylor9517, 2021-06-25:
I followed the instructions exactly. I renamed the kernel and downloaded the correct uImage.buffalo.ls210d and renamed it to uImage.buffalo and transferred the files over. But when I run acp_commander.jar -f I do not see it listed. Is it bricked?

1000001101000, 2021-06-25:
probably the most common mistake folks make is downloading the html of the page for the install rather than download the actual files. We can't tell if you did that or not unless we get into the device. I wouldn't think that's what happened since you got a ping back.Getting a ping back would generally indicate that it booted into something pretty far in most scenarios. 

one other thing to try is to give it a few minutes, there is a pretty big lag between when it joins the network and when it starts ssh (several minutes). It might be worth a couple of restarts too. It's possible, though unlikely, for the installer to fail before starting ssh if it fails to reach out to the debian mirror at startup. I suppose if you had a firewall blocking ftp.debian.org that could explain things too.

In any case the device isn't "bricked". You should be able reinstall the stock firmware via TFTP and then try again. For most devices I recommend connecting the drive to a PC since I'm not a big fan of TFTP though that model is a pain to take apart.

https://buffalonas.miraheze.org/wiki/Linkstation_LS210D
https://buffalonas.miraheze.org/wiki/Restoring_Stock_Firmware_via_TFTP
https://buffalonas.miraheze.org/wiki/Restoring_Stock_Firmware_without_TFTP

ledaicntt, 2021-06-27:
I also got the same error. My Nas is TS-XEL.
This is files and command which I copy to /boot:

initrd.buffalo (MD5: 957E6FC8AFDCF8FE30F9059475A271F8)
uImage-88f6281.buffalo (MD5: 979247D3FA9AD2502650C0726606B2D6) (Rename from uImage-88f6281.buffalo.tsxel)

java -jar acp_commander.jar -t 192.168.1.11 -pw password -xfer initrd.buffalo -xferto /boot/
java -jar acp_commander.jar -t 192.168.1.11 -pw password -xfer uImage-88f6281.buffalo -xferto /boot/
java -jar acp_commander.jar -t 192.168.1.11 -pw password -c reboot

After reboot, LCD show message: "Terastation arm Debian Installer", but I can not connect by ssh to install.
I tried many times, with FW stock version: 1.75, 1.71 but the results are the same: "Connection refused".

Can you tell me where did I go wrong?

1000001101000, 2021-06-27:
I wouldn’t think these two issues would be related given they use different installers/architectures…even network drivers. 

it sounds like they are getting stuck at a similar point which might not be a coincidence.

We can tell the installer boot process must be getting pretty far on the TS-XEL since it’s putting the installer message on the LCD. The ls200 responds to ping which means either the installer or stock fw are getting pretty far.

Does the TS-XEL respond to ping? One possibility is that it just got a different IP address from DHCP than it had with stock fw.

I’ll get out my TS-XEL and ls220 and load the Buster installers and see if I run into issues.

ledaicntt, 2021-06-27:
Thank you for responding.
I'm still trying to fix the problem. 
I checked the DHCP IP and it's correct. I also tried ping successfully. Everything is fine, except error:  [Network error: Connection refused]

1000001101000, 2021-06-27:
I was booted into the Buster installer on both an LS220DE and TS-XEL. I was able to connect to both via SSH without issue.

I did notice that I couldn't connect to the TS-XEL if the ethernet cable was in the bottom port. It's possible auto-detect doesn't work in this scenario. If you haven't already, try with the cable in the top port.

There are a bunch of things that happen in the background while the device starts up which need to succeed before it starts SSH. Specifically it needs to connect to the debian mirror to download and install the SSH server. If there's any chance you have a firewall that would interfere with that process it would explain a lot of what we're seeing.

ledaicntt, 2021-06-28:
Thanks to your guidance, I tried switching from the top port to the bottom port (slightly different from yours). And great, SSH connected successfully.

1000001101000, 2021-06-28:
I made myself a note to look into the ethernet autodetect sometime in the future, this seems like odd behavior unless both of us have a bad ethernet port or something. I'm glad to hear it's working now!

1000001101000, 2021-06-30:
@ledaicntt I did some research and found that increasing the timeout for link detection solves the autodetect issue for my TS-XEL. I think it should work from either ethernet port in the future.

1000001101000, 2021-06-30:
@RobertTTaylor9517 The only thing I can think to look at until you get a chance to restore the firmware and start over would be to boot up the device and wait ~15 min to ensure it finished booting and running any disk checks/etc that might be going on and then do a full port scan of the device to see if anything is listening on it at all. 

Since that model doesn't have flash I don't think it could be booting from flash.
if it was booting from the debian kernel it would also need the debian initrd to start the network, in which case it would be in the installer but not starting ssh for some reason I don't understand.
if it was booting the stock firmware files for some reason I would expect it to eventually appear in acp_commander though that could take a while if it does a disk check for some reason.
if the files were mismatched somehow I wouldn't think it would successfully get an ip address or respond to ping, though it might be possible if it was the stock kernel with the installer initrd but not the other way around.

It should be simple enough to figure out once we can get access to the /boot on the drive either by booting TFTP or connecting the drive to a PC.

1000001101000, 2021-07-08:
@RobertTTaylor9517 let me know if you want additional help with this.

closed 2021-07-16 by 1000001101000