1000001101000/Debian_on_Buffalo #110: LS-QVL reboot and shutdown Issues

VR-Punk, 2021-06-26:
hi,

if i use the reboot command on the ls-qvl, then it will shutdown. What command do i have to use to reboot the system ?

1000001101000, 2021-06-26:
Could you post the output of:

`ip addr`
`cat /usr/local/bin/phy_restart.sh`

VR-Punk, 2021-06-26:
cat: /usr/local/bin/phy_restart.sh: no such file or directory

1000001101000, 2021-06-26:
did you use my installer? that usually puts everything in place.

The issue is that this devices decides whether to reboot or shutdown based on whether a certain register has been set in the ethernet controller. phy_restart.sh handles that during the reboot process, if it's not installed it never gets that restart signal and just powers off.

- grab phy_restart.sh and phytool-armel from here: https://github.com/1000001101000/Debian_on_Buffalo/tree/master/Tools
- then copy them into /usr/local/bin
- rename phytool-armel to  phytool
- run `ln -s /usr/local/bin/phy_restart.sh /lib/systemd/system-shutdown/phy_restart.sh` to install it

after that reboots should work as expected.

1000001101000, 2021-06-26:
I have a note on the wiki, though the wiki needs a ton of work.

https://github.com/1000001101000/Debian_on_Buffalo/wiki/Device-Specific-Notes#linkstation-ls400-series--v-series-ls-vlwvlqvl-and-likely-other-kirkwood-devices-ls-wxl-etc

1000001101000, 2021-06-26:
….they will also need execute privileges which might not get set depending how you copy them over.

If needed you can fix that with:
`chmod 755 <file>`

VR-Punk, 2021-06-26:
I asked this some time ago, but have a solution for the fan of the ls-qvl ? The fan is vey loud at full power !

1000001101000, 2021-06-26:
Did that solve the shutdown issue?

I have notes about fancontrol here:
https://github.com/1000001101000/Debian_on_Buffalo/wiki/Post-Installation-Options#gpio-fans

VR-Punk, 2021-06-26:
i did exactly what you wrote but if i will do 

$ sudo reboot

this system shutdowns

1000001101000, 2021-06-26:
Could you post the output of: `ls -l /usr/local/bin`

VR-Punk, 2021-06-26:
$ ls -l /usr/local/bin
-rwxrwxrwx 1 root root 144312 26. Jun 23:58 phy_restart.sh
-rwxrwxrwx 1 root root  30096 26. Jun 23:59 phytool

$ ls -l /lib/systemd/system-shutdown/
lrwxrwxrwx 1 root root 29 27. Jun 00:01 phy_restart.sh -> /usr/local/bin/phy_restart.sh

1000001101000, 2021-06-26:
The size of your phy_restart.sh looks way to big. Make sure you download the file rather than the html of the github page, that should fix it.

```Linux lsqvl-Buster 4.19.0-16-marvell #1 Debian 4.19.181-1 (2021-03-19) armv5tel
Last login: Sat Jun 26 17:28:26 2021 from 192.168.1.146
root@lsqvl-Buster:~# ls -l /usr/local/bin/
total 76
-rw-r--r-- 1 root root   319 Jul 11  2020 ifup-mac.sh
drwxr-xr-x 3 root root  4096 May  2  2020 micon_scripts
-rwxr-xr-x 1 root root 26468 May  2  2020 micro-evtd
-rwxr-xr-x 1 root root   514 May  2  2020 phy_restart.sh
-rwxr-xr-x 1 root root 30056 May  2  2020 phytool
-rw-r--r-- 1 root root   151 Jul 11  2020 rtc_restart.sh ```

VR-Punk, 2021-06-26:
does not work

$ ls -l /usr/local/bin
insgesamt 36
-rwxrwxrwx 1 root root   513 27. Jun 00:51 phy_restart.sh
-rwxrwxrwx 1 root root 30096 26. Jun 23:59 phytool

$ ls -l /lib/systemd/system-shutdown/
insgesamt 0
lrwxrwxrwx 1 root root 29 27. Jun 00:01 phy_restart.sh -> /usr/local/bin/phy_restart.sh

$ sudo reboot

causes shutdown

1000001101000, 2021-06-26:
Try executing phytool to verify it is working:
`/usr/local/bin/phytool`

VR-Punk, 2021-06-26:
$  /usr/local/bin/phytool
Usage: phytool read  IFACE/ADDR/REG
       phytool write IFACE/ADDR/REG <0-0xffff>
       phytool print IFACE/ADDR[/REG]
where

ADDR := C22 | C45
C22  := <0-0x1f>
C45  := <0-0x1f>:<0-0x1f>
REG  := <0-0x1f>

Examples:
       phytool read  eth0/0:3/1
       phytool write eth0/0xa/0 0x1140
       phytool print eth0/0x1c

Note: Not all MDIO drivers support the `port:device` Clause 45 address
format.

The `read` and `write` commands are simple register level
accessors. The `print` command will pretty-print a register. When
using the `print` command, the register is optional. If left out, the
most common registers will be shown.

Bug report address: https://github.com/wkz/phytool/issues

1000001101000, 2021-06-26:
Could you post the output of:

`ip addr`
and
`cat /usr/local/bin/phy_restart.sh`

VR-Punk, 2021-06-26:
$ cat /usr/local/bin/phy_restart.sh
#!/bin/bash
# place in /lib/systemd/system-shutdown/ for debian based systems
# other systems it's /usr/lib/systemd/system-shutdown/

phytool="/usr/local/bin/phytool"

##I think only needed for dhcp leases file
mount -o remount,rw /

ifup --no-scripts --force eth0
sleep 3

$phytool write eth0/0/22 3
if [ "$1" == "halt" ] || [ "$1" == "poweroff" ]; then
    $phytool write eth0/0/16 0x0881
else
    $phytool write eth0/0/16 0x0981
fi
$phytool write eth0/0/22 0

ifdown --force eth0
mount -o remount,ro /

exit 0

$ ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu * qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether * brd *
    inet * brd * scope global eth0
       valid_lft forever preferred_lft forever
    inet6 * scope global dynamic mngtmpaddr 
       valid_lft 9747sec preferred_lft 2547sec
    inet6 * scope link 
       valid_lft forever preferred_lft forever
10: *: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether * brd * link-netns *
    inet * scope global *
       valid_lft forever preferred_lft forever
    inet6 * scope link 
       valid_lft forever preferred_lft forever

1000001101000, 2021-06-26:
How did you install debian on this system? Is there any chance it wouldnt have systemd or something?

VR-Punk, 2021-06-27:
i use system.d for vopono, it is a tool to run applications through VPN tunnels via temporary network namespaces. i use the --user flag of systemctl.

https://github.com/jamesmcm/vopono

I installed Bullseye from this server

http://ftp.debian.org/debian/dists/testing/main/installer-armel/current/images/kirkwood/network-console/buffalo/ls-qvl/

1000001101000, 2021-06-27:
I don’t know a reason why it wouldn’t work on bullseye but I have not tested it yet. I’ll try upgrading mine and see what happens.

I have used other systemd restart scripts on Bullseye so I expect that behavior to be the same. I imagine there could be a problem with the newer kernel, if so my test should show that.

As far as I can tell you have everything installed correctly now, though I could be missing something.

1000001101000, 2021-06-27:
I tried installing the current Bullseye kernel on my LSQVL. I was able to boot the new kernel without issue but mine also failed to reboot when I tried that.

I’ll look into that when I get a chance. I’ll need to come up with a solution as part of adding Bullseye support to this project.

Before I figured out the whole phy thing I used to make mine reboot by triggering the cpu watchdog. I’ll see if adding that to the script can make it work again in the short term.

1000001101000, 2021-06-27:
I made some adjustments to the script which is working on my device. Could you give it a try?

```
#!/bin/bash
# place in /lib/systemd/system-shutdown/ for debian based systems
# other systems it's /usr/lib/systemd/system-shutdown/

phytool="/usr/local/bin/phytool"

##I think only needed for dhcp leases file
mount -o remount,rw /

ifup --no-scripts --force eth0
sleep 3
mount -o remount,ro /

$phytool write eth0/0/22 3
if [ "$1" == "halt" ] || [ "$1" == "poweroff" ]; then
    $phytool write eth0/0/16 0x0881
else
    $phytool write eth0/0/16 0x0981
fi
$phytool write eth0/0/22 0

exit 0

```

1000001101000, 2021-06-29:
Let me know if that worked for you, if so I'd like to add that version to the project right away.

VR-Punk, 2021-06-30:
thx, i will try that !

VR-Punk, 2021-07-05:
thank you very much, that works for me !

closed 2021-07-05 by VR-Punk

1000001101000, 2021-07-06:
excellent, I've added that version to the repository and installer images.