1000001101000/Debian_on_Buffalo #116: Ls420d not exposing port 22 for ssh, on installer image - buster/armhf

wodgey, 2021-08-02:
Hello again. I have been so impressed with Debian running on my ls420DS that I recomended a similar model to my brother (although the ls420d variant rather than DS).

I'm booting the device into EM mode and then TFTP booting the initrd and Uimage files but port 22 isn;t open, for ssh connection. 

I am in process of poking sshd_config within the hdrootfs.img and have sshd running for admin (I have just removed the x from root:x:0:0:root:/root:/bin/sh in /etc/passwd, hoping to ssh in as admin then su root with blank pass, to then set/change as desired.

I would prefer debian though, rather than buffalo though, if you;re able to help?

1000001101000, 2021-08-03:
Good to hear from you!

I'm not sure if I understood everything you're telling me, please let me know if I am missing something.

Are you saying you are attempting to boot into the Debian installer via TFTP? If so, how are you accomplishing this? The Debian installer images are based on Debian's netboot image which requires access to the internet fairly early in the startup process. If you are directly connecting to a PC for the TFTP process this may explain the issue.

I believe the instructions on the wiki demonstrate copying the files over with ACP Commander and then triggering a reboot. If you've already got access to the command line via ACP Commander this process should work well for you.

wodgey, 2021-08-03:
Ah, that might explain things. Yes, I was trying to get ssh access before going through the hassle of Internet Connecton sharing. Netboot installer would not have had web access because I was waiting to get access to a shell, before grabbing anything from the web.

However, I have acchieved my goal anyway using my own modded stock firmware.

ACP Commander can;t see the device running v1.87 so I am unpacking the hddrootfs.img, making edits before then repacking.
Doing this, I have make the required changes to sshd_config and the /etc/init.d/sshd.sh script so that sshd starts up listening. Whilst having the filesystem cracked wide open on my WSL2 VM, I took the time to create /root/.ssh/authorized_keys.

I now have root access over ssh, and have installed optware IPKG package manager and used that to install transmission-daemon and no-ip.com updater. which is the total sum of what my brother needs.

closed 2021-08-03 by wodgey

wodgey, 2021-08-05:
Ah, ok. problem found! I had a rogue DHCP server running on my laptop, issuing leases without specifying gateway/dns. I had forgotten it was installed and running.
What was happening was the netboot installer was grabbing IP Addr and mask from my laptop but then unable to contact the outside world.
anyway, all good again.
I needed to edit the sshd_config on my device after locking myself out, so I tried your installer again and it worked, after I figured out the networking issue

1000001101000, 2021-08-05:
That makes sense, I'm glad it work out.