1000001101000/Debian_on_Buffalo #117: Network issues

iamnorbiato, 2021-08-31:
Hi there how are you ?

Looks like we are having some kernel issues like this:

Can't allocate skb on queue

will force stop/start to get the messages with m ore details...

iamnorbiato, 2021-08-31:
Aug 31 12:29:25 homenas kernel: [  381.341525] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.348280] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.355035] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.361789] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.368544] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.375299] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.382054] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.388808] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.395563] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.402318] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.409072] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.415827] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.422582] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.429337] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.436092] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.442847] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.449602] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.456357] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.463112] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.469867] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.476622] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.483377] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.490131] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.496886] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.503640] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.510395] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.517149] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.523904] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.530659] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.537414] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.544168] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.550923] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.557679] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0
Aug 31 12:29:25 homenas kernel: [  381.564433] mvneta d0074000.ethernet eth0: Can't allocate skb on queue 0

1000001101000, 2021-08-31:
Good to hear from you!

which kernel version are you running?

are you noticing problems beyond seeing these errors in the log?

iamnorbiato, 2021-08-31:
Hi there.

Yes - not quite sure when it has started - but I was loosing the storage connection twice a day - comes and goes... then 3 times a day (was afraid that might be a disk issue but was not) - today it started to loose the network 10 maybe 15 minutes after reboot (manually reboot).

Checked for update/upgrade - it made some for ssh, systemd no kernel update.
Will reboot the storage now and grab the right kernel version and let you know.

Very nice - very very nice to hear from you again as well !

iamnorbiato, 2021-08-31:
Here it is
$ uname -a
Linux homenas 4.19.0-17-armmp #1 SMP Debian 4.19.194-3 (2021-07-18) armv7l GNU/Linux

1000001101000, 2021-08-31:
I believe someone had a simlar issue and was able to resolve it. Could you give this a try:

https://github.com/1000001101000/Debian_on_Buffalo/issues/35

iamnorbiato, 2021-08-31:
One valuable information:
I stopped webmin and transmission server no more network issues - smb working fine - so far so good
I can live without these 2 guys

iamnorbiato, 2021-08-31:
trying 
$ sudo sysctl -p
vm.min_free_kbytes = 4096 (in my case I added this row in the end of sysctl.conf file)

starting webmin and transmission to see whether or not will crash

closed 2023-06-07 by 1000001101000