1000001101000/Debian_on_Buffalo #121: LS-WXL inreachable after reboot

alphonus, 2022-01-05:
I used Option 1 to install Debian Buster on my Linkstation. But after the reboot the fans only start spinning up und it doesnt show up on my local network. Is there a way to debug using a serial connection?

1000001101000, 2022-01-07:
Could you describe the partioning/formatting option you chose? This sounds like an incorrectly configured /boot

I’m not aware of anyone connecting a serial console to this model. It was around that era Buffalo started making that difficult.

1000001101000, 2022-04-11:
@alphonus Do you still need help with this? I just posted some new installer files which include Bullseye support. As part of the testing I made sure the new installers work on LS-WXL.

closed 2023-06-07 by 1000001101000