1000001101000/Debian_on_Buffalo #122: TS-XL No Ports Open After Reboot

DudDoodoo, 2022-01-07:
I'm trying to install Debian on a TS-XL using the stock firmware, following the instructions in the guide.  I was able to connect with acp-commander, move and load the correct files just fine.  

Upon rebooting, the system boots up and displays terastation arm debian installer on the front panel, and pulls an IP from DHCP.  I scanned all possible ports using nmap and nothing is open, obviously including 22.  Any ideas?

1000001101000, 2022-01-08:
Good to hear from you!

The best way to find out exactly what is going on is to hook up a USB->Serial adapter to the front Serial port, there is probably an error message displayed that could help with troubleshooting. Most other models don't have this and we can usually figure it out without if needed. 

The fact that it set the LCD tells us it got into the early installer commands. If it got an address from DHCP it means that it detected interfaces and set the IP etc. Between the two we can infer that a lot of things worked (like loading kernel modules etc).

The next thing it does is reach out to Debian's servers and start pulling down additional installer components which I believe includes the SSH server. Either way I think it won't get that far if it fails to connect to the Debian mirror. If there's anything about your network that might prevent it from reaching the default mirror (ftp.debian.org?) or generally doing DNS lookups that might explain what you see.

In the past there were issues with the entropy pool not populating sufficiently for the SSH server to generate it's keys at start up which looked similar to what you are seeing. That issue was resolved around when Buster was first released but is the only other issue I can remember that would result in what you see. 

The first thing I would try is just rebooting the device a couple of times and see if it succeeds. If there was some momentary network/internet issue it could cause it to stall which simply trying again could fix. If your network connection is unreliable for some reason you'll likely see a lot of failures since the debian installer doesn't seem to have retry logic. Usually that isn't an issue but I did talk to someone using some weird wifi bridge setup which gave them endless problems. Likewise this will be more sensitive to a bad ethernet cable etc.

1000001101000, 2022-04-11:
@DudDoodoo Do you need help with this? I just posted new installer files for this device which include support for Debian Bullseye with Kernel 5.10, I'd recommend using those to get the latest version of everything.

closed 2023-06-07 by 1000001101000