1000001101000/Debian_on_Buffalo #123: LS441D no SSH after Debian install

Asmodin, 2022-01-08:
Installing Debian seemed to work without any Issues and the Device is pingable after booting into Debian post-installation, but there is no open SSH port.

Did I maybe mess something up with the partitioning for the boot process to stop midway?
[image]

I want to change the Stock created XFS Raid 1 Partition to ext4 since my usecase includes backing up many thousands of very small files and performance is tanking very hard on Stock with XFS, making it pretty much unusable. Can I simply do it like that in the Installer or do I have to leave it untouched and do that post-install?

This is the only thing that comes to mind what I might be doing wrong. Hope you got an Idea and thanks in advance for your time and amazing work on this project

Asmodin, 2022-01-08:
```
Nmap scan report for 192.168.178.36

Host is up (0.0075s latency).

All 1000 scanned ports on 192.168.178.36 are in ignored states.

Not shown: 1000 closed tcp ports (reset)

MAC Address: DC:FB:02:DA:25:3A (Buffalo.inc)
```

post-install nmap scan

1000001101000, 2022-01-08:
Good to hear from you!

assuming your raid arrays correspond to the partitions the way I would expect (same as stock firmware or at least with /boot as partition 1). I would think that work. I typically use ext4 for my rootfs (ext3 is only required or /boot) but I wouldn't expect that to be an issue. 

As far as I know configuring your /mnt/array... that should work and conveniently format it and create the fstab entry for you. I don't recall if I've ever done it myself or not....I assume the reason I think it works is because I did it at some point....but I'm not certain. 

However....What you are seeing is exactly what would happen if that array was failing to mount at boot. The other thing that comes to mind would be some long running fsck, but I don't think that would happen with all newly formatted filesystems like that. 

Looks like you already tried letting it sit for an hour and check again which might be my next suggestion. After that I would probably start over and try again leaving the array alone until after getting into the eventual Debian install.

Asmodin, 2022-01-08:
Hey,

thanks for your quick reply!

I'm already in the process of trying another clean reinstall leaving the raid array alone until after the install is complete and will try your post-install guide just with using ext4 instead of xfs.

Fingers crossed it'll boot this time!

Asmodin, 2022-01-08:
Update:

Sadly it's back to responding to pings but no SSH post-install, will let it sit for a while to see if anything changes but no Idea what I'm doing wrong at this point

1000001101000, 2022-01-08:
Which version did you install? Bullseye?

Starting the network but not ssh is a pretty specific failure mode for these. The network kernel modules are in the initrd, so even starting up the network implies that it successfully loaded the files from /boot and booted them at least far enough to get an IP and probably farther.

There’s generally not much that can be done wrong in the installer other than messing up /boot and that would usually cause it to either boot back into the installer or fail booting entirely.

I’ve got some automation set up that generates new installer images whenever Debian publishes new images (the net installer doesn’t work if they get out of sync). Sadly, this means I haven’t tested this exact installer image myself.

The main things you could look at for now is what the dhcp client id shows as in your router if that is available and whether it shows up in acp_commander/etc in case it somehow went to EM mode. If you can access the filesystem you could look at the syslog and see if there’s anything there.

I’ll see if I can find some time to try it on my TS3400D which is pretty similar and has a serial console built in….there’s nothing like a good error message to help with troubleshooting.

Asmodin, 2022-01-08:
Yeah, I tried the Bullseye Version without realizing there are others too. derp.

Gonna try Buster and see if that fixes it, 

Accessing logs isn't easy for me right now sadly, no Linux System here right now, will try to dl a Debian live Image later if the error persists and try to grab a syslog.

1000001101000, 2022-01-08:
Bullseye is the current “Stable” and in theory what I recommend. As “oldoldstable” if Stretch breaks I’ll probably discontinue it.

Buster might be worth a try just to see if it works, that would give more data.

Asmodin, 2022-01-08:
Tried to gain access to the LS441D logs by plugging a Drive to a Windows Machine and reading with Tools like Linux Reader but that doesn't seem to play nice with a (degraded) Raid 1, no Luck there

Asmodin, 2022-01-08:
Not sure how that's possible or why it would happen, but the installer for Buster already feels much more responsive. On the Bullseye installer I had to wait roughly 5 seconds between almost any Setting I made while it's more or less instant to 1 second on the Buster installer Image.

Very peculiar and probably has nothing to do with my Issue, but thought I'd still mention it.

Asmodin, 2022-01-08:
> The main things you could look at for now is what the dhcp client id shows as in your router if that is available and whether it shows up in acp_commander/etc in case it somehow went to EM mode. If you can access the filesystem you could look at the syslog and see if there’s anything there.

No Access to Router here sadly, but intense nmap scan identified it as Debian at least and the Status LED was constant white / not in EM mode, also didn't show up in acp_commander or LSUpdater. So I think it's pretty safe to assume that it was booting into Debian just fine up to a certain point, just not far enough to make it to SSH.

If you need/want a syslog file for finding potential Issues with auto generated Bullseye installer Image, I could try re-running the setup on a different spare single HDD that I got here so that I should be able to extract it from a Windows Machine after the failed Boot occurs.

Asmodin, 2022-01-08:
Buster installed with no Issues whatsoever and booted all the way to SSH access even with setting up and mounting the Raid Storage as ext4 during Install, so something might be wrong with the LS441D Image for Bullseye

1000001101000, 2022-01-08:
Excellent! Well…. not excellent, but that narrows things down. I’ll take a look.

1000001101000, 2022-01-15:
My ts3400d install worked without issue, i’ll pull out the ls441 and try when I get some time this week

Anti-ctrl, 2022-02-20:
Just wanted to confirm, that i have the same issue. LS441D with no SSH, after a fresh install with Debian Bullseye.

I can Ping the Device, but nmap shows all ports closed.
I will wait a bit to see if the Device "recovers", but after reading this, I doubt it. I will Probably just Install Buster and see if that works.

PS: @1000001101000 Thank you alot for making this Possible! If i hadnt found this Project, i would've just yeeted that thing into a Bin.

1000001101000, 2022-02-20:
@Asmodin  @Anti-ctrl  sorry I haven't gotten around to trying this on mine yet. It's looking likely I'll have some time for testing this device on Bullseye again in the near future now that some other work is wrapping up.

whcrg, 2022-02-28:
Tried installing bullseye on my LS220DE. And ame thing here, bullseye installation went fine but no ssh (connection refused) after booting (and sitting for some 5 hours).

Need to try still powering it completely off and back again but not hoping much. My router tells me it has given the box an ip and box name is "Debian" so seem it has booted the new installation. The installer had some 2 letter name.

whcrg, 2022-03-01:
Try 1: ad to crontab of the system root `@boot sudo systemctl start ssh` in the hope ssh is not started for some reason. Not helping. So it apparently has trouble mounting the root. (Also leds don't light up but I am not sure they used to.) Gets IP.

Try 2: put installer files back in place on boot partition and booted, reinstalled. Again keeping the pre-existing md0 boot and md1 root fs (but root emptied).  This time selected "initrd with all available drivers" (last time took minimal)

Post install in the installer system chrooted to the new installation (`chroot /target/`) and found out flash-kernel is not installed by installer, and /usr/share/flash-kernel/db was of course empty after manually installing flash-kernel. 

Copied this https://github.com/1000001101000/Debian_on_Buffalo/blob/master/Tools/buffalo_devices.db to /etc/flash-kernel/db and wget'd the correct dtb to /etc/flash-kernel/dtbs/ and now flash-kernel seems to do its thing.

Not rebooting and hoping for the best.

whcrg, 2022-03-01:
Nope. Still not booting far enough to offer ssh.

1000001101000, 2022-03-01:
I'm trying to get some other stuff of my plate so I can dig out my devices and test some of this. The ts3400 I've tested the installer with recently tells me that the installer isn't completely broken. 

the ls400 issue very much sounds like something could be missing from the initrd generated by the install, though all the boot-critical drivers are the same as the ts3400 (as far as I can tell) so that seems weird. There is an outstanding network hardware issue with some ls400s that I am working on a fix for, but if your device were affected it would be for all Debian versions. 

@whcrg I don't fully understand all the things you've tried but attempting installs without reformatting the /boot and roofs volumes will usually result in weird failures like what you are seeing. Please format both for your next attempt. The initrd options only come up if you choose the advanced options, that isn't necessarily a problem but it provides additional opportunities to  configure something incorrectly. If you stick to the normal install it should be reliable assuming partitioning/formatting are done correctly.

whcrg, 2022-03-01:
Should not matter if I reuse the filesystem and just rm -rf all files away, vs reformat it. I just don't want to touch working /boot  filesystem more than needed. The / raid volume I just reformatted.

Try 3: removed the other drive (this is on LD220DE) in case there is some problem with mounting and raid is not in sync etc. Booted straght to the installer so seems likely something weird was going on, as this should have had the boot stuff that gets into the answers ping but no ssh installation. 

With just one disk started installer system, mounted the raids and went in with chroot and checked both / and /boot have the expected stuff in place, and ran update-initramfs -u to make sure the boot stuff is in place. Reboot --> WORKS

So something was weird with installing to a box that had previously already a somewhat working debian system, something didn't go right place when both disks were in the box.

1000001101000, 2022-03-01:
@whcrg 
that makes sense. the bootloader on these systems are a bit of a pain in that way. They look at all the "partition 1" partitions (not the same as physically first one) they can find before deciding which to boot from. They often don't choose the one you would want/expect, and if they find certain things they can't handle they will fail to boot entirely (like if partition 1 is the whole drive etc). Your issue makes much more sense if you had a second drive with boot files on it that don't match the rest of the system, in that scenario it would boot the kernel and bring up the network in the initrd then fail to mount the rootfs and do things like start ssh.

whcrg, 2022-03-03:
Pribably the drives were out of sync and it was booting whatever and mounting randomly some wrong initrd. I redid the boot raid totally in other machine and now install worked nicely. Fully up and running with Bullseye, also seems to be totally stable even with high loads.

1000001101000, 2022-03-03:
Nice!

1000001101000, 2022-03-03:
One thing I recommend looking into is email alerts from MDADM, particularly for headless devices like this. If your array becomes degraded without you knowing you can suddenly find the device refusing to boot or doing odd things which are hard to troubleshoot at that point.

whcrg, 2022-03-03:
Sounds like very very good idea!

BrianBenninger, 2023-07-12:
Just wanted to throw my experience on here. Same model. Same issue.

I have completed this process previously on this device but had a drive failure so I am having to redo it.

I am able to move uImage.buffalo and initrd.buffalo to the /boot/ directory. After rebooting the device starts up and everything seems to be working. But I am unable to ssh in. There is no open ssh port when I check nmap.

I've tried with both Bullseye and Bookworm. No dice.

I am still able to use acp_commander and telnet into the device. Just no SSH.

1000001101000, 2023-07-12:
@BrianBenninger Your device sounds like it is booting into EM Mode. The /boot/ you are loading the new boot files into is a ramdisk that doesn’t persist through a reboot.

You might be able to mount the proper partition within EM mode and the  load the files and reboot. It depends on why it is in EM mode in the first place.