1000001101000/Debian_on_Buffalo #124: Possibility install on a flash drive？

sumsethan, 2022-01-14:
Hi，
I‘ve installed Debian on my TS3400D with your project’s help ,and it feels awsome!
seems like installation  on a flashdrive is not working ,the installation period is perfectly passed,
but the boot partition (seemslike) need to be FAT32 format to recognize by buffalo uboot.
is there any possibility to make it run on a flash drive plz？

1000001101000, 2022-01-14:
I’ve never tried that but don’t know why it wouldn’t work. Could you give more detail of how you set your partitions? 

It should be as easy as using the same partition you used to boot the installer as your /boot.

sumsethan, 2022-01-14:
nothing special acctually...

1.format a thumbdrive (of MBR partition table) with FAT32 ,put initrd.buffalo and uImage.buffalo in root folder;

2.put this thumbdrive on TS3400D and put another blank destination thumbdrive on TS too;
3.put the boot swith behind to "USB" position,then hit power botton;
4.ssh to installer;

5.use the partition guide to partition the blank thumbdrive ,and it was parted in 3 partations: /boot (ext2), / (ext4) ,swap  (of MBR partition table)

6.countinue installation and passed , remove the FAT32 thumbdrive, reboot;

(boot swith is still on "USB")

screen on TS3400D turn red and shows "can't find a boot file" (or something) 

that's it

any help?

sumsethan, 2022-01-14:
with this FAT32 thumbdrive ,installation to HDD is worked and bootable(turn swith to HDD boot after installation)...

1000001101000, 2022-01-14:
It does look like you’d need two usb drives:
boot from one like you already do
then on the partitioning screen tell it to use that as the /boot (keep as fat32). Put rootfs and swap on the other usb.

if the partitioner doesn’t like using a whole drive as /boot you may need to skip /boot at that step and then fix it in a shell at the end of the install but before the reboot

sumsethan, 2022-01-14:
thx! will try it later

and with your hint , maybe i can format  /boot partation as FAT32 on another computer , could that work?

1000001101000, 2022-01-14:
that shouldnt be needed really.

ideally the installer will just accept that drive as /boot, it should be okay for it to reformat as long as it’s still fat32. I don’t recall if i’ve ever done that with a non partitioned drive is all.

if it will do that in the menus then the resulting system will be configured to put the new boot files on that drive. If the menu can’t you’d have to do that manually.

All this is because the bootloader os hardcoded as the where too look for the boot files. It has to be in a non-paritioned fat32 volume for usb boot

sumsethan, 2022-01-14:
That explains why a multi-partation usbdrive cant boot even the first partation is FAT32 ....yeah
thx a lot 
i'll try  that

closed 2023-06-07 by 1000001101000