1000001101000/Debian_on_Buffalo #128: TS-XL boot somehow NOT from the 4 HDDs?

4920441, 2022-01-28:
Hi,

I got this device for free:
https://buffalonas.miraheze.org/wiki/Terastation_TS-XL

So I try to get debian running, but I'd like not to use the four HDDs for the system itself. 
It would be much nicer If I could flash a minimal debian in the NAND or on a seperate USB drive like I did with other devices such  as 
https://forum.doozan.com/

Is this somehow possible with the TS-XL?

It also seems it only boots if at least two disks are in the device - is that right? Seems a bit stupid to me... but hey, what do I know :-) Buffalo knows best..:)

Thanks a lot for your help + work.

Cheers

4920441

1000001101000, 2022-01-29:
Good to hear from you!

The 2 drive thing is only a limitation of the stock firmware installer, it can boot with a single drive without issues.

I doubt this device’s uboot can boot from USB though I’ve never actually accessed the uboot shell to try it.

TFTP boot is another option, but it is a pain.

NAND boot could work though I don’t have any experience doing that. If you look at the uboot environment you can see where the kernel/initrd/etc would need to be stored.

4920441, 2022-01-29:
Hmm... AFAIR you have to desolder a component to make the debug serial port on the front  bidirectional again. maybe I try that out. 
Do you have any idea on this device how the environment should be set that it could be able to boot from usb?

On my trusty old Zyxel NSA320 its like that - but for that there is a 'rather new' 2017ish uboot available, which makes things a lot easier...

`bootdev=usb
devices=usb ide mmc
init_usb=usb start
scan_disk=echo running scan_disk ...; scan_done=0; setenv scan_usb "usb start";  setenv scan_ide "ide reset";  setenv scan_mmc "mmc rescan"; for dev in $devices; do if test $scan_done -eq 0; then echo Scan device $dev; run scan_$dev; for disknum in $disks; do if test $scan_done -eq 0; then echo device $dev $disknum:1; if load $dev $disknum:1 $load_uimage_addr /boot/uImage 1; then scan_done=1; echo Found bootable drive on $dev $disknum; setenv device $disknum:1; setenv bootdev $dev; fi; fi; done; fi; done
scan_usb=usb start
uenv_init_devices=setenv init_usb "usb start";  setenv init_ide "ide reset";  setenv init_mmc "mmc rescan"; for devtype in $devices; do run init_$devtype; done;
usb_ready_retry=15
load_dtb=ext2load usb 0:1 0x1c00000 /boot/dts/kirkwood-nsa320.dtb
load_initrd=ext2load usb 0:1 0x1100000 /boot/uInitrd
load_uimage=ext2load usb 0:1 0x800000 /boot/uImage
usb_boot=run load_dtb; run load_uimage; if run load_initrd; then bootm 0x800000 0x1100000 0x1c00000; else bootm 0x800000 - 0x1c00000; fi
`

1000001101000, 2023-06-07:
Didn't notice I didn't really answer your last question. 

I know some of these older devices have uboot versions have USB support for their particular boards/SoC. I don't know for sure if that is the case for the TS-XL or not. You'd have to enable the bidirectional serial port and experiment.

closed 2023-06-07 by 1000001101000