1000001101000/Debian_on_Buffalo #131: Fan speeds inverted on LS220DE (kernel 5.10 and buster dtb)

whcrg, 2022-02-27:
Fan speeds are inverted on LS220DE (kernel 5.10 and buster dtb). When PWM 255 fan is stopped, when PWM 0 fan is full speed. Checking the DTS its seems this should be same with bullseys dtb.

Is there easy way to change this other way? It seems to me fancontrol does not have configurable option that would just invert the relation of temperature and pwm "strenght".

whcrg, 2022-03-03:
Now that I have frest install of Bullseye fan works correctly. So possible problem is likely only in Buster. Prob not worth the trouble any more as Bullseye is stable.

1000001101000, 2022-03-03:
hmm, even weirder. I'm fairly sure I tested all this with Buster's kernel since that was the one that changed how the polarity was handled and caused issues for those older devices. 

That does remind me that older versions of the DTBs did have the pins reversed so that the speed order was off-med-low-high rather than off-low-med-high, I don't recall when I fixed that (other than it was a long time ago but in the Buster era). That doesn't sound like what you were seeing though.

I also noticed that the speed map on these are ordered from high-to-low where a lot of the other devices in the kernel go low-to-high. Looking at the kernel source I didn't see anywhere that appeared to assume what order they are in (though my C skills are not the greatest). Likewise looking through kernel changes I didn't seem anything that looked like it would affect that behavior in a very long time. 

I'm terrible about taking notes when I'm testing so I'm not entirely sure what I all tested other than I remember spending a lot of time on the fan speeds after someone else noticed the off-med-low-high thing.

I think I'll leave this as an unsolved mystery for now, especially if it works correctly in Bullseye. I'm hoping that sometime between now and the Bookworm release I'll find the time to do some thorough hardware/DTB testing which includes taking another look at the fan stuff.

whcrg, 2022-03-03:
The non-linear order was in 2019, ancient times :) https://github.com/1000001101000/Debian_on_Buffalo/issues/4#issuecomment-455850729

Yeah I don't see big reason to use time for Buster specific testing. Bullseye works correctly for me at least.