1000001101000/Debian_on_Buffalo #133: Finsish Partioning return to partition loop

frakier, 2022-03-14:
Followed the instruction and all was fine until I told it to...
Finish partitioning and write changes to disk
[partitionloop.txt](https://github.com/1000001101000/Debian_on_Buffalo/files/8241069/partitionloop.txt)
then it asks, I say no, and it goes back anyway so I cannot complete the process.

1000001101000, 2022-03-14:
I have not run into that before. what model is this?

It sounds like the solution in probably to re-partition it using the installer if that's acceptable.

frakier, 2022-03-14:
lsd210d

1000001101000, 2022-03-14:
yeah, being the ls210 makes it a little harder to recover if it gets in an unbootable state since you'd then need TFTP for to crack the thing open to attach the drive to a PC. 

I'm a bit a surprised the stock partitions are coming up that way, I would think someone would have mentioned that before now.

Do you need any data on the device? If so we should probably switch gears to backing up any data you need before moving forward just in case.

I would expect you could delete the root/boot/swap partitions and re-create them with the installer and then be able to move forward. You just need to ensure that /boot is the first partition, relatively small (1G) and formatted EXT3. If it still complains you may need to wipe out the whole partition table, which you could probably still recover the data partition after but it would be a pain. I'd definitely make sure you have a backup of your data and a record of where the data partition starts before doing that. 

Whatever you do, don't reboot the device until we can get this sorted. You have a lot more options by using a second ssh session into the installer as long as it's still running.

frakier, 2022-03-14:
Data is backed up, just the pain of copying it back over from the original again if it gets wiped out. Yeah it is sitting in the ssh installer now.

1000001101000, 2022-03-14:
okay cool.

What I think I would do:

1. use the installer to delete all partitions but the big data partition at the end and have it make new, equivalant ones. 
2. set those up appropriately for /boot, swap and rootfs. 
3. see if that lets you go forward.

If that fails I would try to get the starting sector of the big data partition from the command line in  another session and write it down somewhere. Then repeat the above only having it start you with a new partition table. As long as you don't overwrite that part of the disk with new partitions you should be able to re-create the partition once you're in debian without loosing the filesystem.

frakier, 2022-03-14:
what is TFTP ?

1000001101000, 2022-03-14:
https://buffalonas.miraheze.org/wiki/Restoring_Stock_Firmware_via_TFTP

frakier, 2022-03-14:
yeah I read that on earlier but since running with a ip and password of course
`java -jar acp_commander.jar -t <device ip address> -pw <admin password> -o`
returns
...
Welcome to ACP Commander v0.6 (2021), the tool for Buffalo stock firmware control!
A SocketTimeoutException was thrown which usually indicates bad firewall settings.
Check that UDP port 22936 can be used by ACP Commander to send/receive and that your device is powered on and has booted successfully.
[ERROR] Exception: SocketTimeoutException (Receive timed out) [ACP Send/Receive (Packet:80A0 = SPECIAL_CMD_EN_ONECMD)]
...
I don't guess that will be happening.

frakier, 2022-03-14:
ok deleted the partition... recreated it and it is proceeding...so far... installing base 6%
reading up on this issue it seems to be related to the sector boundaries issue from older discs... this unit is brand new recently bough but must have old discs in it.. Buffalo is fast loosing my respect. When the power Macintosh came out I had a lot of their stuff and back then it was cutting edge and a step ahead of most others... now looks like they are passing off old "new stock" as new, should not have been surprised when I found out how old the software this unit was running was [8+years old].

frakier, 2022-03-14:
Worked... thanks for the help, was afraid to move forward when i was not sure if I would wipe out the wrong thing.
So there is something to add to the check list.. or as a workaround on the step by step instructions.
If it will not support the steps to make the partition changes and it is complaining about the /boot... go back... delete that partition [was #1 for me]... select the unused 1.0 GB and recreate that partition... proceed from there.

Now to see where I go from here.

closed 2022-03-14 by frakier