1000001101000/Debian_on_Buffalo #136: alt bootstrap dependency problems

frakier, 2022-03-18:
I copied the output to a text file...
in the folder next to the generate_disk_image.sh file
there now exists a img file debian_bullseye_armhf.img

is this a success or failure?

[termainl_output.txt](https://github.com/1000001101000/Debian_on_Buffalo/files/8307654/termainl_output.txt)

1000001101000, 2022-03-18:
looks like partitioning had problems, and kernel install definitely failed. 
Not successful.

frakier, 2022-03-18:
this is the sh I used...
[generate_disk_image.txt](https://github.com/1000001101000/Debian_on_Buffalo/files/8308050/generate_disk_image.txt)

1000001101000, 2022-03-18:
What os are you running this from?

it looks like your version of fdisk doesn’t like the input

frakier, 2022-03-18:
Release Linux Mint 20.2 Uma 64-bit
Kernel Linux 5.4.0-104-generic x86_64
MATE 1.24.0

looking at switching it over to debian so I have a reference when dealing with the debian on the ls210d. Just did waste more time just yet trying to get this thing back working.

1000001101000, 2022-03-18:
I’ve only tested from Debian Buster and Bullseye personally.

frakier, 2022-03-18:
Any advice for getting this done with a clean empty disk in the buffalo ls21d. I tried putting the buffalo software back on but when trying to install the updated firmware no matter what setting in the ini file and going into debug while in em mode will not let me do the update so I can start over and use the acp_commnader trick.

1000001101000, 2022-03-19:
If you get it to em mode you could partition/format/mount /boot and transfer over installer files using acp_commander.

frakier, 2022-03-19:
how? I can get it in em mode.
STEP: 1
set my pc to use 192.168.11.1
set up tftppd64 to share the folder with the two files
aset up the dhcp server to server the linkstation a 192.168.11.50
connect the ls210d directly to the computer in place of the internet cable
start up the linkstation... wait for seven red flashes and press the function button.
the files are downloaded from tftpd64 server to the ls210d
the ls210d reboots and I get a solid light and the NAS Navigator shows that it is online and in em-mode.
STEP: 2
setup the LSUpdater.ini Ver.1.78 add the flags and the special flags
[Flags]
VersionCheck = 0
NoFormatting = 0

[SpecialFlags]
Debug=1
save and run the ls updater
get into the debug section by clicking the very upper-left icon next to BUFFALO
in debug mode...
"Run complete format" is grayed out.
cannot proceed from there

1000001101000, 2022-03-19:
You should be able to start telnet with acp_commander ... I think. Form there you could use fdisk and mkfs.ext3

1000001101000, 2022-03-19:
I think `/usr/local/bin/create_partition.sh` is their partitioning script which you could kick off from the shell as well

frakier, 2022-03-19:
will look into it and report back tomorrow or late tonight, thanks

frakier, 2022-03-19:
nope... can start telnet... can see /usr/local/bin/create_partition.sh cannot do anything about it in telnet though
logged in as admin but do not have the permission to run it. cannot ssh in either.

frakier, 2022-03-19:
worked... did step 1 as above...
used acp commander to get into the machine
started telnet
logged in as root
ran /usr/local/bin/create_partition.sh
sync and exit not sure if that part was needed but did it anyway
added/updated to the lsupdated.ini
[Flags]
VersionCheck = 1
NoFormatting = 1
[SpecialFlags]
Debug=1
get into the debug section by clicking the very upper-left icon next to BUFFALO
in debug mode...
"Run complete format" is still grayed out.
selected force update and ok'd closing the debug and then update
let it do its thing and it is back to factory.

frakier, 2022-03-26:
solved... wondered about changing the title since it went in a different direction from the bootstrap...
something more like [solved] factory partitions from on a blank disk and install firmware ls610d
but closing... and taking notes so I do not forget.

closed 2022-03-26 by frakier