1000001101000/Debian_on_Buffalo #142: LS441d restore the backed up uImage and initrd files

sinankizar, 2022-04-17:
I am on LS441d and i have moved the uImage and initrd files to /boot. Rebooted the device and ssh with installer use and installation has begun. At some point the installation failed and I gave up to try the debian this time. I have backed up original  /boot/initrd and uImage files  as uImage.buffalo.bkp and initrd.buffalo.bak but couldn't manage the restore the files. Whenever the device reboots it reboots into installer. How can I restore the original boot files? Where are those files now and how can I access them to rename the oeiginal files so device can boot into original firmware and settings? If it is possible of course 🤣 

The installation fails to create the user btw, whenever installation comes to user creation it stops and returns to installation options menu.

1000001101000, 2022-04-17:
Good to hear from you!

If it’s booting back into the installer the formatting options for /boot aren’t being set properly. Normally that volume is reformatted at the beginning of the install which wipes out the installer files.

usually the setting users/passwords if part if the last step after reboot, i’m not familiar with how you would get to any menu past that point. Are you using the advanced menu? Could you post a screenshot of what it looks like at that point?

I’ve run through the Bullseye installer on my LS441 as recently as 5 days ago without issue. I’m fairly confident the basic logic works under the cases I’ve tried and am curious what is different in your case. Let me know if there any additional details you can provide, particularly if the differ from the wiki instructions or youtube video.

If you’ve still got the installer working you can use it to access a shell and put the stick boot files back in place to try to get to EM mode. This mode also has a NAND recovery mode if needed.

sinankizar, 2022-04-17:
I have aborted the installation process and the device shut down. After this i have pinged the device and ssh'd into it with the installer user and pass. Now I have to install the debian and no other option is available for me i think. I want to cancel the installation and get back to default firmware. Is there an option to get back to factory firmware without losing data at this point or do I have to continue the debian installation to get back my data?

Thank you for your fast support btw :)

1000001101000, 2022-04-18:
Huh, I assumed you were farther than that…have you not actually made it to the partitioning menu?

any chance you could reboot and record a video of the steps you’re taking to get to this state?

If you haven’t actually gotten to the partitioning screen and as such haven’t reformatted anything it might be possible to get back to stock firmware

sinankizar, 2022-04-18:
Strange things happen 🤣 
I have to go to work now but before I go I wanted to share with you a screenshot and everything worked well until the partitioning menu. Now I just want to back up my data on the array and after that I want to try debian on buffalo. And the answer is I didn't format anything. How can I get back to stock from this point?

Thank you

[image]

sinankizar, 2022-04-18:
So does it look like it is possible to revert back to stock from this point?

1000001101000, 2022-04-18:
If you haven’t formatted anything you should be able to:
1. Use the menu to start a shell
2. mount the md device containing /boot
3. Swap the stock *.buffalo devices back in place.
4. Unmount and reboot

sinankizar, 2022-04-18:
You lost me quickly 🤣 
1. Started.
2. where is this boot and how to find it?
3. this one I can :)

1000001101000, 2022-04-18:
what do you get if you run:
ls /sys/block

sinankizar, 2022-04-18:
md0        md1        md11       md12       md2        mtdblock0  sda        sdb        sdc        sdd

1000001101000, 2022-04-18:
Try “mount /dev/md0 /mnt”
then look in /mnt

sinankizar, 2022-04-18:
Tried and got this:

mount: mounting /dev/md0 on /mnt failed: Device or resource busy

1000001101000, 2022-04-18:
maybe.

the other thing you can do is checkout each md device with “mdadm —detail” and make sure you’ve got the one containing sda1

sinankizar, 2022-04-19:
Thanks for everything my friend. Finally I have solved this with your help. I will try this with an empty drive and after that move over to your image. Thanks a billion aagain.

1000001101000, 2022-04-19:
Cheers!