1000001101000/Debian_on_Buffalo #143: Fan kicks in regularly on Terastation

mholl1983, 2022-04-25:
Hi, I thought I'd open a fresh thread for this new issue. :) Thanks as always for your help!

My fan seems to be kicking in a little more than usual lately. I haven't done much on the box aside from installing new security updates. The only new module I installed that I'm aware of is minidlna, which should be fairly lightweight.

My micon_fan.conf file shows the following values: 

```
[Rackmount]
mediumtempc = 40
hightempc = 50
shutdowntempc = 75

[Desktop]
mediumtempc = 25
hightempc = 35
shutdowntempc = 75
```

Is it worth tweaking these values? Or would it be better to figure out (via logs, perhaps?) what is using more CPU and thus why the fan is kicking in? 

I do have Webmin installed, as you probably know, but I find I'm using it less and less as I get more comfortable with the CLI. I could try removing that altogether, perhaps. Anyways, any other suggestions are appreciated, thanks!

mholl1983, 2022-04-25:
Some `top` output, in case that helps too:

```
  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
12758 root      20   0   11352   3144   2608 R   0.2   1.2   0:00.30 top
  313 root      20   0    3236   1704   1496 S   0.1   0.7   0:17.49 mdadm
  384 root      20   0   16404   5616   4648 S   0.1   2.2   0:06.31 micon_fan_daemo
  440 ntpsec    rt   0   16612  14832   5700 S   0.1   5.9   0:44.45 ntpd
  477 proftpd   20   0   17756   1356   1232 S   0.1   0.5   0:06.79 proftpd
12552 marc      20   0   15732   6668   5468 S   0.1   2.6   0:00.42 sshd
    1 root      20   0   36192   7784   5772 S   0.0   3.1   0:15.67 systemd
```

1000001101000, 2022-04-26:
I would probably adjust the threshold, I made them pretty conservative. 

For a NAS cpu temperature isn't always the best metric to track. Ideally it would include HDD temperatures in the calculation but drives aren't very consistent about what they report for their SMART temperature.

mholl1983, 2022-04-26:
Thank you! Whatever process was making the fans kick in seems to have settled, but I'll play with the settings if the issue comes back.

A funny thing I noticed, though: the display on the box now says Bookworm! Did I inadvertently upgrade to the next Debian release? Seems to be working fine so far! I guess if an upgrade was going on, that could've been the culprit for the fan spikes.

1000001101000, 2022-04-26:
The startup service grabs the version from /etc/debian-release, if it changed you must have done a dist-upgrade with webmin or something.

mholl1983, 2022-04-27:
Hi, no, I didn't do any upgrades with webmin. It may've happened via command line, though. So far, not observing any issues other than when the fans kicked in more for a brief time, so I'll leave things as is for now and report back if anything changes. Thanks again!

1000001101000, 2022-04-27:
I'm not worried if you're not.

doing a dist-upgrade is a multi-step process, I doubt you could do it by accident unless something did some of the steps on your behalf ahead of time.

I seem to remember OMV doing something odd like adding packages from testing or something when I played around with it 5 years ago or whatnot. 

Usually for that to happen you need:

- Something to add the testing/bookworm repo(s) to your sources.list
- A command that will actually install packages from a different version such as apt dist-upgrade or apt -t <repo version>

The Debian version in that file gets set by some package (base-files?) it's possible that some thing you installed (like webmin) could have added testing/bookworm repo and then packages from it got installed to fulfill some dependency somewhere. I keep saying webmin just because I know some of its operations install packages in the background. 

It's probably not a big deal either way. I personally don't usually run Debian Testing because the pace of updates gets annoying but many people do and prefer it over Stable for many applications. 

It might be worth considering running through a dist-upgrade to ensure you're all the way on Bookworm and running the kernel the bookworm packages are intended for.

mholl1983, 2022-04-27:
Thank you for the helpful reply! The only other thing that comes to mind is that I set up unattended upgrades to keep security patches up to date. I thought I commented out the other types of upgrades, but let me double check.

closed 2022-07-21 by mholl1983