1000001101000/Debian_on_Buffalo #144: LS420D Copied files successfully using ACP Commander, but now won't boot, ping is unreachable

denismoc, 2022-05-06:
Buffalo NAS LS420D

I followed option 1 and successfully copied the files and rebooted using acp_commander. 
Now the device is suck during boot up, doesn't connect to the network and I can no longer access it to complete the installation. 
The buttons and power switch do nothing either, to power it off I need to plug it out.
Any tips advice on reseting this back or gaining serial access would be appreciated.

1000001101000, 2022-05-10:
that means it's booting into something. I would start by scanning your network for devices that have port 22 open. Sometimes the DHCP server gives the device a new IP address because the client ID changes even if the MAC address doesn't. There are also some edge cases where the MAC address doesn't get set properly and reverts to a random one which can have a similar effect.

denismoc, 2022-05-16:
So I've had no luck with this at all. I've checked my router and it does say that that there is a device on the IP address but it is unreachable. 
All my other devices show what Eth port they are connected to but for this it just says other and the device name is unknown.
Is there any way of forcing access or connecting directly to my laptop instead of over the network?

denismoc, 2022-05-16:
I've accounted for all other IP addresses on the network too

1000001101000, 2022-05-17:
another possibility is that this device is one of the hardware revisions that suffers from an ethernet hardware issue:
https://openwrt.org/toh/buffalo/ls421de#ethernet_port

If that is the case, one way around that would be to connect a USB NIC and then we could do some tests over that connection. 

I've been testing a new fix for this issue but either my fix doesn't work or the issue is different enough with the LS441 that someone sent me that it needs a different solution. 

I'd be willing to trade for my LS420 that does not have the ethernet problem if you;re interested. I fear I'm going to need more than just this LS441 to develop a solution which works for these devices without re-flashing the bootloader.

denismoc, 2022-06-02:
Sorry I was offline for the past two weeks.
I managed to find some documentation to restore a bricked Buffalo NAS that helped return it to factory settings.
I'll investigate their proposed solution with the different bootloader and see if that works. 

I'll consider swapping it but it might be more worth my time just scrapping it and getting a newer model, maybe with more bays available too!

denismoc, 2022-06-02:
Ha no you're very much out of luck, I'm in Ireland
I'm sure the shipping costs will not make it worth it at all for you!

1000001101000, 2022-06-02:
Thanks for confirming!

sillypig670, 2022-07-21:
Buffalo NAS LS410D shows the same.
I've try all 3 options to flash the image file successfully, but when it boot up with 30-60 seconds white LED blink and turn to a solid white, the ethernet will shutdown, seems the ethernet hardware issue you've mentioned.

1000001101000, 2022-07-21:
@sillypig670 are you in the US? 

I'm at a point now where I think I can develop a permanent fix for this issue for at least the LS410/420/421 and more importantly port that fix to the LS441 which has frustrated me for almost 3 years at this point.

I'm willing to work out a trade for an non-affected device etc provided the shipping isn't to expensive.

sillypig670, 2022-07-22:
@1000001101000 Sorry I'm not in the US.
Although I can give u my LS410D for free to try solving that issue willingly, whether postage or time cost too much to have it shipped from China to the US.

denismoc, 2022-07-25:
> 
[Buffalo NAS LS420D LS420 recovery.pdf](https://github.com/1000001101000/Debian_on_Buffalo/files/9183783/Buffalo.NAS.LS420D.LS420.recovery.pdf)

I can't seem to find it online anymore but here's a copy of it. The links in it should still be working though.

denismoc, 2022-07-25:
@smc87 I can't remeber how I found it online anymore but here's a copy of it. The link to the software should still be working in it though
[Buffalo NAS LS420D LS420 recovery.pdf](https://github.com/1000001101000/Debian_on_Buffalo/files/9183798/Buffalo.NAS.LS420D.LS420.recovery.pdf)
.

1000001101000, 2022-07-26:
If you're looking to restore stock firmware there are guides on this wiki that don't require any special software:
https://buffalonas.miraheze.org/wiki/Restoring_Stock_Firmware_without_TFTP

@sillypig670 Do you know how much postage would cost?

closed 2023-06-07 by 1000001101000