1000001101000/Debian_on_Buffalo #145: Bullseye for ls-wvl boot loop

boudicca99, 2022-06-04:
just seen the new bullseye images for ls-wvl etc. usual process stock, connect with acp drop the uImage.buffalo and initrd.buffalo into /boot. Reboot and it never recovers, just boot loops. Have to go back to stock to recover. Cheers Shaz...

boudicca99, 2022-06-07:
one thing I have noticed is the size of initrd.buffalo, its consistently 7.3Mb for Stretch and Buster but only 5.9Mb for Bullseye... is there an issue there perhaps with the size ?

1000001101000, 2022-06-07:
ah okay, so you have succeeded with buster/etc before. probably means you're past the easy stuff though it never hurts to check.

The size difference with my initrd's is intentional..... and a bit of a long story. A few of the oldest devices I support have a hard 7MB limit for initrd, the bootloader actually segfaults if it tries to load a bigger one....no idea why. As such I put more effort into keeping it small than Debian's equivalent. I achieved a lot of that in the latest update by moving a bunch of drivers into the installer kernel rather than as modules that end up in the initrd.

Once you're up on Buster there are a few things we could look at which might help, A big one is kernel modules. If somehow that model relies on a device driver that the other devices I tested didn't I might just need to add it to my kernel build for the installer. we could also play with combinations of DTB and kernel image, maybe the one I'm using needs to be updated/recompiled for the current kernel in some way.

boudicca99, 2022-06-11:
tried it...buster-bullseye then change the initrd and uImage, got all the way to bullseye running without the temp sensor on 5.10.120-1.   switched in the initrd and uImage for wvl and it returns to boot looping.  Switch only the initrd, and brings the interface part up but we never get to ping or dhcp nor ssh.  revert back and all boots fine....same as my previous wvl and qvl no we have Bullseye but have lost the the temp sensor.....such is life! Thanks I will look to possibly retiring them eventually. Many Ta's

deercg, 2022-06-18:
I've same this issues with ls-wvl too. Back to this version is work for me: https://github.com/1000001101000/Debian_on_Buffalo/tree/34e81e2450ec2ac5456471b7f882c71404eda965. While newest version get boot loop. 

That version may work but have problem with MariaDB. So i tried to update to newest version today and that boot loop happen.

1000001101000, 2022-06-20:
Someone had a similar ssue on a device that I have available for testing. It turns out the latest set of Bullseye images that were generated a few weeks ago had a broken kernel.

Corrected issues have now been generated/posted. There's a good chance this will fix the issue you've encountered as well. Let me know if your get a chance to try them out.

boudicca99, 2022-06-21:
cheers....for that...  I will return to the wvl at the next opportunity....currently trying to get a Netgear nv+ v2 to debian so fun fun fun.

shampeon, 2023-04-23:
I've also been unable to use your LS-WVL Bullseye images on my LS-WV2.0TL.

At one point I got the installer working, but it kept failing at the package level, throwing errors about not being able to find any mirrors regardless of the setting. There was some error about an apt or dpkg process already running when I logged in on the console.

I had to abort and tried it again, but it just went into emergency mode and wouldn't boot anything. I've now tried using your Buster installer (wouldn't boot, went into emergency mode), using deboostrap for Bullseye (wouldn't boot, went into emergency mode), trying to restore the original firmware using the updater by reformatting the drive (the partitions all seemed be there, and it gets recognized as an md device on my Ubuntu server, but putting the disks back in the LS-WV2.0TL just caused a boot loop after the firmware reloaded: blinking blue boot light, restart, repeat).

I'm now in the middle of trying to get a debootstrap image for Buster (I had to remove the bsdextrautils package from the script to generate the image). Not sure how much more time I should devote to this, as it's eaten up a good day plus. Just very annoying I can't ANY image to boot now.