1000001101000/Debian_on_Buffalo #146: Can TS-WXL be Debian?

mospead, 2022-06-19:
Thanks to this git, I think I can find the way to install debian to my TS-WXL.  But I can not find appropriate  UImage and initrd in here.  Could you please instruct me how to install debian on TS-WXL.  
My firmware is already v.1.71 or 1.76.  Is that a critical problem? 
I appreciate for this great work still going and thanks to read my poor english.

1000001101000, 2022-06-19:
Good to hear from you!

The TS-XL uImage is used for the TS-WXL and TS-RXL as well. 

The firmware version doesn’t matter when installing debian on this model.

mospead, 2022-06-19:
Thanks for your quick reply.  This git is one of the most valuable place for anyone want to customize buffalo NAS.  
I think I've tried TS-XL uImage for TS-WXL, and went boot-loop.  But it was my try and can be made some mistake.  I'll try again with care.   Thanks to advice and great works!

1000001101000, 2022-06-19:
The most common reason for a boot loop is incorrect or corrupt boot files. Often folks download the html github page rather than the actual files by accident. The correct files will be around 2-3MB and 6-7MB.

The serial port on the front of the device shows the console output. If you get stuck that output could help find the issue.

mospead, 2022-06-19:
I failed again.  Files seems to be OK.  File size are 2,051KB and 5,776KB for uImage and initrd.  
I am using bootable HDD made by stock firmware, connected to another Linux USB and deleted all files in boot partition.  Then placed two files(uImage.buffalo and initrd.buffalo) and ..... went boot-loop.  Should I place another files in boot partition or set particular permission to files?  Or maybe completely doing strange thing?

1000001101000, 2022-06-19:
If you show the full output of what you are doing I might see where the problem is. It isn’t obvious to me by your description.

most of the issues I can think of (ext3 issue like inode size, broken raid1, other drives with old boot files etc) would result in either a boot error or booting to em mode. 

Boot looping generally happens when it successfully loads boot files but then fails to successfully boot from them. Another reason this could happen would be using the wrong uImage.buffalo file.

could you post the filenames and md5 of the files you’re trying as well as the full model name or a picture of the back of the device?

mospead, 2022-06-20:
I've tried again and failed as bellow;
TS-WXL HDD is sdb attached to another debian(LS-GL) by USB.  

root@debian:~# mdadm --assemble --run /dev/md0 /dev/sdb1
mdadm: /dev/md0 has been started with 1 drive (out of 2).
root@debian:~# mount /dev/md0 /mnt/tsboot/
root@debian:~# ls -l /mnt/tsboot/
合計 18732
-rw-r--r-- 1 root root   46164  6月 13 18:09 conf_save.tgz
-rw-r--r-- 1 root root 9296984  6月 20  2017 initrd.buffalo
drwx------ 2 root root    4096 11月  1  2007 lost+found
-rw-r--r-- 1 root root   20989  6月 13 18:09 splx.tgz
-rw-r--r-- 1 root root  404368  6月 20  2017 u-boot-88f6281.buffalo
-rw-r--r-- 1 root root  480980  6月 20  2017 u-boot-mv78100.buffalo
-rw-r--r-- 1 root root  480980  6月 20  2017 u-boot.buffalo
-rw-r--r-- 1 root root 2738676  6月 20  2017 uImage-88f6281.buffalo
-rw-r--r-- 1 root root 2826004  6月 20  2017 uImage-mv78100.buffalo
-rw-r--r-- 1 root root 2826004  6月 20  2017 uImage.buffalo
root@debian:~# rm -R /mnt/tsboot/
rm: '/mnt/tsboot/' を削除できません: デバイスもしくはリソースがビジー状態です
root@debian:~# ls /mnt/tsboot
root@debian:~# cp /mnt/debian_share/*.buffalo /mnt/tsboot/
root@debian:~# ls -l /mnt/tsboot
合計 7844
-rwxr--r-- 1 root root 5913872  6月 20 17:53 initrd.buffalo
-rwxr--r-- 1 root root 2099952  6月 20 17:53 uImage.buffalo
root@debian:~# cd /mnt/tsboot/
root@debian:/mnt/tsboot# md5sum uImage.buffalo initrd.buffalo
bc185f2b7daa134d59d5ad3a03dff431  uImage.buffalo
bb7ed2a32745aac70a9b77c2fae90463  initrd.buffalo

[image]
[image]

thank you,

1000001101000, 2022-06-20:
New installer images have been posted which I tested on my TS-WXL. Please give them a try and let me know how it goes.

mospead, 2022-06-20:
Thank you for dealing with this issue so quickly.  I'll give them a try and let you know how it goes!!

mospead, 2022-06-20:
Debian installer seems to be booting up, display shows 'Debian Installer'!!  But SSH connection by installer@ipaddress is refused.  Is there a way to proceed installation?

mospead, 2022-06-21:
Great!!  Install is completed.  But SSH can't be connected as root.  Can you give me a way? or Must I buy a serial cable?

mospead, 2022-06-21:
Ah.  Should I attached HDD to another debian and edit /etc/ssh/sshd_config?

mospead, 2022-06-21:
Yes!  I get rooted using SSH in Bullseys on my TS-WXL.  Your script seems to be perfect!!!  I really appreciate your kind help and excellent works.

1000001101000, 2022-06-24:
Cheers!

closed 2022-06-24 by 1000001101000

mospead, 2022-10-11:
Thanks for kind reply.  I'll try again, and take screenshots and
informations you said.  Unfortunately I don't have a serial cable so could
not see the information during a boot loop.