1000001101000/Debian_on_Buffalo #147: Buster / Bullsey on LS220D

smc87, 2022-07-05:
Hello, I have a few old devices around that I am trying to make the best use of.

I've tried a few different ways to get the ls220d to boot with any of the images here but each time it gets itself into a state where there is a solid white light on the front and fans are running at 100%.
The NAS does not respond to the power switch and needs to be unplugged to stop.

I can return it to a good state by TFTP-ing a default image back onto the NAS and pushing default firmware.

If I use acp commander or TFTP to put an image in place from here as soon as a reboot I go back into the above state.

Any ideas?

1000001101000, 2022-07-05:
Good to hear from you!

“solid white light on the front and fans are running at 100%” Describes the expected state of the device if it successfully boots into the installer. Likewise, within the installer the power switch is not configured to do anything.

Can you provide some details of how you are trying to connect to the device and what happens when you do?

smc87, 2022-07-06:
Thanks for the response.
Well, thats good to know im somewhat on the right track, sorry for my ignorance with these devices.

When I get the device into the above mentioned state I am never able to connect via putty - "Network error: connection timedout"

And ping shows: Destination host unreachable

smc87, 2022-07-06:
Aha, DHCP was throwing me to a different IP.

Thanks for your help

1000001101000, 2022-07-06:
Excellent, Good work!