1000001101000/Debian_on_Buffalo #150: TS-XEL hangs at boot after I tried to add USB drive

mholl1983, 2022-09-11:
Me again. :)  Hope you're well.

All's been working smoothly, but I wanted to add a 4 TB USB drive into the mix so I could directly backup files from my RAID array to the USB drives on the TS-side. 

Here are the steps I took before things started hanging after a reboot:
1) Plug in USB drive
2) Make a directory for mounting
3) Add entry to fstab file so it would auto-mount upon boot
4) Reboot TS. Result: hanging on boot with "no init file at hdd2" message

I unplugged the USB drive, shut everything down, and tried to boot cold in the hope that that cleared things up, but no dice. 

4 HDDs in the box seem to have good health, at least when I last checked, and they're all seated properly/being detected. The USB drive experiment is the only new thing since this issue arose.

Any suggestions welcomed!

mholl1983, 2022-09-11:
Previously, when I was setting up the device with Debian, I did get a "no init file" error for a good 10 min or so, but then eventually the boot process would continue. Now, it seems to be taking a lot longer, so I'm worried I buggered something up. :(

mholl1983, 2022-09-11:
Just an update that the TS is getting the assigned IP address and I can ping it, but I'm unable to SSH in with Putty, get files from my Samba share, or anything else.

1000001101000, 2022-09-12:
sounds like a bad fstab entry.

for that model it’s easier than most to access the serial console and fix it via maintenance mode or whatever debian calls it. I think I have some details on the buffalonas wiki.

otherwise you could connect all the drives to a pc and edit the fstab that way… though it’s a pain with 4 drives.

mholl1983, 2022-09-12:
Yes, I suspect the fstab needs to be fixed after my blunder. I hope that's all it is, in any case. 

Serial console sounds intriguing and I do see the connector for that. Not clear on what cables/converters I need, though. Any pointers? And is any soldering involved?

mholl1983, 2022-09-12:
Based on the limited info on various sources online, looks like the Buffalo devices use 3.3V TTL and the TS-XEL has a 4 pin connector. Therefore, I can probably safely assume [this cable](https://www.amazon.ca/DTECH-Serial-Adapter-Prolific-Windows/dp/B08G1JTN4N/ref=cm_cr_arp_d_product_top?ie=UTF8) is a good bet?

1000001101000, 2022-09-13:
That should work

mholl1983, 2022-09-17:
Hi, I just got the cable! However, I want to make sure I connect the correct wires to the correct inputs and not cause more issues! I didn't see any obvious indicators on the TS port side itself, so I wanted to double check with you. 

Here are pictures of the cable itself (separate wire side) and the schematics in the manual, respectively.

[image]

[image]

Let me know if I should be attaching these in any particular order, and then (fingers crossed) I'll be able to fix up my fstab file. Thanks again!

1000001101000, 2022-09-17:
I’ve got the pin order on the wiki:Terastation TS-XELbuffalonas.miraheze.orgDon’t connect the 3.3v line, just rx/tx/gnd.

mholl1983, 2022-09-17:
Thanks!

-TX -

-RX -

-3V -

-GND-

So, just to be 100% clear, this is top to bottom facing the front of the TS?

1000001101000, 2022-09-17:
Yeah

mholl1983, 2022-09-20:
Good news! After more tinkering, I was able to get a proper serial connection to my TS-XEL.

I used the following settings on my UART->USB cable:

[image]

[image]

The underlined setting in the second screenshot seemed to make all the different. Prior to checking this, I was getting gibberish in the output. After checking the setting, I was able to get the correct output. Here's the TS in powered off/plugged in state:

[image]

I'll report back shortly when I'm able to get to the fstab file. Thank you again for all your assistance here and in IRC!

1000001101000, 2022-09-20:
cheers!

closed 2022-09-20 by mholl1983