1000001101000/Debian_on_Buffalo #151: LS-x2.0TL

der-ali, 2022-09-18:
Thanks a lot for this amazing work :)
I just want to ask which installer image i could use for LS-x2.0TL version?

der-ali, 2022-09-18:
ok, i used the script `generate_disk_image.sh` and generated an image, but i am not sure what to do next. How can i boot my buffalo station using this image?

1000001101000, 2022-09-18:
What did the LEDs do during the boot process?

der-ali, 2022-09-20:
Hi @1000001101000 , sorry i haven't got notified about your comment.
The LED is blinking all the time.

1000001101000, 2022-09-20:
I took a look at the device-tree and the user manual for this one. There appears to be less feedback with this device which will make it a bit harder to determine what's going on. 

It looks like I didn't add anything to make the LED turn solid when it boots, I don't recall if the device did that on it's own or if I just forgot.

I think the device would still show an error code with the LED if it couldn't boot from the partition or boot files, but it's not clear since it has far less LEDs/codes.

A few things you can take a look at are:

- look at your routers dhcp/arp entries to see if the device appears to have an IP address (possibly a different one than expected)
- connect the drive to a linux pc and open it in gdisk to confirm it only shows an MBR and no GPT
- take a look at the contents of the boot partition and confirm the *.buffalo files are there.

der-ali, 2022-09-20:
@1000001101000 Thanks again for helping :) 
point 1:
I checked the arp entries, the device doesn't appear at all.

point 2:
it shows MBR
[image]

point 3:
I checked using gparted the partitions, but the partitions has no moint point for file system

[image]

What could here gone wrong? I used dd to write the image to the disk.

1000001101000, 2022-09-20:
That all looks normal, the partitions won't have any mount points from the PC's point of view.

One other thing you can do is manually mount that /dev/sda1 and take a look at the files in there. Specifically whether the two *.buffalo files are present and have reasonable sizes.

der-ali, 2022-09-20:
I needed to create a file system first on the partitions to mount hem. I used `mkfs.ext4`.
All the partitions are empty though. I mounted the `debian_bullseye_armel.img` and it contains 3 volumes, one which includes the os and other which includes the boot system (It contains *.buffalo files) and an 4 GB empty volume.
Should i format the disk and retry writing the image to it ?

1000001101000, 2022-09-20:
Yes, I'd suggest writing the image to the disk again.

You should see the same volumes on the resulting disk as you can see inside the disk image.

der-ali, 2022-09-20:
Really weird! 
I retried using the ubuntu disk tool to write the img to the disk. The partitions look better (They have a file system), but still empty when mounting them.

der-ali, 2022-09-20:
@1000001101000 
I tried copying the files from the image volumes to the disk partitions mount point. I am still unable to discover the device in the network :/

1000001101000, 2022-09-20:
The process should create a disk image that contains:

- A boot partition formatted as ext3
- a swap partition
- a rootfs partition formatted as ext4

DD'ing the image to a disk should create those on the disk. Since you've seen them on the disk image that part sounds like it's working. Could you share the full DD command that you are using?

der-ali, 2022-09-20:
```
sudo dd if=/path/to/debian_bullseye_armel.img of=/dev/sda bs=4k
```

der-ali, 2022-09-20:
u're right. the image consists of the above mentioned partitions
[image]
When ii dd the image to the disk i get different partitions
[image]
Not sure why! i am trying to figure that out

der-ali, 2022-09-20:
Funny, when i dd the image to a usb stick, it works, i get the same partitions as in the image. I will try reboot the pc and try again.

der-ali, 2022-09-20:
Still getting the above partitions on the when i `dd` to the disk, I can't explain why.

der-ali, 2022-09-20:
I did the same steps and still getting the same results.
I compared the first kbyte in  hexdump  of the image with the disk.
They differ:
[image]

der-ali, 2022-09-20:
@1000001101000 
I am using now an ssd instead of the hard disk. I was able to write the same partitions from the image to it.
I am unable to discover the NAS in the network thogu.