1000001101000/Debian_on_Buffalo #153: Buster on LS220D only reboots back into installer? Did I miss a step?

mandeldebugger, 2022-10-13:
Hello there! What a fantastic project. I feel like I'm retaking ownership of my ageing Buffalo hardware!

The installer worked fine and it downloaded the additional installation files and rebooted. But now I can only get back into the NAS with the installer credentials:

`ssh installer@192.168.1.97`

The `root` and `nas-user` accounts I set up during the install don't appear to be accesible:
```
BusyBox v1.30.1 (Debian 1:1.30.1-6+b3) built-in shell (ash)
Enter 'help' for a list of built-in commands.
~ # cat /etc/passwd
root:*:0:0:root:/:/bin/sh
installer:x:0:0:installer:/:/bin/network-console
sshd:*:100:65534::/run/sshd:/bin/false
~ # 
```

Should I have delete the installer image from `/boot`? I couldn't find anything in the instructions mentioning this, so I assumed the installer script removes the installer boot image or renames it before rebooting into the newly installed Linux kernel. That can't be the case though because there's no `/boot` folder visible from the installer kernel.

1000001101000, 2022-10-13:
Good to hear from you!

It sounds like you didn't set up /boot properly in the partitioning menu. For the new boot files to be stored to the proper place /boot must:

From https://github.com/1000001101000/Debian_on_Buffalo/wiki/Running-the-installer:

- located on the first partition of the disk(s)
- either a regular partition or RAID1 array
- formatted as ext3

If you're not sure if you have the configuration correct you can post a screenshot of your partition screen and I can take a look before you commit to the next install attempt.

mandeldebugger, 2022-10-13:
Hi. Ah okay. That's odd. I used the newbie guided partitioning option, which said it would set up the boot partition and then set the /home folder in the largest contiguous partition. I'll try again.

1000001101000, 2022-10-13:
If you used any auto/guided partitioning options it will do some goofy stuff on your behalf. For these devices you need to manually specify things at this step.

That #9 being used for swap may be a problem, I would change it to "do not use".

if you set "RAID1 device #0 - 1.0 GB" as /boot and "RAID1 device #1 - 5.1 GB" as / you should be good to go. 
/boot needs to be ext3
/ can be whatever you want, use ext4 if you're not sure.

mandeldebugger, 2022-10-13:
Right. That makes sense and ... yes, that worked. 

Many thanks.

I think the takeaway is "don't use the guided (newbie) partitioning option" because it won't correctly setup the `/` and `/boot` partitions before proceeding with the installation.

closed 2022-10-13 by mandeldebugger

1000001101000, 2022-10-13:
I think I concluded that I wasn't able to disable those options easily, if I remember I'll look again when I work on the bookworm installer.