1000001101000/Debian_on_Buffalo #154: Tried installing fancontrol onto a new LS220D Bullseye installation. It didn't go well.

mandeldebugger, 2022-10-15:
Everything was going so well!

After noticing the fan being noisy at constant 100%, I tried installing `apt-get install fancontrol` in line with the instructions [here](https://github.com/1000001101000/Debian_on_Buffalo/wiki/Post-Installation-Options#Fans). After package download, the installation made the CPU top out at 100% and stay there for two hours or so. I rebooted but it never recovered. Powered it down and up again a few times, no ping responses, no OMV landing page. It appears to have left this world.

I'm going to see if I can `java -jar acp-commander -t 192.168.1.97 -s` back in, although I'm not brimming with confidence because there's no ping response. Any suggestions welcome.

1000001101000, 2022-10-17:
Good to hear from you!

I don’t have a specific theory about what happened. My limited experience with OMV involved it messing with network and disk configuration stuff in ways that broke the underlying OS on the next boot. 

If the system was still online there are some things we could do to investigate the 100% cpu issue, I wouldn’t expect fancontrol to be the cause but also can’t be sure.

You could always connect the drives to a PC and look at logs/etc to try to figure out the startup issue. Failing that you could load the installer files again and start over with a fresh install.

mandeldebugger, 2022-10-20:
I post-mortemed the disk and noticed that the files in the `/boot` partition were replaced presumably by `apt-get dist-upgrade` or by something updating the boot image. Whatever it was made the drive unbootable so I won't be running that again!

I couldn't get it to boot so wiped the disk and gave it an NTFS partition for the time being and re-docked it as a new disk. Currently, `Nas Navigator 2` can't see it, `LSUpdater.exe` can't see it, no ping response, no IP, seven red flashes on the power LED.

It is actually bricked unfortunately. I'm not sure where to go from here.

mandeldebugger, 2022-10-20:
> My limited experience with OMV involved it messing with network and disk configuration stuff in ways that broke the underlying OS on the next boot.

Can you elaborate on this? That has to be the issue - if any OMV update process replaces files in the `/boot` directory (as I think I've observed), it's worth pointing out the culprit.

mandeldebugger, 2022-10-20:
The boot images bad been moved to `.bak` extensions (like `uImage.bak` etc) and newer versions had been put in.

I might need to go back and re-run some update things and see what breaks.

1000001101000, 2022-10-20:
That is normal behavior for kernel updates or anything that generates a new initrd, on it's own it doesn't indicate an issue.

What version are you dist-upgrade-ing from/to?

mandeldebugger, 2022-10-20:
> What version are you dist-upgrade-ing from/to?

Yes, agreed, it should be normal behaviour, but wasn't in this case. I'm running the reinstall of the OS, then I'm going to do a vanilla `apt dist-update` and `apt dist-upgrade` and reboot to see if I can replicate it.

1000001101000, 2022-10-20:
What version of Debian are you running?

What version are you trying to upgrade to?

1000001101000, 2022-10-20:
Linux Kernel 5.10.149 is the Current current version for Debian 11 (Bullseye). I think you'll find that linux-image-armmp is the kernel that you have installed (That's the only one my installer uses for that device).

mandeldebugger, 2022-10-20:
> Linux Kernel 5.10.149 is the Current current version for Debian 11 (Bullseye). I think you'll find that linux-image-armmp is the kernel that you have installed (That's the only one my installer uses for that device).

Ah, I see. Yes, it is. This is a bit of a mystery because I installed the OMV version using [these](https://docs.openmediavault.org/en/stable/installation/on_debian.html) instructions and it was fine for a while until (I suspect) something happened while I did an 'apt-get install fancontrol'.

Reading around, it appears that [some](https://www.reddit.com/r/OpenMediaVault/comments/n4v6o1/omv_4_safe_for_2021/) have had issues with OMV4 and OMV5 on the LS220D because of the low memory & CPU. I don't doubt it - it maxes out at 100% CPU quite a lot so it might not be very suitable for the device. If anyone's interested I might try again but for the time being I'm giving webmin a go instead.

Thanks for your help.

closed 2022-10-20 by mandeldebugger

1000001101000, 2022-10-21:
Fun fact: I fixed a typo with those instructions a few years ago.
https://github.com/openmediavault/openmediavault-docs/commit/57696085834e45aec2dddba07aaabe18bdbbe7bb

I know webmin can max out the CPU on older generation devices at strange times but I think folks have used it on the LS220 successfully. I played with cockpit a while back and liked it though it has far fewer features than the others. 

My main advice is to learn how to do this stuff using command line tools but I also know that takes time.

mandeldebugger, 2022-10-21:
Actually, I've changed my mind and decided to soldier on with OMV once more. I'm just finishing the installation. I'm also carefully documenting outputs I get.

I also did `apt-get install fancontrol` with no issues before installing OMV and I agree - I think they're probably unrelated.

I've already noted an issue with the OMV installer script when running 'omv-confdbadm populate' with `/usr/sbin` not being in the PATH, which is easily fixed by `export PATH=$PATH:/usr/sbin`. I'm about to reboot ...