1000001101000/Debian_on_Buffalo #157: TSXL Lan light but no IP address

Yaboimer, 2022-10-31:
I tried following the directions but APC commander wasn't transferring files for me and would error out over a port issue. I got around that by creating a shared folder, uploading the files, then opening the basic shell with APC and moving the files to the boot sector.

I sent the reboot command and currently have "Terastation arm Debian installer" displayed and a LAN 2 light becasue my LAN 1 didn't work even with factory software. There is no IP and pressing the Display button does nothing.

I'm working on windows, I know I'm a peasant, and have the TS connected to the Ethernet port of my laptop which should be assigning an address of 168.254.140.*** but running "arp -a" I get nothing and sending a ping command to the address it was assigned when it was running factory firmware turns up nothing.

My device is recognizing its connected to something and sends packets to it but nothing gets sent back.

1000001101000, 2022-10-31:
Good to hear from you!

The installer needs access to the internet to function. I think it should start working properly if you connect the Terastation to your switch/router/etc and reboot it.

Yaboimer, 2022-11-01:
Ive connected it to a router and turned it on. Its been on for the past few hours with still no assigned IP address and no signs of life beyond the displayed message. 
I'm going to try to reload factory firmware and reload the images and try again.
There was something worth noting, when I was loading the images on the share folder the transfer speed was extremely low, in the magnitude of 20-30 kb/s. Thoughts?

Yaboimer, 2022-11-01:
Well it worked! I guess I was expecting an IP address to display like with the factory firmware, but after checking the devices on my router I found an address and was able to remote into the installer without issue!
Thanks so much for the help!

closed 2022-11-01 by Yaboimer