1000001101000/Debian_on_Buffalo #159: LSQL wont install

gonsalves2003, 2022-11-24:
I am trying to install on my LS-QL and it wont install. It comes back up and I can see the device on my network. No ports are open. I am unable to access SSH for installation.. Can you help please?

1000001101000, 2022-11-24:
Good to hear from you!

Can you provide more detail of the steps you’ve taken so far and what the LEDs on the front of the device are showing?

gonsalves2003, 2022-11-24:
Hey, thank You for your response.. 

I had the stock firmware 1.11. Used acp_commander to upload the files and reboot. did not work. Then tried with tftp. failed again. then used a Linux vm and mounted each hard disk ( I have 2*1tb) and copied the installers to the /boot partition and tried. all show the same results. I tried BullsEye, Buster and Stretch.

All the LEDs are normal. It does not power down when I press and hold the power button. Need to remove power chord.

gonsalves2003, 2022-11-24:
To add, after install, it goes to IP 192.169.11.150.

1000001101000, 2022-11-24:
> I had the stock firmware 1.11. Used acp_commander to upload the files and reboot. did not work.
What happened exactly?

> Then tried with tftp. failed again. 
This typically wont work unless the NAS can reach the internet at the same time as the TFTP server

> used a Linux vm and mounted each hard disk ( I have 2*1tb) and copied the installers to the /boot partition and tried
This will have messed up the raid1 array the partitions were in. I’m not sure how the bootloader would react.

> To add, after install, it goes to IP 192.168.11.150.
How did you determine this?

If you aren’t getting red leds flashing it is probably reading the files successfully.

one thing to check it it’s rebooting every minute or so. That would mean certain late bootloader fails or kernel panic.

The device must be able to get an address via DHCP and be able to access the main debian mirror over the internet, otherwise it will get stuck before starting ssh

gonsalves2003, 2022-11-24:
Thank You for the response. looks like the issue was to do with the DHCP server availability. I was connecting the device directly to my system. I was able to install Bullseye and was able to login using SSH after install. Thanks again for your help.. 

I see some issues.. not sure those are expected behaviors.. Pressing and holding power button does not do anything. With Stock Firmware, it used to initiate a shutdown. When I issue Shutdown command through terminal, there's no effect on the LED during the shutdown. That used to blink with Stock firmware.. Any option to bring those behaviors back?

gonsalves2003, 2022-11-25:
@1000001101000 , Thank you for post install notes. I was able to configure Power Button for shut down using triggerhappy.

1000001101000, 2022-11-25:
If you want to change the LED behavior at shutdown you can set the trigger via
/sys/class/led/<led name>/trigger

You may need to load a kernel module to enable additional trigger options.

gonsalves2003, 2022-11-26:
Great! Thats very helpful! Thank You.

gonsalves2003, 2022-11-27:
@1000001101000 I am able to edit brightness, not trigger. looks like some permission issue. the file shows only 0b size and none of the chmod, chown or sudo options let me edit the trigger file - both with echo or editor directly. I know it is an interface method. Anything to do with the build of the image?

1000001101000, 2022-11-27:
What do you get if you `cat` the trigger file what is the output?

gonsalves2003, 2022-11-27:
cat /sys/class/leds/linkstation:power-blue/trigger
[none] usb-gadget usb-host f1072004.mdio-bus-mii:08:link f1072004.mdio-bus-mii:08:1Gbps f1072004.mdio-bus-mii:08:100Mbps f1072004.mdio-bus-mii:08:10Mbps timer disk-activity disk-read disk-write ide-disk mtd nand-disk gpio cpu cpu0 default-on panic

gonsalves2003, 2022-11-27:
not able to edit the file though / both by editor and echo

1000001101000, 2022-11-27:
What are you trying to set it to?

gonsalves2003, 2022-11-27:
I am trying to set heartbeat

1000001101000, 2022-11-27:
cat shows the currently available options, it will give you an error if you try to set it to something not in the list.

It’s been a long time since I looked at it, but it seems like you should be able to enable a suitable trigger with
`modprobe ledtrig-oneshot`

Details on using it are here: https://www.kernel.org/doc/Documentation/leds/ledtrig-oneshot.rst

The process will be similar for heartbeat.

gonsalves2003, 2022-11-27:
Thank You! echo timer > /sys/class/leds/linkstation:power-blue/trigger worked! 

As a final help, I am trying to install JRE17 or above on this. Do you know any JRE binary for armel available for download?

1000001101000, 2023-06-07:
didn't realize I never answered this. The debian repo has a metapackage called "default-jdk" which should install openjdk , I would expect that to work.

closed 2023-06-07 by 1000001101000