1000001101000/Debian_on_Buffalo #161: TS-XEL: "No init file at hdd0" on box; "hard resetting link" errors in console output

mholl1983, 2022-12-03:
Hi again, I hope you're well! It was nice being error free on the TS-XEL for awhile, but sadly some new issues arose that I could use help with.

Today, I noticed that my DLNA server was showing media files, but when I clicked any of them, I got an error. I went upstairs to check the box; it was on without any errors, though I heard what sounded like a HDD spinning and struggling to do so. 

I shut it down and rebooted; then, it hung at the "no init file", so I decided to break out the UART/serial console cable again (grateful for that option; thanks again the tips previously!)

I got output in Putty and let the box do its thing as it booted up. When the "no init file" output appeared on the display, I looked t the Putty output and saw the following:

[image]
 
"hard resetting link" and the output around it appeared to show up in a loop in the output. From what I read, it could be bad contact with a specific drive. I'm guessing it could be a drive failing, too? If this looks familiar and if you have any suggestions, do let me know, and thanks again for all your help!

1000001101000, 2022-12-03:
similar messages show up for all sorts of errors related to drives. loose connections can cause it but so can a failing drive or even some types of issue between the drive/controller firmware/drivers.

If this started happening on a device where nothing changed recently it probably means a drive is failing. It might be worth shutting down and pulling/reinserting all the drives to ensure there isn't a bad connection somehow. 

it might be trying to show you a recovery shell that you can't see because of all the sata errors. you could try sending ctrl+alt+f2 to see if you can access via an alternate virtual terminal.... I don't actually know if that shell supports that or not but it's worth a try. otherwise the prompt usually says something like "enter root password for recovery shell", you could try blindly typing your root password and hitting enter. If you manage to get to a recovery shell you could comment whatever file system is failing to start etc.

mholl1983, 2022-12-03:
Thank you for the reply! I'll try to reseat the drives and perhaps first use compressed air to blow out any dust/debris that may be in the connections.

Interesting about the recovery shell. I'll give that a spin if I continue to have issues. Will report back with results. Thank you!

mholl1983, 2022-12-03:
I have a fairly recent backup on a USB drive (lesson learned after losing some data in the past, as you know!) But if I can't get back into the TS box, it'd be nice to try to retrieve data from the 4 drives/raid array. Did you say if I wanted to do that, I'd have to hook up all 4 to a computer running Linux?

mholl1983, 2022-12-03:
>  you could try sending ctrl+alt+f2 to see if you can access via an alternate virtual terminal

When would I do this command? At any particular point during the serial connection output?

mholl1983, 2022-12-03:
I isolated drive 1 as what seems to be the "problem drive". It was making a loud clicking noise the last time I tried a boot. 

Currently, the TS is going through its boot process and seems to be proceeding. Very interesting!

mholl1983, 2022-12-03:
I now see this prompt. 

[image]

mholl1983, 2022-12-03:
And I'm now in emergency mode and can access fstab, etc. Not sure what to check for, but at least it's something!

1000001101000, 2022-12-03:
Nice!

I'm guessing the reason it didn't boot all the way is that the degraded array didn't start which prevented it from being mounted which in turn prevented boot because it isn't listed as optional in the fstab. 

What I would probably do:

1. edit `/etc/fstab` and comment out the RAID5 array temporarily
2. reboot the device which should now succeed since it shouldn't care if that array starts or not. 
3. use `mdadm --detail`  to figure out what's going on with the array
4. replace the failed drive or otherwise deal with whatever the problem is.

mholl1983, 2022-12-03:
Sounds like a good plan! I packed it in for the day, but will definitely follow suit on this tomorrow or shortly after. 

If I replace a failed drive and rebuild the array, I shouldn't lose my files? 

Thanks again!

1000001101000, 2022-12-03:
> Thinking of contingency plans in the meantime. I'd likely build a mid-tower with Linux or Windows to replace the TS, though I hope I can still squeeze a bit more life out of the TS. Still, I'm aware it's an older piece of kit and it won't last forever, but it's just such a formidable little tank!
> 
> Anyways, I have a fairly recent backup on a USB drive (lesson learned after losing some data in the past, as you know!) But if I can't get back into the TS box, it'd be nice to try to retrieve data from the 4 drives/raid array. Did you say if I wanted to do that, I'd have to hook up all 4 to a computer running Linux?

Sorry, just noticed the previous messages. Once you're in you should be able to force start the array which you could then access normally to retrieve any data etc. You could then replace the drive and re-sync etc. 

Someone (possibly even me) should probably develop a better set of best practices for these devices. I can see why it might default to not starting a degraded RAID5....but for a device with no console that does way more harm than good. Same for non-optional filsystems though that depends heavily on what the filesystem does. 

Fairly recently I repurposed some old PC parts into a sort of NAS/files server which has proved to be an interesting experiment for me. It's now my main storage system which backs itself up to some Terastations. I find they complement each other will since I can enable some more CPU intensive features on the PC/NAS and am getting surprisingly good backup performance by limiting the TS to just receiving those backups etc.

1000001101000, 2022-12-03:
> Sounds like a good plan! I packed it in for the day, but will definitely follow suit on this tomorrow or shortly after.
> 
> If I replace a failed drive and rebuild the array, I shouldn't lose my files?
> 
> Thanks again!

Like you said, RAID5 can handle a single drive failure. Re-syncing can take a long time and does stress the other drives, so if you don't have a full backup you should probably do that first once you get the array started and mounted again.

1000001101000, 2022-12-08:
Hmm, that is worrying. The partitions on sda don’t appear to be active in their arrays, the question is: when did that happen?

could be something that happened while you were troubleshooting the sdd problems. If so you can probably force assemble your RAID5 and move forward.

If this happened a long time ago and you didn’t notice the the array might have been degraded for a long time until failing with the death of sdd. That might explain why it failed to boot the way it did. Hopefully not since that might not be recoverable. 

First thing I would do is see what mdadm says about sda. Try: `mdadm —examine /dev/sda6` that should tell you the state of that part of the array, you should be able to compare that to the output from for the other two

mholl1983, 2022-12-08:
Thanks for writing and for the tips!

Here's what the command shows. I'm guessing I missed something.

[image]

1000001101000, 2022-12-08:
guessing something about the command is mangled, maybe the double-hyphen. Here's output from one of my boxes (my disk isn't partitioned but it should otherwise be the same)

```
# mdadm --examine /dev/sda
/dev/sda:
          Magic : a92b4efc
        Version : 1.2
    Feature Map : 0x1
     Array UUID : 60497d:2e957a:fda249:022cc6
           Name : md99
  Creation Time : Tue Jan 18 19:57:30 2022
     Raid Level : raid5
   Raid Devices : 7

 Avail Dev Size : 7813776048 (3725.90 GiB 4000.65 GB)
     Array Size : 23441316864 (22355.38 GiB 24003.91 GB)
  Used Dev Size : 7813772288 (3725.90 GiB 4000.65 GB)
    Data Offset : 261120 sectors
   Super Offset : 8 sectors
   Unused Space : before=261040 sectors, after=3760 sectors
          State : clean
    Device UUID : 19ebc0:0d0e82:34210cfc:1a38fdc

Internal Bitmap : 8 sectors from superblock
    Update Time : Thu Dec  8 11:45:16 2022
  Bad Block Log : 512 entries available at offset 24 sectors
       Checksum : 1366004 - correct
         Events : 103301

         Layout : left-symmetric
     Chunk Size : 512K

   Device Role : Active device 1
   Array State : AAAAAAA ('A' == active, '.' == missing, 'R' == replacing)
```

mholl1983, 2022-12-08:
D'oh! It was that, apologies. Here's the output:

```
marc@debian:~$ sudo mdadm --examine /dev/sda6
/dev/sda6:
          Magic : a92b4efc
        Version : 1.2
    Feature Map : 0x1
     Array UUID : 46c1d87e:e7b2970a:3edd105f:c51f08fe
           Name : TS-XEL02C:2
  Creation Time : Sat Sep 26 15:27:40 2020
     Raid Level : raid5
   Raid Devices : 4

 Avail Dev Size : 1923497952 sectors (917.20 GiB 984.83 GB)
     Array Size : 2885245440 KiB (2.69 TiB 2.95 TB)
  Used Dev Size : 1923496960 sectors (917.19 GiB 984.83 GB)
    Data Offset : 2048 sectors
   Super Offset : 8 sectors
   Unused Space : before=1968 sectors, after=992 sectors
          State : active
    Device UUID : 18ae8cf5:e463c0b7:824d70b5:b88498c3

Internal Bitmap : 8 sectors from superblock
    Update Time : Sat Nov 26 14:46:23 2022
       Checksum : e8dd79b4 - correct
         Events : 39140

         Layout : left-symmetric
     Chunk Size : 512K

   Device Role : Active device 1
   Array State : AAAA ('A' == active, '.' == missing, 'R' == replacing)
```

mholl1983, 2022-12-09:
Deja vu!

```
marc@debian:~$ sudo mdadm --assemble /dev/md2 /dev/sda6 /dev/sdb6 /dev/sdc6  --force
mdadm: forcing event count in /dev/sda6(1) from 39140 up to 39795
mdadm: Marking array /dev/md2 as 'clean'
mdadm: /dev/md2 has been started with 3 drives (out of 4).
marc@debian:~$ sudo mdadm --detail /dev/md1/
mdadm: cannot open /dev/md1/: Not a directory
marc@debian:~$ sudo mdadm --detail /dev/md2
/dev/md2:
           Version : 1.2
     Creation Time : Sat Sep 26 15:27:40 2020
        Raid Level : raid5
        Array Size : 2885245440 (2.69 TiB 2.95 TB)
     Used Dev Size : 961748480 (917.19 GiB 984.83 GB)
      Raid Devices : 4
     Total Devices : 3
       Persistence : Superblock is persistent

     Intent Bitmap : Internal

       Update Time : Sat Nov 26 14:46:23 2022
             State : clean, degraded
    Active Devices : 3
   Working Devices : 3
    Failed Devices : 0
     Spare Devices : 0

            Layout : left-symmetric
        Chunk Size : 512K

Consistency Policy : bitmap

              Name : TS-XEL02C:2
              UUID : 46c1d87e:e7b2970a:3edd105f:c51f08fe
            Events : 39795

    Number   Major   Minor   RaidDevice State
       -       0        0        0      removed
       1       8        6        1      active sync   /dev/sda6
       2       8       22        2      active sync   /dev/sdb6
       3       8       38        3      active sync   /dev/sdc6
```

mholl1983, 2022-12-09:
Now I'm a little worried. `sudo mdadm --detail /dev/md2` gives a "no such file or directory" error.

1000001101000, 2022-12-09:
`mdadm --detail --scan` will list all the arrays it knows about, it might have started as `/dev/md127` or something like that. 

If it's still active I'd recommend mounting it and backing up any data you care about before you worry about anything else.

mholl1983, 2022-12-09:
```
Last login: Fri Dec  9 09:25:46 2022 from 192.168.50.125
marc@debian:~$ sudo mdadm --detail --scan
[sudo] password for marc:
ARRAY /dev/md0 metadata=0.90 UUID=e7194c7c:f3579d9d:529ae12f:b827b7aa
ARRAY /dev/md/10 metadata=1.2 name=TS-XEL-EM02C:10 UUID=db00a7ae:43c0dfeb:89959a8c:b4b581b3
ARRAY /dev/md/1 metadata=1.2 name=TS-XEL-EM02C:1 UUID=d6c17ec0:cc93ec32:218a896f:4358e4cf
INACTIVE-ARRAY /dev/md2 metadata=1.2 name=TS-XEL02C:2 UUID=46c1d87e:e7b2970a:3edd105f:c51f08fe
marc@debian:~$ stopped /dev/md1
-bash: stopped: command not found
marc@debian:~$ sudo mdadm -S /dev/md2
mdadm: stopped /dev/md2
marc@debian:~$ sudo mdadm --assemble /dev/md2 /dev/sdb6 /dev/sdc6 /dev/sdd6 --force
mdadm: forcing event count in /dev/sdb6(1) from 39797 up to 39799
mdadm: /dev/md2 has been started with 3 drives (out of 4).
marc@debian:~$ mount /dev/md2 /mnt/array
mount: /mnt/array: must be superuser to use mount.
       dmesg(1) may have more information after failed mount system call.
marc@debian:~$ sudo mount /dev/md2 /mnt/array
mount: /mnt/array: special device /dev/md2 does not exist.
       dmesg(1) may have more information after failed mount system call.
marc@debian:~$ mount /dev/md2 /mnt/array
mount: /mnt/array: must be superuser to use mount.
       dmesg(1) may have more information after failed mount system call.
marc@debian:~$ su
Password:
root@debian:/home/marc# mount /dev/md2 /mnt/array
mount: /mnt/array: special device /dev/md2 does not exist.
       dmesg(1) may have more information after failed mount system call.
root@debian:/home/marc# mdadm --detail --scan
bash: mdadm: command not found
root@debian:/home/marc# mdadm --detail --scan
bash: mdadm: command not found
root@debian:/home/marc# sudo mdadm
Usage: mdadm --help
  for help
root@debian:/home/marc# sudo mdadm --detail --scan
ARRAY /dev/md0 metadata=0.90 UUID=e7194c7c:f3579d9d:529ae12f:b827b7aa
ARRAY /dev/md/10 metadata=1.2 name=TS-XEL-EM02C:10 UUID=db00a7ae:43c0dfeb:89959a8c:b4b581b3
ARRAY /dev/md/1 metadata=1.2 name=TS-XEL-EM02C:1 UUID=d6c17ec0:cc93ec32:218a896f:4358e4cf

```

mholl1983, 2022-12-09:
Seems like I'm able to force build the array but can't mount it, and then it seems to disappear? Let me know if I am missing something. Like you suggested, just trying to make it active w/ 3 drives to try to do a full backup or as much as possible at this point. Thanks again!

1000001101000, 2022-12-09:
take a look at `dmesg`

1000001101000, 2022-12-09:
also, if your data is worth thousands of dollars you might consider taking all your drives to one of those data recovery companies.

1000001101000, 2022-12-09:
you may want to get a full root shell to make this a bit easier, `sudo su -`

1000001101000, 2022-12-09:
`ls /sys/block`

mholl1983, 2022-12-09:
Thanks! It'll be sad, but not the end of the world if I lose any or all data. I made a point of keeping our important files elsewhere on my main computer and backed up to an external USB drive. The TS-XEL raid array houses all of our media, so it's kind of a local "streaming box", as I affectionately call it.

Will try what you suggested above and let you know how it goes. I have a new Seagate drive on the way, but I have a feeling there's more damage than just one drive going bad. Thanks so much again and I'll keep you posted!

mholl1983, 2022-12-09:
Output from ls /sys/block

`md0  md1  md10  mtdblock0  mtdblock1  sdc  sdd`

mholl1983, 2022-12-09:
Output from dmesg (sorry for the long message!)

```

root@debian:/home/marc# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.0.0-5-marvell (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-9) 12.2.0, GNU ld (GNU Binutils for Debian) 2.39) #1 Debian 6.0.10-2 (2022-12-01)
[    0.000000] CPU: Feroceon 88FR131 [56251311] revision 1 (ARMv5TE), cr=0005317f
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] OF: fdt: Machine model: Buffalo Terastation TS-XEL
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 65024
[    0.000000] Kernel command line: console=ttyS0,115200
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] mem auto-init: stack:all(zero), heap alloc:on, heap free:off
[    0.000000] Memory: 244732K/262144K available (5031K kernel code, 833K rwdata, 1508K rodata, 320K init, 246K bss, 17412K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 24229 entries in 72 pages
[    0.000000] ftrace: allocated 71 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] clocksource: orion_clocksource: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 9556302233 ns
[    0.000002] sched_clock: 32 bits at 200MHz, resolution 5ns, wraps every 10737418237ns
[    0.000042] Switching to timer-based delay loop, resolution 5ns
[    0.000287] Calibrating delay loop (skipped), value calculated using timer frequency.. 400.00 BogoMIPS (lpj=800000)
[    0.000318] pid_max: default: 32768 minimum: 301
[    0.000922] LSM: Security Framework initializing
[    0.001274] landlock: Up and running.
[    0.001285] Yama: disabled by default; enable with sysctl kernel.yama.*
[    0.001581] AppArmor: AppArmor initialized
[    0.001962] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001996] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.004131] CPU: Testing write buffer coherency: ok
[    0.006581] cblist_init_generic: Setting adjustable number of callback queues.
[    0.006598] cblist_init_generic: Setting shift to 0 and lim to 1.
[    0.006891] cblist_init_generic: Setting shift to 0 and lim to 1.
[    0.007266] Setting up static identity map for 0x8220 - 0x825c
[    0.007563] mvebu-soc-id: MVEBU SoC ID=0x6281, Rev=0x3
[    0.009272] devtmpfs: initialized
[    0.015204] VFP support v0.3: not present
[    0.015382] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.015423] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.015646] pinctrl core: initialized pinctrl subsystem
[    0.018097] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.018890] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.019926] audit: initializing netlink subsys (disabled)
[    0.020419] audit: type=2000 audit(0.020:1): state=initialized audit_enabled=0 res=1
[    0.021350] thermal_sys: Registered thermal governor 'fair_share'
[    0.021366] thermal_sys: Registered thermal governor 'bang_bang'
[    0.021375] thermal_sys: Registered thermal governor 'step_wise'
[    0.021381] thermal_sys: Registered thermal governor 'user_space'
[    0.021479] cpuidle: using governor ladder
[    0.021845] Feroceon L2: Enabling L2
[    0.021881] Feroceon L2: Cache support initialised.
[    0.031035] No ATAGs?
[    0.071452] clocksource: Switched to clocksource orion_clocksource
[    0.072354] VFS: Disk quotas dquot_6.6.0
[    0.072448] VFS: Dquot-cache hash table entries: 1024 (order 0, 4096 bytes)
[    0.073369] AppArmor: AppArmor Filesystem Enabled
[    0.075123] NET: Registered PF_INET protocol family
[... 1563 lines cut ...]
[19734.023726] md/raid:md2: read error not correctable (sector 2248 on sdb6).
[19734.030664] md/raid:md2: read error not correctable (sector 2256 on sdb6).
[19734.037604] md/raid:md2: read error not correctable (sector 2264 on sdb6).
[19734.044536] md/raid:md2: read error not correctable (sector 2272 on sdb6).
[19734.051468] md/raid:md2: read error not correctable (sector 2280 on sdb6).
[19734.058531] sd 0:1:0:0: [sdb] tag#6 FAILED Result: hostbyte=DID_OK driverbyte=DRIVER_OK cmd_age=61s
[19734.067657] sd 0:1:0:0: [sdb] tag#6 Sense Key : Medium Error [current]
[19734.074340] sd 0:1:0:0: [sdb] tag#6 Add. Sense: Unrecovered read error - auto reallocate failed
[19734.083113] sd 0:1:0:0: [sdb] tag#6 CDB: Read(10) 28 00 73 7c 07 00 00 00 08 00
[19734.090480] I/O error, dev sdb, sector 1937508096 op 0x0:(READ) flags 0x800 phys_seg 1 prio class 2
[19734.099613] ata1: EH complete
[19734.103050] I/O error, dev sdb, sector 1937508304 op 0x0:(READ) flags 0x800 phys_seg 1 prio class 2
[... cut ...]
[19734.189647] sd 0:1:0:0: [sdb] Synchronizing SCSI cache
[19734.215123] sd 0:1:0:0: [sdb] Synchronize Cache(10) failed: Result: hostbyte=DID_BAD_TARGET driverbyte=DRIVER_OK
[19734.225416] sd 0:1:0:0: [sdb] Stopping disk
[19734.273491] sd 0:1:0:0: [sdb] Start/Stop Unit failed: Result: hostbyte=DID_BAD_TARGET driverbyte=DRIVER_OK
[19734.888262] md2: detected capacity change from 5770490880 to 0
[19734.894197] md: md2 stopped.
[19744.644300] /dev/md2: Can't open blockdev
[19763.213036] /dev/md2: Can't open blockdev

```

mholl1983, 2022-12-09:
Of note are some errors that are called out where "md2" is referenced.

mholl1983, 2022-12-09:
I rebooted and did ls /sys/block, and md2 is back

`md0  md1  md10  md2  mtdblock0  mtdblock1  sda  sdb  sdc
`

1000001101000, 2022-12-09:
looks like one of your drives is throwing IO errors. 
```
[    9.263620] ata1.00: Read log 0x10 page 0x00 failed, Emask 0x1
[    9.269517] ata1.00: failed to read log page 10h (errno=-5)
[    9.275147] ata1.00: failed to read SCR 1 (Emask=0x40)
[    9.280337] ata1.00: NCQ disabled due to excessive errors
[    9.285791] ata1.01: failed to read SCR 1 (Emask=0x40)
[    9.290973] ata1.02: failed to read SCR 1 (Emask=0x40)
[    9.296162] ata1.03: failed to read SCR 1 (Emask=0x40)
[    9.301352] ata1.00: exception Emask 0x1 SAct 0x4000 SErr 0x0 action 0x6 frozen
[    9.308715] ata1.00: dev err
[    9.311633] ata1.00: failed command: READ FPDMA QUEUED
[    9.316815] ata1.00: cmd 60/02:70:02:00:00/00:00:00:00:00/40 tag 14 ncq dma 1024 in
                        res 51/04:70:02:00:00/04:00:00:00:00/40 Emask 0x1 (device error)
[    9.332406] ata1.00: status: { DRDY ERR }
[    9.336452] ata1.00: error: { ABRT }
[    9.811509] ata1.15: SATA link up 3.0 Gbps (SStatus 123 SControl F300)
[    9.818852] ata1.00: hard resetting link
```

It also seems to complain about some bad data in the MD device, not shocking since you had to force it to assemble.
`md/raid:md2: read error not correctable (sector 2280 on sdb6).`

1000001101000, 2022-12-09:
if the drive is still mostly working you might be able to clone it to a new drive and then force add that to the array. There might even be a way to make mdadm ignore a few data errors so you can try to recover what's left. To be honest I've never gotten to that point personally.

mholl1983, 2022-12-09:
Cool, thanks. Interesting about cloning to a new drive. I'm guessing I'd have to try that outside of the TS box itself? Like on a Linux desktop or something. Will cross that bridge if and when I have to, but it's a good suggestion.

I'll try for the SMART data now, momentarily, and post back here.

mholl1983, 2022-12-09:
sdd6:

```root@debian:/home/marc# sudo smartctl -a /dev/sdd
smartctl 7.3 2022-02-28 r5338 [armv5tel-linux-6.0.0-5-marvell] (local build)
Copyright (C) 2002-22, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Family:     Seagate Constellation ES.3
Device Model:     ST1000NM0033-9ZM173
Serial Number:    Z1W0MD3D
LU WWN Device Id: 5 000c50 064cad930
Firmware Version: 0001
User Capacity:    1,000,204,886,016 bytes [1.00 TB]
Sector Size:      512 bytes logical/physical
Rotation Rate:    7200 rpm
Form Factor:      3.5 inches
Device is:        In smartctl database 7.3/5319
ATA Version is:   ACS-2 (minor revision not indicated)
SATA Version is:  SATA 3.0, 6.0 Gb/s (current: 1.5 Gb/s)
Local Time is:    Fri Dec  9 17:27:36 2022 NST
SMART support is: Available - device has SMART capability.
SMART support is: Enabled

=== START OF READ SMART DATA SECTION ===
SMART overall-health self-assessment test result: FAILED!
Drive failure expected in less than 24 hours. SAVE ALL DATA.
See vendor-specific Attribute list for failed Attributes.

General SMART Values:
Offline data collection status:  (0x82) Offline data collection activity
                                        was completed without error.
                                        Auto Offline Data Collection: Enabled.
Self-test execution status:      (   0) The previous self-test routine completed
                                        without error or no self-test has ever
                                        been run.
Total time to complete Offline
data collection:                (  592) seconds.
Offline data collection
capabilities:                    (0x7b) SMART execute Offline immediate.
                                        Auto Offline data collection on/off supp                                                                           ort.
                                        Suspend Offline collection upon new
                                        command.
                                        Offline surface scan supported.
                                        Self-test supported.
                                        Conveyance Self-test supported.
                                        Selective Self-test supported.
SMART capabilities:            (0x0003) Saves SMART data before entering
                                        power-saving mode.
                                        Supports SMART auto save timer.
Error logging capability:        (0x01) Error logging supported.
                                        General Purpose Logging supported.
Short self-test routine
recommended polling time:        (   1) minutes.
Extended self-test routine
recommended polling time:        ( 116) minutes.
Conveyance self-test routine
recommended polling time:        (   2) minutes.
SCT capabilities:              (0x50bd) SCT Status supported.
                                        SCT Error Recovery Control supported.
                                        SCT Feature Control supported.
                                        SCT Data Table supported.

SMART Attributes Data Structure revision number: 10
Vendor Specific SMART Attributes with Thresholds:
ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_                                                                           FAILED RAW_VALUE
  1 Raw_Read_Error_Rate     0x000f   037   036   044    Pre-fail  Always   FAILI                                                                           NG_NOW 12263131
  3 Spin_Up_Time            0x0003   097   097   000    Pre-fail  Always       -                                                                                  0
  4 Start_Stop_Count        0x0032   100   100   020    Old_age   Always       -                                                                                  617
  5 Reallocated_Sector_Ct   0x0033   077   077   010    Pre-fail  Always       -                                                                                  1285
  7 Seek_Error_Rate         0x000f   090   060   030    Pre-fail  Always       -                                                                                  1179813657
[... 11 lines cut ...]
194 Temperature_Celsius     0x0022   030   053   000    Old_age   Always       -                                                                                  30 (0 12 0 0 0)
195 Hardware_ECC_Recovered  0x001a   012   012   000    Old_age   Always       -                                                                                  12263131
197 Current_Pending_Sector  0x0012   095   095   000    Old_age   Always       -                                                                                  856
198 Offline_Uncorrectable   0x0010   095   095   000    Old_age   Offline      -                                                                                  856
199 UDMA_CRC_Error_Count    0x003e   200   200   000    Old_age   Always       -                                                                                  0

SMART Error Log Version: 1
No Errors Logged

SMART Self-test log structure revision number 1
Num  Test_Description    Status                  Remaining  LifeTime(hours)  LBA                                                                           _of_first_error
# 1  Short offline       Completed without error       00%     15256         -
[... cut ...]  00%         3         -

SMART Selective self-test log data structure revision number 1
 SPAN  MIN_LBA  MAX_LBA  CURRENT_TEST_STATUS
    1        0        0  Not_testing
    2        0        0  Not_testing
    3        0        0  Not_testing
    4        0        0  Not_testing
    5        0        0  Not_testing
Selective self-test flags (0x0):
  After scanning selected spans, do NOT read-scan remainder of disk.
If Selective self-test is pending on power-up, resume after 0 minute delay.
```

mholl1983, 2022-12-09:
sdc:

```root@debian:/home/marc# sudo smartctl -a /dev/sdc
smartctl 7.3 2022-02-28 r5338 [armv5tel-linux-6.0.0-5-marvell] (local build)
Copyright (C) 2002-22, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Family:     Seagate Constellation ES.3
Device Model:     ST1000NM0033-9ZM173
Serial Number:    Z1W03Z5C
LU WWN Device Id: 5 000c50 050ace05a
Firmware Version: 0001
User Capacity:    1,000,204,886,016 bytes [1.00 TB]
Sector Size:      512 bytes logical/physical
Rotation Rate:    7200 rpm
Form Factor:      3.5 inches
Device is:        In smartctl database 7.3/5319
ATA Version is:   ACS-2 (minor revision not indicated)
SATA Version is:  SATA 3.0, 6.0 Gb/s (current: 3.0 Gb/s)
Local Time is:    Fri Dec  9 17:28:43 2022 NST
SMART support is: Available - device has SMART capability.
SMART support is: Enabled

=== START OF READ SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

General SMART Values:
Offline data collection status:  (0x82) Offline data collection activity
                                        was completed without error.
                                        Auto Offline Data Collection: Enabled.
Self-test execution status:      (   0) The previous self-test routine completed
                                        without error or no self-test has ever
                                        been run.
Total time to complete Offline
data collection:                (  600) seconds.
Offline data collection
capabilities:                    (0x7b) SMART execute Offline immediate.
                                        Auto Offline data collection on/off support.
                                        Suspend Offline collection upon new
                                        command.
                                        Offline surface scan supported.
                                        Self-test supported.
                                        Conveyance Self-test supported.
                                        Selective Self-test supported.
SMART capabilities:            (0x0003) Saves SMART data before entering
                                        power-saving mode.
                                        Supports SMART auto save timer.
Error logging capability:        (0x01) Error logging supported.
                                        General Purpose Logging supported.
Short self-test routine
recommended polling time:        (   1) minutes.
Extended self-test routine
recommended polling time:        ( 127) minutes.
Conveyance self-test routine
recommended polling time:        (   2) minutes.
SCT capabilities:              (0x50bd) SCT Status supported.
                                        SCT Error Recovery Control supported.
                                        SCT Feature Control supported.
                                        SCT Data Table supported.

SMART Attributes Data Structure revision number: 10
Vendor Specific SMART Attributes with Thresholds:
ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
  1 Raw_Read_Error_Rate     0x000f   081   059   044    Pre-fail  Always       -       156575972
  3 Spin_Up_Time            0x0003   096   096   000    Pre-fail  Always       -       0
  4 Start_Stop_Count        0x0032   100   100   020    Old_age   Always       -       622
  5 Reallocated_Sector_Ct   0x0033   100   100   010    Pre-fail  Always       -       5
  7 Seek_Error_Rate         0x000f   090   060   030    Pre-fail  Always       -       1298740013
  9 Power_On_Hours          0x0032   061   061   000    Old_age   Always       -       34166
 10 Spin_Retry_Count        0x0013   100   100   097    Pre-fail  Always       -       0
[... 105 lines cut ...]
  40 51 00 ff ff ff 0f  Error: UNC at LBA = 0x0fffffff = 268435455

  Commands leading to the command that caused the error were:
  CR FR SC SN CL CH DH DC   Powered_Up_Time  Command/Feature_Name
  -- -- -- -- -- -- -- --  ----------------  --------------------
  25 00 00 ff ff ff 4f 00      06:10:05.112  READ DMA EXT
  25 00 00 ff ff ff 4f 00      06:09:24.315  READ DMA EXT
  25 00 00 ff ff ff 4f 00      06:09:21.575  READ DMA EXT
  35 00 a8 ff ff ff 4f 00      06:09:21.568  WRITE DMA EXT
  ea 00 00 00 00 00 40 00      06:09:21.553  FLUSH CACHE EXT

SMART Self-test log structure revision number 1
Num  Test_Description    Status                  Remaining  LifeTime(hours)  LBA_of_first_error
# 1  Short offline       Completed without error       00%        48         -

SMART Selective self-test log data structure revision number 1
 SPAN  MIN_LBA  MAX_LBA  CURRENT_TEST_STATUS
    1        0        0  Not_testing
    2        0        0  Not_testing
    3        0        0  Not_testing
    4        0        0  Not_testing
    5        0        0  Not_testing
Selective self-test flags (0x0):
  After scanning selected spans, do NOT read-scan remainder of disk.
If Selective self-test is pending on power-up, resume after 0 minute delay.```

mholl1983, 2022-12-09:
sdb:

```root@debian:/home/marc# sudo smartctl -a /dev/sdb
smartctl 7.3 2022-02-28 r5338 [armv5tel-linux-6.0.0-5-marvell] (local build)
Copyright (C) 2002-22, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Family:     Seagate Constellation ES.3
Device Model:     ST1000NM0033-9ZM173
Serial Number:    Z1W02XDJ
LU WWN Device Id: 5 000c50 050519a6e
Firmware Version: 0001
User Capacity:    1,000,204,886,016 bytes [1.00 TB]
Sector Size:      512 bytes logical/physical
Rotation Rate:    7200 rpm
Form Factor:      3.5 inches
Device is:        In smartctl database 7.3/5319
ATA Version is:   ACS-2 (minor revision not indicated)
SATA Version is:  SATA 3.0, 6.0 Gb/s (current: 3.0 Gb/s)
Local Time is:    Fri Dec  9 17:29:44 2022 NST
SMART support is: Available - device has SMART capability.
SMART support is: Enabled

=== START OF READ SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

General SMART Values:
Offline data collection status:  (0x82) Offline data collection activity
                                        was completed without error.
                                        Auto Offline Data Collection: Enabled.
Self-test execution status:      (   0) The previous self-test routine completed
                                        without error or no self-test has ever
                                        been run.
Total time to complete Offline
data collection:                (  584) seconds.
Offline data collection
capabilities:                    (0x7b) SMART execute Offline immediate.
                                        Auto Offline data collection on/off support.
                                        Suspend Offline collection upon new
                                        command.
                                        Offline surface scan supported.
                                        Self-test supported.
                                        Conveyance Self-test supported.
                                        Selective Self-test supported.
SMART capabilities:            (0x0003) Saves SMART data before entering
                                        power-saving mode.
                                        Supports SMART auto save timer.
Error logging capability:        (0x01) Error logging supported.
                                        General Purpose Logging supported.
Short self-test routine
recommended polling time:        (   1) minutes.
Extended self-test routine
recommended polling time:        ( 112) minutes.
Conveyance self-test routine
recommended polling time:        (   2) minutes.
SCT capabilities:              (0x50bd) SCT Status supported.
                                        SCT Error Recovery Control supported.
                                        SCT Feature Control supported.
                                        SCT Data Table supported.

SMART Attributes Data Structure revision number: 10
Vendor Specific SMART Attributes with Thresholds:
ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
  1 Raw_Read_Error_Rate     0x000f   081   063   044    Pre-fail  Always       -       151642236
  3 Spin_Up_Time            0x0003   096   096   000    Pre-fail  Always       -       0
  4 Start_Stop_Count        0x0032   100   100   020    Old_age   Always       -       620
  5 Reallocated_Sector_Ct   0x0033   100   100   010    Pre-fail  Always       -       0
  7 Seek_Error_Rate         0x000f   090   060   030    Pre-fail  Always       -       1157132968
  9 Power_On_Hours          0x0032   062   062   000    Old_age   Always       -       34158
 10 Spin_Retry_Count        0x0013   100   100   097    Pre-fail  Always       -       0
 12 Power_Cycle_Count       0x0032   100   100   020    Old_age   Always       -       619
184 End-to-End_Error        0x0032   100   100   099    Old_age   Always       -       0
187 Reported_Uncorrect      0x0032   100   100   000    Old_age   Always       -       0
188 Command_Timeout         0x0032   100   100   000    Old_age   Always       -       0
189 High_Fly_Writes         0x003a   098   098   000    Old_age   Always       -       2
190 Airflow_Temperature_Cel 0x0022   068   050   045    Old_age   Always       -       32 (Min/Max 31/32)
191 G-Sense_Error_Rate      0x0032   100   100   000    Old_age   Always       -       0
192 Power-Off_Retract_Count 0x0032   100   100   000    Old_age   Always       -       618
193 Load_Cycle_Count        0x0032   100   100   000    Old_age   Always       -       620
194 Temperature_Celsius     0x0022   032   050   000    Old_age   Always       -       32 (0 14 0 0 0)
195 Hardware_ECC_Recovered  0x001a   027   012   000    Old_age   Always       -       151642236
197 Current_Pending_Sector  0x0012   100   100   000    Old_age   Always       -       0
198 Offline_Uncorrectable   0x0010   100   100   000    Old_age   Offline      -       0
199 UDMA_CRC_Error_Count    0x003e   200   200   000    Old_age   Always       -       0

SMART Error Log Version: 1
No Errors Logged

SMART Self-test log structure revision number 1
Num  Test_Description    Status                  Remaining  LifeTime(hours)  LBA_of_first_error
# 1  Short offline       Completed without error       00%         6         -

SMART Selective self-test log data structure revision number 1
 SPAN  MIN_LBA  MAX_LBA  CURRENT_TEST_STATUS
    1        0        0  Not_testing
    2        0        0  Not_testing
    3        0        0  Not_testing
    4        0        0  Not_testing
    5        0        0  Not_testing
Selective self-test flags (0x0):
  After scanning selected spans, do NOT read-scan remainder of disk.
If Selective self-test is pending on power-up, resume after 0 minute delay.```

mholl1983, 2022-12-09:
And I tried to put in the first (missing) drive ,but it holds up booting and it even messes things up when I try to add it while the TS is booted up properly.

1000001101000, 2022-12-09:
yeah, SDD basically dead. SDC with the 5 relocated sectors and all those read errors is on its way too most likely.

mholl1983, 2022-12-09:
Sucks, but at least it's good to know officially. Not sure if it's worth the effort, but is there any value in stripping out the drive and plugging them into a full desktop to try to run recovery? 

Had I had my time back, I would've run smartmon a little more often to anticipate these things. Good news is I was looking at the backup on my external 4 TB drive and it looks to be about 70-80% of what was available on the array, so it could be worse. I've learned a lot along the way and am grateful for all your help!

mholl1983, 2022-12-10:
Got to thinking again. When I get my new drive in the mail, is it worth swapping out the existing SDD for the new one and see if the array rebuilds w/ the three drives?

mholl1983, 2022-12-10:
```root@debian:/home/marc# sudo mdadm --examine --scan
ARRAY /dev/md0 UUID=e7194c7c:f3579d9d:529ae12f:b827b7aa
ARRAY /dev/md/1  metadata=1.2 UUID=d6c17ec0:cc93ec32:218a896f:4358e4cf name=TS-XEL-EM02C:1
ARRAY /dev/md/10  metadata=1.2 UUID=db00a7ae:43c0dfeb:89959a8c:b4b581b3 name=TS-XEL-EM02C:10
ARRAY /dev/md/2  metadata=1.2 UUID=46c1d87e:e7b2970a:3edd105f:c51f08fe name=TS-XEL02C:2
root@debian:/home/marc# sudo mount /dev/md/2 /mnsudo mdadm -D /dev/md0t/array
mount: invalid option -- 'D'
Try 'mount --help' for more information.
root@debian:/home/marc# sudo mdadm -D /dev/md2
mdadm: cannot open /dev/md2: No such file or directory
root@debian:/home/marc# sudo mdadm -D /dev/md/2
mdadm: cannot open /dev/md/2: No such file or directory
root@debian:/home/marc# sudo mdadm -D /dev/md/2
mdadm: cannot open /dev/md/2: No such file or directory
root@debian:/home/marc# mdadm --assemble --force --uuid 46c1d87e:e7b2970a:3edd105f:c51f08fe /dev/md/2
bash: mdadm: command not found
root@debian:/home/marc# sudo mdadm --assemble --force --uuid 46c1d87e:e7b2970a:3edd105f:c51f08fe /dev/md/2
mdadm: /dev/md/2 assembled from 2 drives - not enough to start the array.
```

mholl1983, 2022-12-10:
Hadn't tried w/ the uuid option before. That seems to result in a more "realistic" outcome, given the 2 drives assembling (as opposed to 3 when i manually force them, but then the array does a disappearing act)

1000001101000, 2022-12-10:
it needs all 3 stable to do anything with a fourth.

it might be worth a shot taking a new drive and the drive that is failing and putting them into a dock or something and trying to clone it. there are tools like ddrescue that will keep trying to read the drive even if hitting IO errors and even support multiple tries at reads/etc. 

I thought there used to be tricks for trying to get a dying drive to behave long enough to try to clone it, but the top google result makes it sound the the "freezer trick" isn't really a thing any more:
https://drivesaversdatarecovery.com/why-the-freezer-trick-for-hard-drives-doesnt-work/#:~:text=Bottom%20line%2C%20modern%20hard%20drives,destroy%20the%20hard%20drive%20permanently.

mholl1983, 2023-02-05:
Hey again! So just an update that I decided to retire the TS-XEL, at least for the time being. I might keep it around and do a side project with it at some point, but I'll be shifting to a full tower setup with Windows or Linux (likely TrueNAS, but will think on it more). A few questions:

* Would I need to use all 4 drive bays if i wanted to revive the TS down the road? Can I get away with 2 drives as a minimum?
* If I do decide to truly retire it/sell/donate, is there a way to factory reset the box even with Debian on there?

Thanks again for all your help. I learned a lot along the way.

1000001101000, 2023-02-06:
The OS is one the drives so you can always wipe the drives and install the stock firmware or debian using the instructions on the wiki etc.

Debian can run on any number of disks. The stock firmware can too though you need to tweak something to use less than 4 in some cases. Theres a note on the wiki about that part.

I’ve been playing around with a few different backup strategies lately mixing zfs send/receive and simpler rsync based backups. I’ve also played a little bit with iscsi. most terastations are mostly good enough for most of that though zfs on 32-bit devices is unsupported now (though I did get it to work on a device as an experiment).

as write all of this I realize it’s probably better suited to discord rather than a github issue. In any case, good luck!

mholl1983, 2023-02-06:
np, sounds good! Thanks so much again for all your help. I can reach out on Discord if anything arises and I need help. Will close this ticket for now. Cheers!

closed 2023-02-06 by mholl1983