1000001101000/Debian_on_Buffalo #164: LS-WVL can not SSH and file browser after install Webmin

khoahoc0508, 2022-12-26:
First, thank you for the great guide to installing Debian. My ls-wvl is working perfectly, but after trying to install Webmin, I can not ssh and can not access the shared folder. I still access and config Webmin, so please help me with this situation.

Sincerely,
Trung

1000001101000, 2022-12-26:
Good to hear from you!

I haven't done much with Webmin. It sounds like it may have made some sort of change to your SSH config on your behalf that broke things but that is just a guess.

khoahoc0508, 2022-12-27:
Can you give a guide on how to check and fix if the ssh and samba servers are broken?

Sincerely,
Trung

1000001101000, 2022-12-27:
You could try restoring sshd.conf and/or smb.conf. I could imagine the issues could also be related to file permissions (for samba shares) or user account/group membership.

“What changes did webmin make to my system and how do I undo them?” is probably a better question for the webmin folks since I know very little about webmin.

khoahoc0508, 2022-12-31:
Thank but I can make it work again, so I reinstalled and forgot webmin.

closed 2023-06-07 by 1000001101000