1000001101000/Debian_on_Buffalo #165: [LS-WXL] No temperature sensors?

Panakotta00, 2022-12-26:
Not sure if the LS-WXL in fact doesn't have any temperature sensors,
but when I run `sudo sensors-detect` I only get as result:
```
# sensors-detect version 3.6.0
# DMI data unavailable, please consider installing dmidecode 2.7
# or later for better results.
# Kernel: 5.10.0-20-marvell armv5tel
# Processor: Feroceon 88FR131 rev 1 (v5l)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no):
modprobe: FATAL: Module cpuid not found in directory /lib/modules/5.10.0-20-marvell
Failed to load module cpuid.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no):
Sorry, no supported PCI bus adapters found.

Sorry, no sensors were detected.
Either your system has no sensors, or they are not supported, or
they are connected to an I2C or SMBus adapter that is not
supported. If you find out what chips are on your board, check
https://hwmon.wiki.kernel.org/device_support_status for driver status.
```
Output of `sudo sensors`:
```
gpio_fan-isa-0000
Adapter: ISA adapter
fan1:        5000 RPM  (min =    0 RPM, max = 5000 RPM)
```

So it seems it can not detect any temperature sensors.

1000001101000, 2022-12-26:
Good to hear from you!

I don't think that SoC (88f6281) has a temperature sensor, at least the device-tree in the linux kernel source doesn't appear to list one. 

One option would be to use the `drivetemp` module to monitor the hard drive temperature rather than the CPU.

Panakotta00, 2022-12-26:
Thanks for your help ^^

closed 2022-12-26 by Panakotta00